bytespider/Meross #1: Certificate Management

nebula-it, 2019-03-13:
Hey bytespider,

Thank you for the extensive work. I finally got my MSS310 and Im trying to push the config change to point it to my local MQTT server. I had quite some trouble with running the node.js app, I wasn't able to install the dependencies, netroute was failing to install on Windows. I even tried it on WSL Ubuntu but it was failing there as well so I finally gave up and hard coded gateway address and removed netroute from package.json. With this I was finally able to run "meross info wifi0" and it returned the values, Next I'm moving to "meross setup" but I want to confirm the process beforehand, If I create a cert from my local CA and use it on MQTT server how are we forcing the MSS310 to trust that cert? 

Also, I'm no developer by any shot and this is my first time working a node.js app, I tried to go over the code seems like its doing everything with the HTTP requests (Get, Post) is that correct? If so I would try to implement this in PowerShell.
Finally, the console that you got to via serial cable can be accessed by telnet as well by connecting to MSS310's Wi-Fi but once its connected to a Wi-Fi and is communicating with iot.meross.com it will block any attempts to its IP by telnet or SSH. The menu options were in different place than in serial output that you posted but that might have changed with the firmware. Also MSS310 I got has lot more pins on the backside than just 4 I can see on the photos you posted. (Just little observation)
Thank you again for the great work. Cheers

bytespider, 2019-03-13:
I must admit i worked with MacOS entirely developing it, so I’m not surprised netroute didnt work. Perhaps I could add a option to override the gateway address.

All the node app does is to send http requests to the device, so you could easily write an app in any language to speak to the device.

You bring up a great point about certificate trust, it didnt seem to be an issue, perhaps the device only cares that the packets are encrypted. That would be an issue with people using stock deviced with meross servers and someone spoofing messages but thats off topic.

Did you open up your MSS310? Id be intregued to see whats inside yours, perhaps a newer or older version. I know there was mention of a second version on https://github.com/albertogeniola/MerossIot/issues

I’ll certainly try telnet via wifi. As for menu location you’re probably right about it being differing firmware versions.

nebula-it, 2019-03-13:
I'm getting an ubunut box setup just to test it on there with netroute. But yeah I think giving an option to add a switch for IP of MSS310 would be better option. Something like `./meross info -IP 10.10.10.1` should work.

I'm learning powershell so this will be a great project for me to do.

That is good to know that device only cares about encryption. So are you getting power usage data fine in your local MQTT? For the stock devices they might have added their own root CA cert to the devices. Its kinda bad that devices do not check the trust of the cert but i guess its good thing for us :)

Yes I did open it up, it has newer manufacturing date (January 2019) than one you have. I didn't take the circuit board out though, did it require desoldering the AC power pins? I'll post some photos shortly.

What FW version are you currently running? Mine is running 1.1.18. 

Also my device originally came with "Factory mode: ON" so it was not able to connect to my home WiFi once I turned it off. Factory mode might allow some additional configuration from the telnet menu, I'm still exploring and will update here if I find something interesting.
Cheers

[image]

nebula-it, 2019-03-15:
Here are the internals of mine

bytespider, 2019-03-18:
Hi sorry for the delayed response.

My FW version according to the device is 2.1.7 BuildDate :2018/11/08 09:58:45 GMT +08:00 Also your date looks to be JAN 2018, Giving the different FW versions It would seem they changed a lot. I would upgrade the firmware to a newer version if possible.

Yes i desoldered the power pins, but that doesn't seem needed as your pins are easily accessible.

nebula-it, 2019-03-19:
No problem.
Yeah I read the date wrong. I looked for the FW upgrade but 1.1.18 is the latest one for mine.

bytespider, 2019-04-01:
Closing due to inactivity

closed 2019-04-01 by bytespider