bytespider/Meross #2: Unable to connect to WiFi

nebula-it, 2019-03-15:
Hey bytespider,

I have been fighting with this for past few days but still unable to connect to my WiFi using the ./meross setup command. I ran packer capture on my WiFi interface no DHCP requests there or even a single packet from the device, seems like its not able to connect at all. Any ideas?

bytespider, 2019-03-18:
Do you have any luck pairing it normally following the Meross instructions?

If you can then you need put it back into setup mode by removing the device from the app. Then you need to connect directly to the Meross AP, hopefully then the tool will work.

nebula-it, 2019-03-19:
Yes, Im able to pair using Meross's instructions.
It even accepts the ./meross setup command from the tool and then restarts to connect with my WiFi, its flashing green as well means its trying to connect. But I don't see any DHCP request packets from it. I will try using a different Wireless AP and will try to connect to that.
Also, my MQTT server does not have a cert installed, Its running on port 8883 though. Will that have any affect on this ? I'm guessing it should still get the IP address from DHCP before reaching out to MQTT.

nebula-it, 2019-03-19:
I was checking the log from telnet while connecting from the ./meross tool and while connecting from the Meross official app:
Here is the difference, but I'm still trying to decipher it.
When config is done from the Meross APP:

`==========[local]=========>>>>
[http]: file "config" not found
<<<<======[local]=========
messageId: 2954dd0fe67891ed15392da981fd83a8 namespace:Appliance.Config.Wifi method: SET from: http://10.10.10.1/config timestamp:1552968027
CFG_commit: 45 update!
CFG_save: write 4912 bytes to flash:CFG_CHKSUM=0e61eda9, len=4912
CFG_save: old offset=10b4, new=212c
..CFG_commit: 0 update!
messageId: 2954dd0fe67891ed15392da981fd83a8 namespace:Appliance.Config.Wifi method: SETACK from: /appliance/18030627209728251a0134298f11665f/publish timestamp:1552968027
body len 289
==========[local]=========>>>>
ntp_update!
http_config(): zweb_location:801e808e
mon_main:line:283 | WAN_start 0`

When ./meross setup is used from the tool:

`==========[local]=========>>>>
[http]: file "config" not found
<<<<======[local]=========
messageId: 1 namespace:Appliance.Config.Wifi method: SET from: <null> timestamp:-1
CFG_commit: 26 update!
CFG_save: write 4264 bytes to flash:CFG_CHKSUM=c44a008e, len=4264
CFG_save: flsh run-out, erase.
.....CFG_commit: 0 update!
messageId: 1 namespace:Appliance.Config.Wifi method: SETACK from: /appliance/18030627209728251a0134298f11665f/publish timestamp:105
body len 250
==========[local]=========>>>>
mon_main:line:283 | WAN_start 0`

bytespider, 2019-03-19:
Yes it should request an IP from your DHCP device and once it has one attempt to connect to the MQTT server.

I found that I needed to have a certificate on the MQTT server or else mosquito and other clients wouldnt use secure messaging.

If you are able to telnet, maybe you can see what the settings are after you did './meross setup` might give you a clue as to why it’s not cnnecting to your AP.

nebula-it, 2019-03-24:
The ./meross setup didn't work for me (Im pretty sure, Im doing something wrong) so to improvise I set up the MS310 with the app, using their standard procedure. Once connected to WiFi I powered off MS310, change my DNS to point iot.meross.com and smart.meross.com to my local MQTT server(Its running on port 2001). Powered on the MS310 and voila! its connected to my local MQTT.

Now its posting the onoff action from switch to the MQTT, however Im unable to get the power consumption data. Do you know if MS310 constantly posts the power consumption data to MQTT or you have to request it manually?

Also between your guide and @albertogeniola 's I think I read somewhere the detailed make up of how the payload for requests are made up from timestamps and other values but I'm not able to find it anymore. If you recall it can you please link it here. I'm trying to publish onoff value to MQTT server.  Thanks

bytespider, 2019-03-24:
I think what you’re referring to is https://github.com/albertogeniola/MerossIot/wiki/HTTP-APIs

I beleive you have to request the data as I too expected it to stream the data periodically.

bytespider, 2019-04-01:
Closing due to inactivity

duceduc, 2024-02-05:
@nebula-it 

    Once connected to WiFi I powered off MS310, change my DNS to point iot.meross.com and smart.meross.com to my local MQTT server(Its running on port 2001). Powered on the MS310 and voila! its connected to my local MQTT.

Hi. Can you eleborate on how you connect your device to wifi locally using your router?? I am having issue as well. The device would reboot after running ./meross setup, but it goes back to pairing mode.

bytespider, 2024-02-05:
@duceduc are you able to connect using the official app and does your device use Bluetooth or WIFI pairing?

duceduc, 2024-02-05:
Yes, I am able to connect to wifi using the app. The device I have is a mss110 plug which only has wifi. No bluetooth. When I was in the app, there was an update to this device which I updated. Tried to pair again locally. Still no go.

bytespider, 2024-02-06:
Try this:
1. Set up the device using the app
2. Find your device key and Meross user ID, [Meross LAN for Home Assistant](https://github.com/krahabb/meross_lan) can do this.
3. Reconfigure the device using the IP of the device on your network with a command like `npx meross setup --nowifi --gateway 192.168.1.123 --mqtt mqtts://192.168.1.100`

closed 2024-11-11 by bytespider