bytespider/Meross #3: (node:2690) UnhandledPromiseRejectionWarning : Problem

EddyBishop, 2019-03-31:
Hi Rob,
I am not a skilled programmer, but I'm still trying, following your instructions, to connect my MSS310 to the MQTT broker I use with Home Assistant.

I installed the [merossiot](https://github.com/albertogeniola/MerossIot) libraries and the various dependencies related to the execution of your scripts. Mosquitto is regularly installed and working and I entered the basic configuration in the Mosquitto .conf file.

By executing the command:

> ./meross info wlan0

I receive the information concerning the device ID.

However, I execute the command: 

> ./meross setup wlan0 --wifi-ssid xWIFISSIDx --wifi-pass xWIFIPSWx --mqtt 192.168.1.xx: 8883

I receive this error message:

> Setting up device with IP 10.10.10.1 on wlan0
> (node: 2690) UnhandledPromiseRejectionWarning: ReferenceError: URL is not defined
>     at mqtt.map (/home/pi/Meross/bin/src/api.js:51:23)
>     at Array.map (<anonymous>)
>     at API.configureMqttServers (/home/pi/Meross/bin/src/api.js:48:24)
>     at / home / pi / Meross / bin / src / meross-setup: 43: 30
>     at Object. <anonymous> (/ home / pi / Meross / bin / src / meross-setup: 51: 3)
>     at Module._compile (module.js: 652: 30)
>     at Object.Module._extensions..js (module.js: 663: 10)
>     at Module.load (module.js: 565: 32)
>     at tryModuleLoad (module.js: 505: 12)
>     at Function.Module._load (module.js: 497: 3)
> (node: 2690) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, $
> (node: 2690) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.j $

I can't understand where I'm wrong.
Could you help me?

Sorry for my bad English, I'm writing from Italy hahaha
Thanks a lot!

bytespider, 2019-03-31:
>     at Object. (/ home / pi / Meross / bin / src / meross-setup: 51: 3)

It is likey your version of NodeJS. Can you tell me which version you are using?

I tend to run multiple versions and its possible I coded against an older version by mistake.

EddyBishop, 2019-03-31:
Hi Rob!
Thanks for your quick response!

My NodeJS vers is: v8.15.1

bytespider, 2019-03-31:
Ah yes URL wasnt in that version. I will code arround it and include the URL dependancy

EddyBishop, 2019-03-31:
Oh thanks a lot!

With which version are you sure it works?

I can always downgrade NodeJS to a version compatible with your script

bytespider, 2019-04-01:
I've updated the code to be used on Node 8.
I've also remove the interface in favour of --gateway <ip>

Let me know if it works for you.

EddyBishop, 2019-04-01:
Hi Rob,
thank you very much for your interest!
I just tested the new script and...

> sudo ./meross info --gateway 10.10.10.1

I receive the information of the object correctly

> sudo ./meross setup --gateway 10.10.10.1 --wifi-ssid WIFISSID --wifi-pass WIFIPSW --mqtt 192.168.1.xx: 8883

The script works and the socket gets the new setup, then it restarts and tries to connect to the wifi, but unfortunately it can't connect correctly to my wifi and restarts after a few minutes :(

This is the code I receive after sending the "./meross setup":
> Setting up device with IP 10.10.10.1
> Setting MQTT servers [ { host: '192.168.1.15', port: '8883' } ]
> sending payload { header: 
>    { method: 'SET',
>      namespace: 'Appliance.Config.Key',
>      messageId: '1' },
>   payload: { key: { gateway: [Object], key: '', userId: '' } } }
> sending payload { header: 
>    { method: 'SET',
>      namespace: 'Appliance.Config.Wifi',
>      messageId: '1' },
>   payload: 
>    { wifi: 
>       { ssid: 'VEFSRElTX0VYVA==',
>         password: 'RGVjaW1vRG90dG9yZTEwIQ==' } } }
> { header: 
>    { messageId: '1',
>      namespace: 'Appliance.Config.Wifi',
>      method: 'SETACK',
>      payloadVersion: 1,
>      from: '/appliance/1805034347250425180134298f116cf7/publish',
>      timestamp: 454,
>      timestampMs: 670,
>      sign: '3bbfdde8842a5c44a0323518eec97cbe' },
>   payload: {} }

Is it possible that the problem is in this part of the code?
> sending payload { header: 
>    { method: 'SET',
>      namespace: 'Appliance.Config.Wifi',
>      messageId: '1' },
>   payload: 
>    { wifi: 
>       { ssid: 'VEFSRElTX0VYVA==',
>         password: 'RGVjaW1vRG90dG9yZTEwIQ==' } } }

I noticed that it shows an ssid ('VEFSRElTX0VYVA==') and a password ('RGVjaW1vRG90dG9yZTEwIQ==') different from the ones I enter on the command string.

Sorry again for my horrible English!

bytespider, 2019-04-01:
No the values are base64 encoded and sent to the device as thats what the devices expect.

If the device light flashes green then resets I expect either your WIFI credentials are incorrect (or you’re using 5Ghz? I think these only support 2.4Ghz) or it cannot connect to your mqtt server.

Test if you can connect to the MQTT server via mosquito, you have to pass it the certificate using the `—cafile` option something along the lines of `mosquitto_sub -h 192.168.1.15 -p 8883 -t topic --cafile /path/tp/cafile.pem`

EddyBishop, 2019-04-03:
Hi rob,
the script works great, I think the problem is my Mosquitto configuration.

Does Mosquitto need to be installed on the same system that I use to configure the MSS310 with your setup script?

I have to connect the MSS310 to the mosquitto broker installed on my Raspberry (on which Home Assistant runs), is it possible that the right certs for connection to the MQTT broker are missing on my Raspberry?

Can you recommend a good guide to configure Mosquitto with certs?

Thanks!

bytespider, 2019-04-04:
You'll need Mosquitto client applications order to test the broker on you RasPi.

I followed this tutorial to get mine set up https://mcuoneclipse.com/2017/04/14/enable-secure-communication-with-tls-and-the-mosquitto-broker/

My resulting minimal config is detailed here: [https://github.com/bytespider/Meross/wiki/MQTT](https://github.com/bytespider/Meross/wiki/MQTT)

bytespider, 2019-05-02:
Closing due to inactivity

closed 2019-05-02 by bytespider