bytespider/Meross #4: Interacting with topics of MS310 on local MQTT

nebula-it, 2019-04-17:
Hello again @bytespider 
Was busy for few day with a server migration and now back to this. I was able to connect to local MQTT server by just pointing the iot.meross.com and smart.meross.com DNS entries to my local MQTT server.  The device publishes info to MQTT fine, e.g if I use the physical button to turn on/off this is the MQTT message that is posted:
`{"header":
    {"messageId":"ac071f0d81fe2e07d1c2a9c4c8f523a7",
        "namespace":"Appliance.Control.Toggle",
        "method":"PUSH",
        "payloadVersion":1,
        "from":"/appliance/mydeviceUUID/publish",
        "timestamp":1553413590,
        "timestampMs":440,
        "sign":"e410105ebe776395645a9aa33ce8b919"},
        "payload":
        {"toggle":{
            "onoff":0,
            "lmTime":1553413590}
            }
        }`

I'm wondering how you are posting mesages to topic that MS310 is subscribed to so we can control it directly from there. I looked on other MS310 libraries but they are all using HTTP POST requests to interact with MS310. This seems like a json message that can be published and then MS310 will read it from its subscribed topics. Have you done something in this regards?
Cheers

bytespider, 2019-04-18:
I haven't done much that way so far, other commitments, but based on the work by [albertogeniola/MerossIot](https://github.com/albertogeniola/MerossIot) I believe that you just need to publish the correct JSON to `/appliance/<device uuid>/subscribe`

nebula-it, 2019-04-19:
Thank you, I'll try that.

closed 2019-04-19 by nebula-it