bytespider/Meross #6: Info fails for MSS520H

accuser, 2019-05-02:
Getting info from an MSS520H with 3.1.1 firmware fails:

```
sending payload { header:
   { method: 'GET',
     namespace: 'Appliance.System.All',
     messageId: '1' },
  payload: {} }
{ RequestError: Error: read ECONNRESET
    at new RequestError (/home/pi/Meross/bin/src/node_modules/request-promise-core/lib/errors.js:14:15)
    at Request.plumbing.callback (/home/pi/Meross/bin/src/node_modules/request-promise-core/lib/plumbing.js:87:29)
    at Request.RP$callback [as _callback] (/home/pi/Meross/bin/src/node_modules/request-promise-core/lib/plumbing.js:46:31)
    at self.callback (/home/pi/Meross/bin/src/node_modules/request/request.js:185:22)
    at Request.emit (events.js:189:13)
    at Request.onRequestError (/home/pi/Meross/bin/src/node_modules/request/request.js:881:8)
    at ClientRequest.emit (events.js:189:13)
    at Socket.socketErrorListener (_http_client.js:392:9)
    at Socket.emit (events.js:189:13)
    at emitErrorNT (internal/streams/destroy.js:82:8)
  name: 'RequestError',
  message: 'Error: read ECONNRESET',
  cause:
   { Error: read ECONNRESET
       at TCP.onStreamRead (internal/stream_base_commons.js:111:27) errno: 'ECONNRESET', code: 'ECONNRESET', syscall: 'read' },
  error:
   { Error: read ECONNRESET
       at TCP.onStreamRead (internal/stream_base_commons.js:111:27) errno: 'ECONNRESET', code: 'ECONNRESET', syscall: 'read' },
  options:
   { url: 'http://192.168.1.171/config',
     headers: { 'Content-Type': 'application/json' },
     json: { header: [Object], payload: {} },
     method: 'POST',
     callback: [Function: RP$callback],
     transform: undefined,
     simple: true,
     resolveWithFullResponse: false,
     transform2xxOnly: false },
  response: undefined }
undefined
```

Node is version V10.15.3.

bytespider, 2019-05-02:
Does the device itself do anything? Reset, Flash indicator lights? I don't have this device so it will be tricky to debug.

The IP of the device seems to be `192.168.1.171` which is not typically a device in pairing mode. This tool hasn't been tested on already paired devices.

accuser, 2019-05-02:
Device has been setup to access the wireless network, hence it has an IP address. Not sure how I'd access it otherwise, i.e. there is a need to specify a gateway address.

bytespider, 2019-05-02:
This tool connects to the AP provided by the device during pairing.

1. Put the device in pairing mode.
2. Connect your computer to the AP the device presents (MEROSSxxx)
3. Find the gateway IP of your connection, typically 10.0.0.1
4. Run `./meross info --gateway 10.0.0.1` replacing 10.0.0.1 with your gateway ip.

This should show you information if it connected correctly.

accuser, 2019-05-03:
Appreciate the support, especially as I'd obviously misread the instructions. I can setup the device using the utility but something isn't working correctly, but I'm not sure what yet.

bytespider, 2019-05-03:
The other problem I faced was setting the MQTT broker up in a way the device likes.

accuser, 2019-05-03:
Oh my gosh. How unnecessarily complicated was that? :-)

I've managed to get the switch talking to MQTT locally. I'm only subscribing, not publishing yet, but everything seems to be working. Thanks for your help - I really appreciate it.

closed 2019-05-03 by accuser

bytespider, 2019-05-04:
I’m glad you got it. 👍🏻

accuser, 2019-05-06:
It was working, but now it is not. Been away for the weekend and now it is flashing orange on and off. Can't put it back into pairing mode. :-(

bytespider, 2019-05-06:
Does pressing and holding the button for 10 seconds (or so) put the device into pairing mode? 

If I recall, orange flashing means it can not connect to the WIFI AP or the MQTT server.