bytespider/Meross #11: Not working with hardware version 1.1.18

freakinpenguin, 2020-10-06:
First of all many thanks for you efforts! 

I bought two 3-sets from the MSS310 plugs and while one set works very well with your instructions, the other 3 don't.

I'm quite sure, that this is related to different hardware versions (2.0.0 vs.1.1.18).

I'm just wondering whether you already know this, or maybe there is someone around who knows a solution.

I attached the output of "meross info..." for both versions.
[1_InfoFromMerossScript.txt](https://github.com/bytespider/Meross/files/5336679/1_InfoFromMerossScript.txt)
[2_InfoFromMerossScript.txt](https://github.com/bytespider/Meross/files/5336680/2_InfoFromMerossScript.txt)

bytespider, 2020-10-07:
Hey there,

What specifically isn't working? Do you have the messages that show when
you try to initialize the device?
I don't have hardware version 2 to test against, I don't foresee that being
an issue, more likely my code is targeting v1 firmware as I can see there
is a difference in the message response.

freakinpenguin, 2020-10-08:
Hey!
The problem is, that after the "meross setup"-command is sent (see command + log in the attached file), the socket powers off (expected) and reboots. When the socket starts it is blinking rapid green (according to doc this means it's trying to connect to Wifi - maybe also to the MQTT server). The socket stays in this state and after maybe 3-5 mins it returns into the configuration mode (yellow-green blinking).
I don't see it entering my wifi in my router logs nor a connection to the MQTT server.
[2_SetupCommand+Result.txt](https://github.com/bytespider/Meross/files/5350945/2_SetupCommand%2BResult.txt)

Concerning your wiki-page-edits:
I tried to add mqtt:// and mqtts:// in advance of the mqtt-server-ip-address+port but unfortunately that didn't change anything.

bytespider, 2020-10-09:
If i recall correctly the Meross devices now require port 8883 and TLS
Heres my Mosquitto config

```nginx
port 8883

require_certificate false
use_identity_as_username false

# replace with your CA Root
cafile ../certs/ca.crt

# replace with your server certificate and key paths
certfile ../certs/server.crt
keyfile ../certs/server.key
```

Here's the commands I used to create the certificates and keys

**Create the Certificate Authority**
```sh
openssl genrsa -des3 -out ca.key 2048
openssl req -new -x509 -days 1826 -key ca.key -out ca.crt
```

**Create the certificate signing request.**
It's important when asked for the FQDN in these next step to use the IP or domain name of the machine your MQTT instance is on. 192.168.188.43 according to the details in 2_SetupCommand+Result.txt above.
```sh
openssl genrsa -out server.key 2048
openssl req -new -out server.csr -key server.key
```

**Create the final certificate**
```sh
openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 360
```

freakinpenguin, 2020-10-10:
I tried the values you suggested, but didn't see any difference.
This would also be strange, as I can't see the socket in my routers wifi list. So I guess that setup procedure is slightly different in this hardware version. If I find some time I will power up wireshark and investigate...

Nevertheless, thanks for your suggestions!

bytespider, 2020-10-14:
I'm surprised that your router can't see the device, the only thing I can think is the device, router, and MQTT are not on the same subnet.

Also if you haven't done so, try setting up using the cloud. Some users have reported that worked for them. I'm not convinced this is nessacary one of my plugs never saw the cloud until recently when I paired it for a firmware upgrade.

colinfitzpatrick, 2020-12-21:
Can the offline + mqtt method work with the mss210?  

I've tried without success - the ./meross setup --gateway 10.10.10.1 --wifi-ssid <your-ssid> --wifi-pass <your-pass> --mqtt <mqtt-host-ip>:8883 process failed with a socket hangup error?

bytespider, 2020-12-28:
> --wifi-ssid --wifi-pass --mqtt :8883 process failed with a socket hangup

No idea as I only have MSS310. I advise you connect to Meross and make sure
your devices work correctly before you attempt any hardware or software
hacking.

closed 2021-02-03 by bytespider