bytespider/Meross #12: Failing to authenticate in mqtt server

chegancasb, 2020-10-11:
Hi,

I'm trying to configure a MSS210 switch to connect to my mosquitto broker using your tool.

I've followed your tool and it correctly configures the swtich connect to my wifi network. I also send the data for the mqtt broker:

The command a sent and its response is below (somewhat sensitive data ofuscated).

    $ ./meross setup --gateway 10.10.10.1 --wifi-ssid <SSID> --wifi-pass <SSID-PASSWORD> --mqtt mqtts://192.168.1.<OCT>:8883
    Setting up device with IP 10.10.10.1
    Setting MQTT servers [ { host: '192.168.1.<OCT>', port: '8883' } ]
    sending payload { header:
       { method: 'SET',
         namespace: 'Appliance.Config.Key',
         messageId: 'bea90e4e1bc5f3f921479de4f4ccae45',
         timestamp: 1602454838,
         sign: '2274d8d30f46d2d59f2612564065a0eb' },
      payload:
       { key:
          { gateway: { host: '192.168.1.<OCT>', port: '8883' },
            key: '',
            userId: '' } } }
    sending payload { header:
       { method: 'SET',
         namespace: 'Appliance.Config.Wifi',
         messageId: '1fe27ef18f573a87344984faa71a692d',
         timestamp: 1602454838,
         sign: 'f33a8fe0204aaefdcb3f5540bcd293ad' },
      payload:
       { wifi: { ssid: '<SSID-BASE664>', password: '<SSID-PASSWORD-BASE664>' } } }
    Got response: { header:
       { messageId: '1fe27ef18f573a87344984faa71a692d',
         namespace: 'Appliance.Config.Wifi',
         method: 'SETACK',
         payloadVersion: 1,
         from: '/appliance/<UUID>/publish',
         timestamp: 230,
         timestampMs: 85,
         sign: 'c21cc4813383639bb321519b6c0fe54c' },
      payload: {} }
    $

A few seconds later in the mosquitto logs the following entries appear:

    (...)
    2020-10-11T23:29:47: New connection from 192.168.1.<OCT2> on port 8883.
    2020-10-11T23:29:47: Sending __CONNACK__ to 192.168.1.<OCT2 __(0, 5)__
    2020-10-11T23:29:47: Socket error on client <unknown>, disconnecting.
    (...)

The mosquitto has the following configuration for the tls listener:
    listener 8883

    require_certificate false
    use_identity_as_username false

    cafile /etc/mosquitto/ca_certificates/RootCA.crt
    certfile /etc/mosquitto/certs/mqtt.vpnserver.crt
    keyfile /etc/mosquitto/certs/mqtt.vpnserver.key

I've tried several different configuration options such as the `allow_annonymous` without success.
I've also tried issuing different certificates to use as the server certificates. To the IP address to the host name, etc but nothing worked.

From the log results __CONNACK (...) (0, 5)__ it indicates that the TCP connection was estabilished, that the TLS handshake was successful (and an encrypted TLS channel was estabilished) and also that a MQTT session was requested but failed due to "Connection refused, not authorized" (result code taken from https://www.hivemq.com/blog/mqtt-essentials-part-3-client-broker-connection-establishment/)
This leads me to think that some authentication data is not being sent to/retrieved from the switch so that it's connection is accepted.

Are you configuring anything related with the client authentication on the switch or on the mqtt broker?
Could you please help me?

Thanks.
Regards,
chegancasb

chegancasb, 2020-10-11:
Just a update. 
Had a view on one of the projects mentioned in the close issues https://github.com/shodge12/Meross-Node-Red-Comm/
and with his mosquitto configuration file, with some trial and error, I was able to spot the configuration error:

Setting the parameter __tls_version__ to __tlsv1.1__solved the issue but I am now allowing annonymous logins, which is not what I intended.

If you could please check my previous questions, I would appreciate.

Regards,
chegancasb

bytespider, 2020-10-14:
Older firmware required TLS1.1, it's my understanding that firmware 2.x can use TLS1.2.

Credentials are:
Username: `{device_mac_address}`
Password: `{user_id}_` + md5(`{device_mac_address}{key}`)

For example:
key = 123456
device_mac_address = xx:xx:xx:xx:xx:xx
user_id = 987654

Result:
username: xx:xx:xx:xx:xx:xx
password: 987654_98386d15983ca399b225f92d0b9f6850

chegancasb, 2020-10-14:
Thanks for that. I assume you created a password file with that content and
then setup the broker to use that password file, correct?

Regards,
Bruno

bytespider, 2020-10-14:
I haven't tried it and it may not be possible with Mosquitto because of this:

`Usernames must not contain ":". Passwords are stored in a similar format to crypt(3).`
[https://mosquitto.org/man/mosquitto_passwd-1.html](https://mosquitto.org/man/mosquitto_passwd-1.html)

paolo-hub, 2020-10-18:
I have tried it and I have de following message from my mqtt broker:
`Login attempt or request with invalid authentication from supervisor (172.30.32.2) (Python/3.8 aiohttp/3.6.2)`

paolo-hub, 2020-10-20:
I'm using MQTT Mosquitto broker extension for Home Assistant.

bytespider, 2020-10-21:
Mosquitto won't work unless you use an auth plugin, the password file auth doesn't allow colon in usernames and the username sent by the devices contain colons.

paolo-hub, 2020-10-23:
Do you mean a plug-in on the mosquitto server?

bytespider, 2020-10-26:
Ignore my previous message, I've checked and the HA version of Mosquitto does use the Auth plugin to talk to HA and get the valid logins. https://github.com/home-assistant/hassio-addons/blob/master/mosquitto/DOCS.md

You may need to double-quote your usernames

paolo-hub, 2020-10-26:
Ok you mean something like this?
`logins:
  - username: "mqtt_user"
    password: my_pass_here
anonymous: false
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false`

bytespider, 2020-10-26:
Yes, I would expect that to work.

paolo-hub, 2020-10-26:
Ok many thanks, I will try and I will say to you the result

paolo-hub, 2020-10-27:
hello, ok something happen, the device not connect to the network also if the script not return any error. What happen is that the device go in "on" mode.
here the response with the command:

```
./meross setup --gateway 10.10.10.1 --wifi-ssid "My_Wi_Fi_ID" --wifi-pass “xxxxxx” --mqtt mqtts://192.168.100.5:1883
```

```
Setting up device with IP 10.10.10.1
Setting MQTT servers [ { host: '192.168.100.5', port: '1883' } ]
sending payload {
  header: {
    method: 'SET',
    namespace: 'Appliance.Config.Key',
    messageId: 'd2bcbfd3f47ce01b9ca0f52422225215',
    timestamp: 1603826361,
    sign: '74d3dfe81fce5d914f8b422308db79f7'
  },
  payload: {
    key: {
      gateway: { host: '192.168.100.5', port: '1883' },
      key: '',
      userId: ''
    }
  }
}
sending payload {
  header: {
    method: 'SET',
    namespace: 'Appliance.Config.Wifi',
    messageId: '291d2fd3e3a725ed0574b95e3509b577',
    timestamp: 1603826362,
    sign: '4d821620faf9f3224a25af9c6d602548'
  },
  payload: { wifi: { ssid: 'TWluaSBXLUxBTg==', password: 'xxxxxxx' } }
}
Got response: {
  header: {
    messageId: '291d2fd3e3a725ed0574b95e3509b577',
    namespace: 'Appliance.Config.Wifi',
    method: 'SETACK',
    payloadVersion: 1,
    from: '/appliance/1911158614864390805148e1e9118d52/publish',
    timestamp: 23,
    timestampMs: 636,
    sign: 'add63bb87406cce0f3353d4ca0d62113'
  },
  payload: {}
}

```

bytespider, 2020-10-28:
That all looks good.

paolo-hub, 2020-10-28:
Yes, but I don't understand what's happen, because even if the script provide the wi-fi settings, the meross device (smart plug in this case) don't connect the network, simply change the mode exactly as I if press the botton on/off

bytespider, 2020-10-28:
That script sets up the device to connect to your wifi and to the MQTT broker you passed.

You can then send a message via MQTT to turn the device on or off. You can see how to do that here: https://github.com/albertogeniola/MerossIot/issues/1#issuecomment-713093330

paolo-hub, 2020-10-28:
Ok, but also if the script response without errors (as above) the meross device not connect to the wi-fi

bytespider, 2020-10-28:
What's the LED doing?
https://github.com/bytespider/Meross#led-status

paolo-hub, 2020-10-28:
The device stay in red flashing. When I start the script I can listen a click sound as if I press the on/off power button, and consequentli the below half part or the button mecome blue (to indicate that the power is enable), but the upper half part of the button continuing flashing in red color. In addition The Meros Wi-Fi network remain available.

bytespider, 2020-10-29:
Have you ever had the device working on the Meross Cloud using the official app?

paolo-hub, 2020-10-29:
Yes, I have the device working with the official app, and I use the Meross pligin for Home Assistant to integrate it in the home automation. The problem is that the Meross cloud and Meross plugin not works good together and Home Assistant losts the connection to the Meros device frequently, so I'm trying to move locally on my MQTT broker.
Obviously to connect the meros device to my MQTT broker, I put it in descovery mode and the device disconnect from my wi-fi and the meross app.
The device that I use for test is [Meros MSS310RKIT(IT)]( https://www.amazon.it/gp/product/B079KY6666/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)

bytespider, 2021-02-02:
Did you have any luck?

Does your WIFI SSID or password use non ASCII characters? Whilst the tool supports UTF8 it's possible the issue is encoding related

Please use single quotes to quote the WIFI SSID and password.

paolo-hub, 2021-02-02:
Hi, no The situation is the same. I have SSID and password in UTF 8 compatibility characters.
Ok I will try with single quote '.
Thanks!

bytespider, 2021-02-02:
Perhaps also try with an SSID and password using just a-zA-Z0-9 just in case

bytespider, 2021-02-06:
I've investigated this and it seems that the issue is that the Meross devices don't honour the MQTT 1.3 spec and send a ClientId > 23 characters. As a result, Mosquitto needs to be configured with `use_username_as_clientid true`, however, the way the Home Assistant Mosquitto broker is set up you cannot do this.

krahabb, 2021-02-23:
> Hi, no The situation is the same. I have SSID and password in UTF 8 compatibility characters.
> Ok I will try with single quote '.
> Thanks!

Hello @paolo-hub,
I had the very same issue with the plug accepting the config script setup and connecting to the mosquitto.
After this and the mosquitto dropping the connection (because of auth mess) the plug would revert to 'pairing mode' (led blinking)
I've resolved by using 2 servers in the config script (because the plug has both a primary and a secondary and, my guess is that if the plug encounters an issue on the first one it tries the second but then, missing the second one, it reverts to pairing mode as a failsafe). Of course you can put the very same host:port in the config script and the plug will then loop between the 2 until one works.
Also, to resolve the auth issue, I had to configure allow anonymous in my listener configuration for the broker. no other option working, probably because of all the inconsistencies in the process (but I didn't really dig that much from the time I was able to succesfully connect the plug to mosquitto.
The 'double server' in configuration looks like a must from what I've read so far because, as stated, if the first connection to the broker fails (and that could happen at any time though) the firmware is going to stall and never tries reconnecting.

Dizzyikea, 2021-03-16:
I have been struggling with this for a few days now but today got it working so i thought i would share how

I use Hassio stand alone 64bit image on pi4 with the MQTT addon. I have 3 x [MS310](https://www.amazon.co.uk/gp/product/B083Q162VP/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) plugs.

Trying to connect to the MQTT addon with no SSL on port 1883 failed and i read here i must use SSL and port 8883. Problem is i already have a smappee and some sonoff devices connected to that so i need 1883 available with auth for those, as well as 8883 without auth for the Meross. Nothing i tried would make this work using the default config or turning on "customize:
  active: true" and using a mosquitto.conf. It seems the hassio addon really does not like something about mixing auth 1883 and non auth 8883

My soloution was to setup a 2nd mosquitto broker on my pihole rpi as it does not have much load on it and is always on. So i reverted my hassio MQTT addon back to just 1883 and auth for my exisiting devices. Then on the 2nd broker i used this config

```
allow_anonymous true
require_certificate false

port 1883

listener 8883

cafile /etc/mosquitto/certs/ca.crt
keyfile /etc/mosquitto/certs/server.key
certfile /etc/mosquitto/certs/server.crt
```

The Meross's connect to this broker with no issues.

So now to get this data back over to Hassio's addon i do this

```
customize:
  active: true
  folder: mosquitto
```

This gives me the option to add addition config in \share\mosquitto\mosquitto.conf as follows

```
connection bridge-01
address <ip of 2nd broker>:1883
topic # out 0
topic # in 0
```
All my messages are now copied from the 2nd broker to my hassio addon. That's it all working my Hassio addon has no SSL messing with my existing clients and the other broker has 1883 available to communicate with hassio and 8883 to communicate with the Meross devices.

I'm sure there probably was another way around this but once i had figured this out it took less than 1 hour to implmentate and add all my devices.

Hope this is of use to someone.

closed 2021-03-30 by bytespider

bytespider, 2021-04-16:
Using Meross devices with [Home Assistant Mosquitto](https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO))

userMak, 2021-05-16:
Hi everyone, I have problem to connect me Meross devices to HA with Meross IOT integration, and thought to try with MQTT.
My skills and knowledge are very limited in linux, MQTT so I really don't understand all the steps I have to follow.
So my HA is in a NUC with ubuntu 18.04 and I have already installed Mosquito in HA and is working fine with some of my devices.
Below is the config of my mosquito broker.

[image]

my best and only successful attempt to connect one of my Meross wall switches to HA was with albertogeniolas android app
https://github.com/albertogeniola/MerossIot/issues/162

I install it to my phone, and I manage to connect on of my switches with app and tried to pair it with my mosquito broker but I got the following error
`1621157966: Socket error on client <unknown>, disconnecting.
1621157990: New connection from 192.168.1.50 on port 1883.`

I have read that I probably need a second MQTT broker but I am sure I will be lost in order to set it up etc but now I saw the above post from @bytespider https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO)
and I am wondering if with my above setup it could work or if there is an easy way (for noobs like me) to finally use this devices.

bytespider, 2021-05-16:
@userMak You can absolutely do this without a second broker.

If you're using my tool then the guide above should get you in the right direction. It looks to me like you used the non-TLS port. **Meross devices require a TLS port such as 8883**.

The simplified steps are as follows:
Prerequisite: You need to have NodeJS installed and be able to run the Meross tool from the command line.
1. On a computer with a WIFI card/adapter connect to the Access Point of a Meross device in pairing mode. It's probably best to use something you're familiar with such as Windows or Mac.
2. From the command line run `npx meross info`. This should spit out a bunch of info, the most important being the MAC address and password.
3. Using the details from 2, create a user in HA with the MAC as the username and the password as the password.
4. From the command line run `npx meross setup` providing the `--wifi-ssid`, `--wifi-pass`, and `--mqtt` options with the correct information for your set up. The MQTT URL needs to be the TLS variant. For example `mqtts://192.168.2.100`
5. The device will reboot and you should watch your Mosquitto logs for any info pertaining to the connection.

userMak, 2021-05-16:
@bytespider 
That's good news. So I will try to install in my NUC the NodeJS (I just saw how to do it) but how can I use your tool? 
Sorry fir my ignorance but how can I install it in my NUC?
Thanks

bytespider, 2021-05-16:
is the NUC the only computer you have?

userMak, 2021-05-16:
I have a windows 10 laptop too

bytespider, 2021-05-16:
@userMak youll probably find it easier to use that as you only need my tool once to set up the devices. Here's a tutorial on installing node https://www.fosstechnix.com/how-to-install-node-js-on-windows/

userMak, 2021-05-16:
@bytespider 
ok, I have the password and I have added the user in HA. Now I am stuck a little with
From the command line run npx meross setup providing the --wifi-ssid, --wifi-pass, and --mqtt options with the correct information for your set up. The MQTT URL needs to be the TLS variant. For example mqtts://192.168.2.100

please can you give me an example with your credentials to adapt them?

bytespider, 2021-05-16:
`npx meross setup --wifi-ssid 'my-wifi' --wifi-pass 'supersecretpassword' --mqtt 'mqtts://192.168.2.100'` 
You're literally providing the device with your wifi credentials and the IP of you HA NUC

userMak, 2021-05-16:
well I am getting close but although I listened the device (clicked) and get the following the led is blinking red and not green.
any thoughts? 

[image]

bytespider, 2021-05-16:
did anything show up in your mosquitto logs?

Is it red or orange/red alternating?

userMak, 2021-05-16:
no I can't see anything in mosquito logs.

usually when I try to connect it to meross app is red/green alternating 
now I am not sure 
the led is red and blinking I think

bytespider, 2021-05-16:
Okay, it's possible that you need to give more info to the device.

Put the device back into pairing, connect to it's AP and do `npx meross info --include-wifi`. This time it will also ask the device to scan for local wifi and provide a list.

Find yours and then proceed to set up the device again.

This time when you do `npx meross setup` add the options for `--wifi-encryption`, `--wifi-cipher`, and `--wifi-channel` as listed in the previous command's output.

Example: `npx meross setup --wifi-ssid 'my-wifi' --wifi-pass 'supersecretpassword' --wifi-encryption 7 --wifi-cipher 6 --wifi-channel 1 --mqtt 'mqtts://192.168.2.100'`

userMak, 2021-05-16:
I tried but I am getting the same result.
In the HA users I had in display username the name of the device (meross dining) and at username the mac address.
should I make like this?
[image]

userMak, 2021-05-16:
now I am thinking about it, I have set username and password in mosquito so all the other devices are using password and username. is there a way to add these credentials?

bytespider, 2021-05-16:
> now I am thinking about it, I have set username and password in mosquito so all the other devices are using password and username. is there a way to add these credentials?

No the Meross devices connect to the MQTT broker using the predefined MAC address and password.

The user display name in HA doesn't matter only the username and password.

Is your broker and device on the same subnet? For instance 192.168.1.x

userMak, 2021-05-16:
Yes they are. 
That's a pitty . I was so close

bytespider, 2021-05-16:
Does your WiFi credentials contain any accented characters?

userMak, 2021-05-16:
No just letters and numbers

bytespider, 2021-05-16:
Does your wall switch appear to reboot after setup? For instance with the wall plugs you hear the internal relay click on and off?

userMak, 2021-05-16:
Yes, i can hear the click.wifi is 2.5ghz and meross app is working correctly.

bytespider, 2021-05-16:
Can you try `anonymous:true` in your Mosquito broker and see if that helps

userMak, 2021-05-16:
I just did. I can hear the click that when it is connecting, and what I notice after the click the red led is blinking faster than before (when it is in pair mode), but the led is red and no logs in mosquito.

Lets assume that it will connect to mosquito which topic should I use. now I just place #

bytespider, 2021-05-16:
The topics are /appliance/<device uuid>/subscribe which the device subscribes to and /appliance/<device uuid>/publish to which the device publishes. The uuid of the device is detailed in the info stage but you can easily also just watch the messages coming into the broker to see them.

I'm running out of ideas of why your device cannot connect to the broker. I'm sure it's an issue in my tool but I'm not sure what.

Perhaps you could try https://github.com/albertogeniola/meross_pair an android app that has similar function to this tool and see if that helps.

userMak, 2021-05-16:
the only links I have found for the android app is the following
https://github.com/albertogeniola/meross_pair/releases/tag/0.0.1a

The logic is the same like yours. I will keep the user in HA with the same credentials and pair the device with the phone?

bytespider, 2021-05-16:
Yeah try it. It may work. It may not. This stuff is hard to debug when you're not there with the same  device and the same setup

userMak, 2021-05-16:
I just did. Unfortunately when I use the 8883 port it has the same results like yours. 
When I use the 1883 port it is trying to connect to the mosquito broker but I get the following
`1621195895: New connection from 192.168.1.50 on port 1883.
1621195895: Socket error on client <unknown>, disconnecting.`

the good thing here is that the 192.168.1.50 ip is correct (is the ip I have given to particular wall switch).
Anyway thanks for your time and your help!

bytespider, 2021-05-17:
Did you try anonymous:true in your Mosquito config?

bytespider, 2021-05-17:
Can you post your Mosquitto config, both the ports and the config tabs.

userMak, 2021-05-17:
good morning!
yes I have try with anonymous:true. I leave it "true" from yesterday evening

[image]

bytespider, 2021-05-17:
Okay lets try this.
Connect the switch to the Meross servers and check if there is a firmware update. If there is, please report back before updating.

userMak, 2021-05-17:
No there are no updates

bytespider, 2021-05-17:
@userMak can you email me to continue this?

[7 more replies]