bytespider/Meross #14: MQTT and Homekit

wullimann, 2021-01-12:
Hi,

I configured a custom mqtt server to my switches that are already setup with homekit. The servers a stored properly in the devices but the mqtt server logs don't show any activity from the switches to connect the server. Do you know if mqtt action is disabled when setup with homekit?

Martin

bytespider, 2021-01-13:
Can you describe the steps you took to set the devices up?

wullimann, 2021-01-13:
First I set it up for Homekit with my iPad. After it registered with my wifi I changed the mqtt server

    meross setup --gateway 192.168.64.124 --mqtt 192.168.64.15:8883 --wifi-ssid xxxxxx --wifi-pass xxxxx 

I checked:

```
meross info --gateway 192.168.64.124
Getting info about device with IP 192.168.64.124
sending payload {
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '121ff074341e256b6bce3f0c46207cef',
    timestamp: 1610539791,
    sign: '2ee444722d192346a86f5979e7b0706c'
  },
  payload: {}
}
Got response: {
  header: {
    messageId: '121ff074341e256b6bce3f0c46207cef',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/2009091137638951852148e1e93155f4/publish',
    timestamp: 1610539792,
    timestampMs: 695,
    sign: 'fbb03c858deb13206d79ab20e0f15513'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss210',
          subType: 'us',
          version: '4.0.0',
          chipType: 'MT7686',
          uuid: '2009091137638951852148e1e93155f4',
          macAddress: '48:e1:e9:31:55:f4'
        },
        firmware: {
          version: '4.1.8',
          homekitVersion: '2.0.1',
          compileTime: 'Aug  1 2020 10:11:00',
          wifiMac: '74:ac:b9:b1:c5:67',
          innerIp: '192.168.64.124',
          server: '192.168.64.15',
          port: 8883,
          userId: 0
        },
        time: { timestamp: 1610539792, timezone: '', timeRule: [] },
        online: { status: 0, bindId: 'NFm1h1kUsR7LmoGz', who: 1 }
      },
      digest: {
        togglex: [ { channel: 0, onoff: 0, lmTime: 1610468663 } ],
        triggerx: [],
        timerx: []
      }
    }
  }
}
```

But there is ablosutely no activity in die mosquitto log:

```
/var/log/mosquitto$ sudo tail -f mosquitto.log 
1610522949: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
1610524750: Saving in-memory database to /var/lib/mosquitto/mosquitto.db.
```

The MQTT Explorer on my Mac shows no activity either on the server. Since my setup differs from the setup in the docs here in terms of homekit, I deduced that this is the catch.

bytespider, 2021-01-13:
I would suggest you follow the guide first to make sure that your MQTT broker is configured correctly and you can control the device as per this comment https://github.com/albertogeniola/MerossIot/issues/1#issuecomment-718904673
(you'll need to change the UUID in the messages)

Once you're happy it works, try again to setup with Homkeit and then MQTT

closed 2021-04-16 by bytespider