bytespider/Meross #15: MSS110 network configuration fails

midnightgrue, 2021-01-20:
Hello,

I've been trying for a few days, but I'm afraid I am totally unable to make this code work. I'm sure it's my fault, but I would be grateful for some suggestions on where to direct my troubleshooting because I'm feeling pretty stupid at the moment...

If I put the plug in pairing mode and connect to its AP, this utility is successfully able to query the plug with the 'info' command, and it is successfully able to send payloads to the plug with the 'setup' command. The plug restarts as expected, but it never manages to join the wireless network. There is no indication on my router that the plug is attempting to connect; there are no errors, no connection attempts, nothing at all from that MAC address as far as the router is concerned. (Obviously this explains why no mqtt connections ever show up in the server log either.) The plug just blinks green for a long time, and eventually it gives up and resets itself back into pairing mode.

My network is all one single subnet. The official Meross app has no trouble joining the plug to any of my 2.4GHz SSIDs; it is only when I try to use this code that it fails.

I have tried:
-- setting up a separate wifi network with a simple SSID and password (to avoid any problems with password special characters)
-- using both guest mode and normal mode for the 2.4GHz network on my router
-- many various iterations of quoting and double-quoting the SSID and password in the setup command
-- using two separate node.js installs on different platforms (Kali Linux and MacOS Catalina)
-- upgrading the plug firmware using the Meross app (successfully: it's on 1.1.28 now)
-- running wireshark to see if I can spot any clues to the connection failure (none: I see the payloads, but after the plug restarts I see no broadcasts or anything on the destination SSID net)

I have also followed the SSL configuration you describe, and I am able to connect via TLS to my mqtt server no problem (using the openssl client), but as the plug never actually shows up in the router I don't *think* the mqtt or tls configuration is the issue.

Here is the output of the commands when I execute them:

~~~
$ bin/src/meross info --gateway 10.10.10.1
Getting info about device with IP 10.10.10.1
sending payload {
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '52555f34249a6e75a64f2309df0fe846',
    timestamp: 1611114830,
    sign: '083d7ffaf12355231cfbe3a692f51f8b'
  },
  payload: {}
}
Got response: {
  header: {
    messageId: '52555f34249a6e75a64f2309df0fe846',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/1712126291560325130734298f107d35/publish',
    timestamp: 63,
    timestampMs: 610,
    sign: '7766643a4c322e46f7e6c10400b22dcd'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss110',
          subType: 'us',
          version: '1.0.0',
          chipType: 'MT7688',
          uuid: '1712126291560325130734298f107d35',
          macAddress: '34:29:8f:10:7d:35'
        },
        firmware: {
          version: '1.1.28',
          compileTime: '2019-11-25 16:15:37',
          wifiMac: '01:00:00:00:a8:e5',
          innerIp: '0.0.0.0',
          server: '',
          port: 0,
          secondServer: '',
          secondPort: 0,
          userId: 0
        },
        time: { timestamp: 63, timezone: '', timeRule: [] },
        online: { status: 2 }
      },
      control: { toggle: { onoff: 1, lmTime: 0 }, trigger: [], timer: [] }
    }
  }
}

$ bin/src/meross setup --gateway 10.10.10.1 --wifi-ssid TEST --wifi-pass 123456789 --mqtt mqtts://192.168.2.7:8883
Setting up device with IP 10.10.10.1
Setting MQTT servers [ { host: '192.168.2.7', port: '8883' } ]
sending payload {
  header: {
    method: 'SET',
    namespace: 'Appliance.Config.Key',
    messageId: 'aec466c8270c05f6c23fe351c38322a1',
    timestamp: 1611115057,
    sign: '8ba32cd78956600b277393cf965a663a'
  },
  payload: {
    key: {
      gateway: { host: '192.168.2.7', port: '8883' },
      key: '',
      userId: ''
    }
  }
}
sending payload {
  header: {
    method: 'SET',
    namespace: 'Appliance.Config.Wifi',
    messageId: '7445c822f426ba1669a69cd48381107c',
    timestamp: 1611115057,
    sign: 'a5a869b5f6681544c30dfbbb3f9e7f8e'
  },
  payload: { wifi: { ssid: 'VEVTVA==', password: 'MTIzNDU2Nzg5' } }
}
Got response: {
  header: {
    messageId: '7445c822f426ba1669a69cd48381107c',
    namespace: 'Appliance.Config.Wifi',
    method: 'SETACK',
    payloadVersion: 1,
    from: '/appliance/1712126291560325130734298f107d35/publish',
    timestamp: 290,
    timestampMs: 980,
    sign: '353be22eb626cbd30cdc2122a9797050'
  },
  payload: {}
}
$ 
~~~

Any suggestions on where I should look, or something else I should try?

Thanks so much for your efforts on this! This project is what actually first prompted me to try and gain local control over my Meross devices. Now if only I could make it work...

bytespider, 2021-02-02:
Is your router configured for DHCP?
Does your wifi SSID or password contain any non ASCII characters?

I would suggest if you really don't see the device on your network, that the issue lies there. As per your command log, the device understood what it was told and responded with the SETACK response.

I would try with node 12, and as a rule of thumb, single quote strings like wifi on the command line as double quotes will get interpreted.

midnightgrue, 2021-02-03:
> Is your router configured for DHCP?

Yes, definitely.

> Does your wifi SSID or password contain any non ASCII characters?

I actually set up the test network I referenced in the posted log (--wifi-ssid TEST --wifi-pass 123456789) to make sure this wasn't the problem, as my normal SSID does include an underscore. Same results, unfortunately.

> I would suggest if you really don't see the device on your network, that the issue lies there. As per your command log, the device understood what it was told and responded with the SETACK response.

I concur -- but I can't seem to figure out where. The official Meross app has no trouble joining the device to my normal SSID, so obviously there's something happening there that is able to deal with whatever problem exists on my network, something which doesn't happen when I use your utility. Do I need to have a local ntp server running or something? Some kind of timing or DNS issue? I mean, I'm seriously stumped. Everything looks like it should work. It just doesn't :( 

> I would try with node 12, and as a rule of thumb, single quote strings like wifi on the command line as double quotes will get interpreted.

That's what I thought on the quotes, but I figured it was worth a try (shrug) I will try with node 12 and see if anything changes. I also just liberated another MSS110 device from service, so I'll try with that too, to make sure it isn't just a wonky plug or something. I'll post results here (probably tomorrow before I get to it).

Thanks very much for the help!

bytespider, 2021-02-03:
> Some kind of timing or DNS issue? I mean, I'm seriously stumped. Everything looks like it should work. It just doesn't :(

I don't have either of these things that I'm aware of, though my Eero system might be doing something I'm not aware of. I do have my network connection to the outside world which I know other people have experienced issues with when the device can't connect out to an NTP server or such.

bytespider, 2021-02-09:
Hi @midnightgrue, the utility has been updated to include a few fixes and quirks that other users have discovered about these devices if you're still interested in getting yours to work, please try the latest build.

midnightgrue, 2021-02-14:
Thanks for the ping! I did make sure I was using node 12 and tried a different device and got the same results, which is why I never got around to posting again, sorry.

Unfortunately the update doesn't work for me either :( In fact, after updating to the new utility, the info command itself doesn't work quite right anymore. Here's the output (I used your clever --verbose option to see that, despite the weird error, the utility was actually managing to connect to the plug and get at least some information):

~~~
$ ./meross info --gateway 10.10.10.1
Getting info about device with IP 10.10.10.1
Error Unable to connect to device
$ ./meross info --verbose --gateway 10.10.10.1
Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '0fc2686d7f7966638666afca38bf98a4',
    timestamp: 1613283870,
    sign: '9197c850bd407dc4f433b208326c5676'
  },
  payload: {}
}
< 200 OK
< server: HTTPD
< date: Thu, 01 Jan 1970 00:02:47 GMT
< pragma: no-cache
< cache-control: no-cache
< content-type: application/json;charset=UTF-8
< connection: close
<
{
  header: {
    messageId: '0fc2686d7f7966638666afca38bf98a4',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/1712217231635225131434298f10c74a/publish',
    timestamp: 167,
    timestampMs: 390,
    sign: '90454e8b55c0acb3effa5b56286a9460'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss110',
          subType: 'us',
          version: '1.0.0',
          chipType: 'MT7688',
          uuid: '1712217231635225131434298f10c74a',
          macAddress: '34:29:8f:10:c7:4a'
        },
        firmware: {
          version: '1.1.27',
          compileTime: '2018-12-20 09:42:41',
          wifiMac: '01:00:00:00:18:dc',
          innerIp: '0.0.0.0',
          server: '',
          port: 0,
          secondServer: '',
          secondPort: 0,
          userId: 0
        },
        time: { timestamp: 167, timezone: '', timeRule: [] },
        online: { status: 2 }
      },
      control: { toggle: { onoff: 1, lmTime: 0 }, trigger: [], timer: [] }
    }
  }
}
Error Unable to connect to device
$ ./meross setup --verbose --gateway 10.10.10.1 --wifi-ssid '<redacted>' --wifi-pass '<redacted>'
Setting up device with IP 10.10.10.1
┌───────────────────────────────────────────────┬───────────────────────────────
│Encoded WIFI SSID                              │cml2YV9lY2hv                   
├───────────────────────────────────────────────┼───────────────────────────────
│Encoded WIFI password                          │QW1hejBuMXM1cDNjMUBs           
└───────────────────────────────────────────────┴───────────────────────────────
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    method: 'SET',
    namespace: 'Appliance.Config.Wifi',
    messageId: '5c99d19bf1d5a9d91e1420af4277bf80',
    timestamp: 1613283948,
    sign: 'd647e6d7142189dadc1f6eaa07120601'
  },
  payload: { wifi: { ssid: 'cml2YV9lY2hv', password: 'QW1hejBuMXM1cDNjMUBs' } }
}
< 200 OK
< server: HTTPD
< date: Thu, 01 Jan 1970 00:04:05 GMT
< pragma: no-cache
< cache-control: no-cache
< content-type: application/json;charset=UTF-8
< connection: close
<
{
  header: {
    messageId: '5c99d19bf1d5a9d91e1420af4277bf80',
    namespace: 'Appliance.Config.Wifi',
    method: 'SETACK',
    payloadVersion: 1,
    from: '/appliance/1712217231635225131434298f10c74a/publish',
    timestamp: 245,
    timestampMs: 940,
    sign: 'bb6feb429bc61dc57179e28061d92705'
  },
  payload: {}
}
Rebooting device
$ 
~~~

As before, the device reboots successfully after issuing the setup command, and then it blinks green for a long time, reboots again, and returns to blinking orange/green. No change...no visibility on my router...nothing. I do think it's likely I'm doing something wrong, but I'm really not sure what it could be (shrug)

I'll be away from home for a few weeks starting tomorrow, so I won't be able to try again for a while, but if you have any ideas or suggestions, please let me know and I'll definitely give them a try when I get back. 

Thanks again for your work and all your help!

bytespider, 2021-02-15:
Actually, this is super helpful. 

I realise now the lastest firmware for you seems to be a v1 firmware whilst my code has been strictly v2 up to this point. I had assumed that all devices have the same or similar firmware, which seems to not be the case.

On your setup command, I see that you didn't pass any `--mqtt` options.

kmitch95120, 2021-02-23:
I'm running into the same issue with a new MSS110 that appears to have version 6 firmware and a new Realtek chip.  I'll add the details to the issue I started to track the new firmware and chip.

bytespider, 2021-03-30:
@midnightgrue @kmitch95120
> Unfortunately the update doesn't work for me either :( In fact, after updating to the new utility, the info command itself doesn't work quite right anymore.

Sorry about that, it turns out there are some differences in the responses from various devices. I've removed that code as I don't think it's really needed. Can you try the latest version? The utility is now on NPM so you can just do `npx meross info` if you don't want to download and install manually.

bytespider, 2021-03-31:
Update: We've found a solution. Just need to implement it in code.

bytespider, 2021-04-16:
Command line options `--wifi-encryption`, `--wifi-cipher`, and `--wifi-channel` have been added. Please used these along with `meross info --include-wifi` if your device doesn't connect with just the SSID and Password.

closed 2021-04-16 by bytespider

midnightgrue, 2021-04-19:
Just wanted to report back and say the updated code works for me! With the additional options, I'm able to configure my old MSS110s to join my wi-fi and start publishing successfully to MQTT. Of course I now have other problems -- I'm using the meross-lan integration with the MQTT add-on in Home Assistant, and while the integration discovers the plugs just fine, the devices it creates are all showing as unavailable (sigh) -- but that's not a problem with your utility, I don't think. My seemingly endless quest will now be continued on another github issues page :) Thank you so much for all your help!