bytespider/Meross #17: MSH300 to Mosquitto on Raspberry PI connection fail

elitno, 2021-02-06:
Hi,
I followed the instructions to the letter but I can't in any way get an MSH300 to communicate with my Mosquitto Broker running on a raspberry pi4.

This is my mosquitto.conf

port 1883

listener 8883

require_certificate false
use_identity_as_username false
allow_anonymous true

cafile /etc/mosquitto/certs/ca.crt
certfile /etc/mosquitto/certs/server.crt
keyfile /etc/mosquitto/certs/server.key

But after a single try after pairing tentative I see in the log

06/02/2021 15:56:49 
1612623408: Socket error on client <unknown>, disconnecting.
06/02/2021 15:56:49(-0.1 seconds) 
1612623408: New connection from 192.168.1.141 on port 8883.

And my MSH300 start again to flashing orange-green

Thanx

p.s. nice work !!!

bytespider, 2021-02-06:
Thanks.

I suspect there is an issue with your SSL certificate How did you create it?.

elitno, 2021-02-06:
pi@raspberrypi:/etc/mosquitto/certs $ openssl genrsa -des3 -out ca.key 2048
Generating RSA private key, 2048 bit long modulus (2 primes)
......+++++
......................................+++++
e is 65537 (0x010001)
Enter pass phrase for ca.key:
Verifying - Enter pass phrase for ca.key:
pi@raspberrypi:/etc/mosquitto/certs $ openssl req -new -x509 -days 1826 -key ca.key -out ca.crt
Enter pass phrase for ca.key:
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:IT
State or Province Name (full name) [Some-State]:Bologna
Locality Name (eg, city) []:Bologna
Organization Name (eg, company) [Internet Widgits Pty Ltd]:
Organizational Unit Name (eg, section) []:
Common Name (e.g. server FQDN or YOUR name) []:192.168.1.224
Email Address []:
pi@raspberrypi:/etc/mosquitto/certs $ openssl genrsa -out server.key 2048
Generating RSA private key, 2048 bit long modulus (2 primes)
.....+++++
........................................................+++++
e is 65537 (0x010001)
pi@raspberrypi:/etc/mosquitto/certs $ openssl req -new -out server.csr -key server.key
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:IT
State or Province Name (full name) [Some-State]:Bologna
Locality Name (eg, city) []:Bologna
Organization Name (eg, company) [Internet Widgits Pty Ltd]:
Organizational Unit Name (eg, section) []:
Common Name (e.g. server FQDN or YOUR name) []:192.168.1.224
Email Address []:

Please enter the following 'extra' attributes
to be sent with your certificate request
A challenge password []:
An optional company name []:
pi@raspberrypi:/etc/mosquitto/certs $ openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 360
Signature ok
subject=C = IT, ST = Bologna, L = Bologna, O = Internet Widgits Pty Ltd, CN = 192.168.1.224
Getting CA Private Key
Enter pass phrase for ca.key:
pi@raspberrypi:/etc/mosquitto/certs $ sudo systemctl restart mosquitto

Thanx again !

bytespider, 2021-02-07:
I think the issues is the Common Name on the CA cert and server cert is the same. Try using something else for the CA Common Name, your name for instance.

elitno, 2021-02-07:
I changed the name, filled in all the certificate fields but I get the same error.
The strange thing is that from mosquitto_sub I can connect perfectly:

mosquitto_sub -h 192.168.1.224 -p 8883 -t "/appliance/XXXX/publish" --cafile /etc/mosquitto/certs/ca.crt

1612695903: New client connected from 192.168.1.224 as mosq-xxx (p2, c1, k60). (OK)

bytespider, 2021-02-07:
Yeah these Meross devices can be fussy.

Do you have the latest firmware on it?

elitno, 2021-02-07:
Yeah last firmware version 4.1.26.
But I think that the problem is in my mosquitto configuration because It is possible connect to "test.mosquitto.org" successfully but not to mine.

bytespider, 2021-02-07:
I've included some certificates and example configs in the repo, they may help.
[https://github.com/bytespider/Meross/tree/develop](https://github.com/bytespider/Meross/tree/develop)

elitno, 2021-02-08:
I finally found the problem: it depends on the use of a password file.
Is not sufficient to specify "allow_anonymous true"  for authorizing both anonymous (empty username) and unknown users (any username), the string password_file must be deleted or commented out.  This is because in mosquitto it is not possible to specify usernames containing a colon (:) so mac-addres-username can't be used.

bytespider, 2021-02-09:
> This is because in mosquitto it is not possible to specify usernames containing a colon (:) so mac-addres-username can't be used.

Yes and No. 😀

I have discovered, in the past week, it's possible to use mosquitto-auth-plug [my fork with fixes](https://github.com/bytespider/mosquitto-auth-plug) and a MySQL or HTTP back end to authenticate the user. But yes if you don't care about authentication you can simply set allow_anonymous true.

> Is not sufficient to specify "allow_anonymous true" for authorizing both anonymous (empty username) and unknown users (any username)

With no password file, which my suggested config did not set and `allow_anonymous true`, for me and a few others as well, all devices were allowed to connect. That was your config as detailed above.

Are things working for you now? If you are running your own Mosquitto not backed by Home Assistant and you want authentication of your devices you can [try the new config](https://github.com/bytespider/Meross/wiki/MQTT#authenticated-users)

closed 2021-04-16 by bytespider

bytespider, 2021-04-16:
Using Meross devices with [Home Assistant Mosquitto](https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO))