bytespider/Meross #18: Meross over MQTT with Home Assistant

bytespider, 2021-02-06:
The Home Assistant version of Mosquitto ships with mosquitto-auth-plug. I've had some success with connecting Meross smart plugs to the HA Mosquitto instance, however, there are a few issues:
- The Meross devices (may not be all) send a ClientId > 23 characters which contravene the MQTT 3.1 spec.
- Home Assistant does allow you to inject custom options, however, new listeners cannot be seen by the network
- Home Assistant doesn't allow you to inject custom options into the config "Per Listener"

krahabb, 2021-02-23:
Hello,
I've developed this [integration](https://github.com/krahabb/meross_lan) by implementing a sort of mqtt discovery for meross protocol into the code.
After the initial issues with re-pairing the plugs I've finally managed to let it work with this configuration:
MQTT broker: mosquitto
MQTT TLS: according to your guide and any other to setup TLS on mosquitto with private certificate
MQTT authentication: none (this is important even if not the best use-case: this way I do not have to deal with Meross user_id and passwords at the mqtt level)

Important: when using your meross config utility I had to configure both of the mqtt server keys on the plug to the same mqtthost else the plug would fail unrecoverably and fall back to pairing mode in case the broker was down. Also I did not setup any user/key on the plug itself so all of the signing code in the protocol implementation is based on a null key at the moment.
Let me know if I could help or if my integration could be useful. I can work a bit on that even if the main problem is that I only managed to buy MSS310 plugs so testing is *almost impossible* for any other device

bytespider, 2021-02-24:
Thanks for sharing your experience. The current version of this tool already sets both servers to the same if only one is provided.

closed 2021-02-24 by bytespider

bytespider, 2021-04-16:
See [https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO)](https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO)) for a rough guide to using Meross plugs with Home Assistant

nickelnine, 2021-04-18:
Hi @bytespider , do you mind explaining in more detail the 'Device credentials' for using with Hassio? When running `meross-setup` as you say, I only see the `--wifi` parameters and not `--user`.

bytespider, 2021-04-19:
`--user` and `--key` sets the Meross Cloud user id and key and makes up part of the password generated when communicating with the MQTT broker. You can safely ignore them as they offer no additional security on local MQTT