bytespider/Meross #20: MSS110: New Chip and Firmware

kmitch95120, 2021-02-20:
I just picked up a 4 pack of MSS110 US plugs and thought I'd share what the info command shows. Hardware version 6.0.0 and firmware 6.1.4 with a Realtek RTL8710cf. Just now working on getting RLS working in mosquitto so I'll let you know if there's any issues.  Let me know if you want me to collect and specific information. 

Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: 'a3548a78936b95d462f51dbef5198e16',
    timestamp: 1613841348,
    sign: '3b7da8133320c9d6a1f352995272991b'
  },
  payload: {}
}
< 200 OK
< content-type: application/json
< connection: close
<
{
  header: {
    messageId: 'a3548a78936b95d462f51dbef5198e16',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/2008192442191190825548e1e92cb133/publish',
    timestamp: 250848,
    timestampMs: 165,
    sign: 'c4a60115316d723da29e7f12d2b199cb'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss110',
          subType: 'us',
          version: '6.0.0',
          chipType: 'rtl8710cf',
          uuid: '2008192442191190825548e1e92cb133',
          macAddress: '48:e1:e9:2c:b1:33'
        },
        firmware: {
          version: '6.1.14',
          compileTime: '2020/07/27-17:24:33',
          wifiMac: '00:00:00:00:00:00',
          innerIp: '10.10.10.1',
          server: '',
          port: 0,
          userId: 0
        },
        time: { timestamp: 250848, timezone: '', timeRule: [] },
        online: { status: 2 }
      },
      digest: {
        togglex: [ { channel: 0, onoff: 1, lmTime: 2 } ],
        triggerx: [],
        timerx: []
      }
    }
  }
}

bytespider, 2021-02-20:
Thanks for the update. Donut would seem the firmware version is closely tied to the hardware version despite having little differences 🤦

kmitch95120, 2021-02-23:
I had a bit more time to play with the new device last night but I can't seem to get it to connect to WiFi.  The first thing I noticed is that the LED behavior is not the same as what's documented in the installation guide.  The docs talk about amber solid, amber/green flashing, green solid, and red solid, but the devices I have (4x of them) only have a blue LED and it's either On or Off. 

[image]

I should also note that these particular MSS110's DO NOT support HomeKit.  Meross now has the MSS110 and the MSS110HK for HomeKit, which is more expensive.  I'd love to get these working, Let me know what info I can collect or what tasks I need to do.

bytespider, 2021-02-24:
I'm curious, is there a new version of the firmware for you? 6.1.16?

kmitch95120, 2021-02-24:
I created a Meross account and added one of the plugs to it and the app says 6.1.14 is the latest.

bytespider, 2021-04-16:
Hi @kmitch95120 could you try the latest version if this tool?

bytespider, 2021-04-16:
Command line options --wifi-encryption, --wifi-cipher, and --wifi-channel have been added. Please used these along with meross info --include-wifi if your device doesn't connect with just the SSID and Password.

closed 2021-04-16 by bytespider