bytespider/Meross #22: mss310 us MT7688 (hardware:1.0.0 firmware:1.1.18)  not joining Wi-Fi after reboot

mthlvt, 2021-03-30:
Hi @bytespider ! thanks for the new release which is promising!!!

I'm also stuck with 2018 plugs, the behaviour is similar to what is described in #15 i.e.:
- info and setup seem to work 
- plug reboots but fails to join the wifi network (blinks green then reboot back into configuration mode)

Note:
- No logs showing the plug on the wi-fi router side, DCHP is enabled
- Plugs work with Meross app and I've tried various combination of plugs and even different physical Wifi APs.
- using npm version 12, and latest Meross commit of today 

Posting the logs in case this can help identifying the issue. (x's = sanitized data)

<pre><font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>~/Meross/bin</b></font>$ ./meross-info --verbose  --gateway 10.10.10.1 
Getting info about device with IP 10.10.10.1
&gt; POST /config
&gt; Host: 10.10.10.1
&gt; Accept: application/json, text/plain, */*
&gt; Content-Type: application/json
&gt;
{
  header: {
    method: &apos;GET&apos;,
    namespace: &apos;Appliance.System.All&apos;,
    messageId: &apos;a28817986ef4a6ba2b0baa16c4d54221&apos;,
    timestamp: 1617135599,
    sign: &apos;47de1547d7a9af6779d014dc0520dbd1&apos;
  },
  payload: {}
}

&lt; 200 OK
&lt; server: HTTPD
&lt; date: Thu, 01 Jan 1970 00:06:14 GMT
&lt; pragma: no-cache
&lt; cache-control: no-cache
&lt; content-type: application/json;charset=UTF-8
&lt; connection: close
&lt;
{
  header: {
    messageId: &apos;a28817986ef4a6ba2b0baa16c4d54221&apos;,
    namespace: &apos;Appliance.System.All&apos;,
    method: &apos;GETACK&apos;,
    payloadVersion: 1,
    from: &apos;/appliance/1801164732796825131634298f1xxxxx/publish&apos;,
    timestamp: 374,
    timestampMs: 440,
    sign: &apos;5ab121e75515dae92db1698ddd764b95&apos;
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: &apos;mss310&apos;,
          subType: &apos;us&apos;,
          version: &apos;1.0.0&apos;,
          chipType: &apos;MT7688&apos;,
          uuid: &apos;1801164732796825131634298f1xxxxx&apos;,
          macAddress: &apos;34:29:8f:11:15:5c&apos;
        },
        firmware: {
          version: &apos;1.1.18&apos;,
          compileTime: &apos;2018-12-03 11:15:36&apos;,
          wifiMac: &apos;01:00:00:00:d0:88&apos;,
          innerIp: &apos;0.0.0.0&apos;,
          server: &apos;&apos;,
          port: 0,
          secondServer: &apos;&apos;,
          secondPort: 0,
          userId: 0
        },
        time: { timestamp: 374, timezone: &apos;&apos;, timeRule: [] },
        online: { status: 2 }
      },
      control: { toggle: { onoff: 1, lmTime: 0 }, trigger: [], timer: [] }
    }
  }
}

┌─────────────────┬───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">Device           </font>│mss310 us MT7688 (hardware:1.0.0 firmware:1.1.18)                          
├─────────────────┼───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">UUID             </font>│1801164732796825131634298f1xxxxx                                           
├─────────────────┼───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">Mac address      </font>│34:29:8f:1x:xx:xx                                                          
├─────────────────┼───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">WIFI             </font>│0.0.0.0 (01:00:00:00:d0:88)                                                
├─────────────────┼───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">Status           </font>│<span style="background-color:#2E3436"><font color="#729FCF">Configuration</font></span>                                                              
├─────────────────┼───────────────────────────────────────────────────────────────────────────
│<font color="#C4A000">Credentials      </font>│User: <span style="background-color:#2E3436"><font color="#34E2E2">0</font></span>                                                                    
│<font color="#C4A000">                 </font>│Password: <span style="background-color:#2E3436"><font color="#34E2E2">0_93249be6091bab57e8ec2370258bafb1</font></span>                               
└─────────────────┴───────────────────────────────────────────────────────────────────────────
<font color="#8AE234"><b>mint@mint</b></font>:<font color="#729FCF"><b>~/Meross/bin</b></font>$ ./meross setup --verbose  --gateway 10.10.10.1 --wifi-ssid XXXX --wifi-pass xxxxxxxxxxxxxxxxxxxxx --mqtt mqtts://192.168.x.x
Setting up device with IP 10.10.10.1
┌────────────────────────────────────────────────────┬────────────────────────────────────────
│<font color="#C4A000">Primary MQTT broker                                 </font>│192.168.x.x:8883                        
├────────────────────────────────────────────────────┼────────────────────────────────────────
│<font color="#C4A000">Secondary MQTT broker                               </font>│192.168.x.x:8883                        
└────────────────────────────────────────────────────┴────────────────────────────────────────
&gt; POST /config
&gt; Host: 10.10.10.1
&gt; Accept: application/json, text/plain, */*
&gt; Content-Type: application/json
&gt;
{
  header: {
    method: &apos;SET&apos;,
    namespace: &apos;Appliance.Config.Key&apos;,
    messageId: &apos;c54df717ed6e05a09ec5d5e551774a20&apos;,
    timestamp: 1617135673,
    sign: &apos;f0e130470adea4d23c9e4b4d8494e553&apos;
  },
  payload: {
    key: {
      gateway: {
        host: &apos;192.168.x.x&apos;,
        port: &apos;8883&apos;,
        secondHost: &apos;192.168.x.x&apos;,
        secondPort: &apos;8883&apos;
      },
      key: &apos;&apos;,
      userId: &apos;&apos;
    }
  }
}

&lt; 200 OK
&lt; server: HTTPD
&lt; date: Thu, 01 Jan 1970 00:07:28 GMT
&lt; pragma: no-cache
&lt; cache-control: no-cache
&lt; content-type: application/json;charset=UTF-8
&lt; connection: close
&lt;
{
  header: {
    messageId: &apos;c54df717ed6e05a09ec5d5e551774a20&apos;,
    namespace: &apos;Appliance.Config.Key&apos;,
    method: &apos;SETACK&apos;,
    payloadVersion: 1,
    from: &apos;/appliance/1801164732796825131634298f1xxxxx/publish&apos;,
    timestamp: 448,
    timestampMs: 610,
    sign: &apos;ca9f6544d0aa70ca87586e33f592a75f&apos;
  },
  payload: {}
}

┌───────────────────────────────────────┬─────────────────────────────────────────────────────
│<font color="#C4A000">Encoded WIFI SSID                      </font>│xxxxxxxxxxxxxxxx=                                     
├───────────────────────────────────────┼─────────────────────────────────────────────────────
│<font color="#C4A000">Encoded WIFI password                  </font>│xxxxxxxxxxxxxxxxxxxxxxxxxxxxx=                         
└───────────────────────────────────────┴─────────────────────────────────────────────────────
&gt; POST /config
&gt; Host: 10.10.10.1
&gt; Accept: application/json, text/plain, */*
&gt; Content-Type: application/json
&gt;
{
  header: {
    method: &apos;SET&apos;,
    namespace: &apos;Appliance.Config.Wifi&apos;,
    messageId: &apos;77a3a9e286dc27bc21e2cc67a1e8e719&apos;,
    timestamp: 1617135674,
    sign: &apos;8afe91cb6c2d71b6e3de19c9eddaa183&apos;
  },
  payload: {
    wifi: {
      ssid: &apos;xxxxxxxxxxxxxxxx=&apos;,
      password: &apos;xxxxxxxxxxxxxxxxxxxxxxxxxxxxx=&apos;
    }
  }
}

&lt; 200 OK
&lt; server: HTTPD
&lt; date: Thu, 01 Jan 1970 00:07:28 GMT
&lt; pragma: no-cache
&lt; cache-control: no-cache
&lt; content-type: application/json;charset=UTF-8
&lt; connection: close
&lt;
{
  header: {
    messageId: &apos;77a3a9e286dc27bc21e2cc67a1e8e719&apos;,
    namespace: &apos;Appliance.Config.Wifi&apos;,
    method: &apos;SETACK&apos;,
    payloadVersion: 1,
    from: &apos;/appliance/1801164732796825131634298f1xxxxx/publish&apos;,
    timestamp: 448,
    timestampMs: 730,
    sign: &apos;45e008d3664c08ea59c06fb8f519ec97&apos;
  },
  payload: {}
}

Rebooting device
</pre>

cheers!

bytespider, 2021-03-31:
If you have access to curl, can you try put a device into pairing and connect to it's AP then try this command?
```
curl --request POST \
   --url http://10.10.10.1/config \
   --header 'Content-Type: application/json' \
   --data '{
     "header": {
         "method": "GET",
         "namespace": "Appliance.Config.WifiList",
         "messageId": "cce78e0c91e9c9446323775340534878",
         "timestamp": 1617176364,
         "sign": "3bd15b3447a3347a9107c386bf5ee516"
     },
     "payload": {}
 }'
```

mthlvt, 2021-03-31:
Thanks!

The wifi list contains multiple networks, among them I confirm the same SSID I pass to setup:

```json
            {
                "ssid": "xxxxxxxxxxxxxxxx=",
                "bssid": "38-10-d5-5c-cd-de",
                "signal": 99,
                "channel": 8,
                "encryption": 6,
                "cipher": 3
            },
```
perhaps the encryption/cipher needs to be adjusted?

bytespider, 2021-03-31:
> perhaps the encryption/cipher needs to be adjusted?
This was the avenue of thinking I was heading down. Will check what my WIFI setup uses and compare.

mthlvt, 2021-03-31:
yes, I didn't modify the rest of the json above, it was hyphenated

if you let me know what encryption cipher (e.g. WPA2+PSK) you used successfully on your network I can setup a virtual ssid with same settings to test.

bytespider, 2021-03-31:
Do you want to try this?

```
curl --request POST \
  --url http://10.10.10.1/config \
  --header 'Content-Type: application/json' \
  --data '{
  "header": {
    "method": "SET",
    "namespace": "Appliance.Config.Wifi",
    "messageId": "0be2318bac0a9237164cade9063fe803",
    "timestamp": 1617183880,
    "sign": "cb9178873b940b65e291b5e3fef64c32"
  },
  "payload": {
    "wifi": {
      "ssid": "<your base64 encoded ssid>",
      "password": "<your base64 encoded password>",
      "bssid": "<your bssid each octet separated by colon>",
      "encryption": 6,
      "cipher": 3,
      "channel": 8
    }
  }
}'
```

mthlvt, 2021-03-31:
That worked, now it is joining the wi-fi and gets an IP. Thanks !

Likely I have to tune MQTT config now as it keeps blinking green, will redo the certificate process from scratch to be sure.

bytespider, 2021-03-31:
That's excellent. I'll get working on an implementation in the tool.

bytespider, 2021-03-31:
> Now it blinks green, what's the next command to push MQTT server settings please?

Sorry I missed this before.

```
curl --request POST \
  --url http://10.10.10.1/config \
  --header 'Content-Type: application/json' \
  --data '{
  "header": {
    "method": "SET",
    "namespace": "Appliance.Config.Key",
    "messageId": "efed8e6db4b629fbeeaa766b7c8e7d1d",
    "timestamp": 1617186747,
    "sign": "2bdfbbed9795faa187ffeb493e606c66"
  },
  "payload": {
    "key": {
      "gateway": {
        "host": "0.0.0.0",
        "port": "8443",
        "secondHost": "0.0.0.0",
        "secondPort": "8443"
      },
      "key": "",
      "userId": ""
    }
  }
}'
```

Replace the `0.0.0.0` with your MQTT IP and the `8443` with your MQTT port, just incase that wasn't obvious 😀

mthlvt, 2021-03-31:
it works, thanks a lot.

bytespider, 2021-03-31:
@mthlvt are you able to try something for me?
Could you, if you have a spare device, try running the WIFI setup command again but set `encryption` and `cipher` to `0`?

mthlvt, 2021-03-31:
sure I'll try tomorrow !

mthlvt, 2021-04-01:
I tried encryption and cipher : 0 but this doesn't work for me, the LED becomes fixed orange after reboot and then it reverts to config mode after a while.

bytespider, 2021-04-01:
Okay thanks for doing that. I was hoping it was just a case of expecting some value an not caring. Its going  a pretty horrible set of cli flags when there 5 cipher types and 9 encryption types. Will have to provide that WIFI list command so users know what values to use.

mthlvt, 2021-04-02:
Thanks, for now I'm just hardcoding cipher/encryption values I need in api.js and it scales ok.

closed 2021-04-16 by bytespider

bytespider, 2021-04-16:
Command line options `--wifi-encryption`, `--wifi-cipher`, and `--wifi-channel` have been added. Please use these along with meross info `--include-wifi` for devices that don't connect just from the SSID and Password

lcavalli, 2021-06-03:
Hi, I have a smart plug mss310 (hw: 2.0.0, fw: 2.1.6) and I'm still not able to configure my device.
I also tried with
`./meross-info --include-wifi`
and then
`./meross-setup --wifi-channel 6 --wifi-encryption 7 --wifi-cipher 6 --gateway 10.10.10.1 --wifi-ssid XXXXX --wifi-pass ******* --mqtt mqtts://192.168.1.xxx --mqtt mqtts://192.168.1.xxx`
I tried also  `--wifi-encryption 0 --wifi-cipher 0`. No way to connect to my wifi. Each time the device resets and led blinks amber/green.

bytespider, 2021-06-03:
@lcavalli can your tell me more about your broker?

lcavalli, 2021-06-03:
No, standalone (native, no docker) on the same raspberry pi running HA in a docker container. Configured with TSL (certificates generated following your wiki) without authentication listening on port 8883. Port 1883 still active for plain mqtt used by zigbee2mqtt (running on the very same rpi in a docker container). Thank you!

lcavalli, 2021-06-03:
Bingo! You were right!
```
1622752444: New connection from 192.168.1.150:63109 on port 8883.
1622752444: Client <unknown> disconnected, not authorised.
```
The smart plug resets because the broker reject the connection.
Thank you. I will look at mosquitto configuration.

lcavalli, 2021-06-03:
Solved. Since I'm using multiple configurations in /etc/mosquitto/conf.d/ (one for zigbee2mqtt and one for meross) I had to add `per_listener_settings true` before `include_dir /etc/mosquitto/conf.d`. Thank you again for your help and for your software!