bytespider/Meross #25: Smart plug not connecting to broker

enomam, 2021-05-17:
Hi,
I just got a bunch of MSS310 smart plugs, with firmware 2.1.16, h/w version 2.0.0.
I've already got an existing mosquitto server (v1.4.10).

When I run the setup as desribed in the wiki, the device does connect to my wifi (I can see it on my routers dashboard), but it doesn't connect to the broker  - I see lots of the following messages, and the light on the device remains green.
```
1621232550: New connection from 10.0.10.147 on port 1883.
1621232550: Socket error on client <unknown>, disconnecting.
```

The device works fine when connected via the Meross servers, so not sure what else to try here. My mosquitto server is not configured with certificates - if that helps, and there is no username auth.

bytespider, 2021-05-17:
When I test my devices they require the use of TLS. Try setting up the certificate in your broker.

enomam, 2021-05-17:
Gotcha - yes.
That seems to work, but has wreaked havok with my zigbee2mqtt setup.

I'll take a pause and revisit

bytespider, 2021-05-17:
You can leave port 1883 as non-TLS and use port 8883 for TLS, if you aren't already.

enomam, 2021-05-17:
OMG! That worked - I gotta learn more of this stuff!

closed 2021-05-17 by enomam