bytespider/Meross #27: Turning off the leds of the plug

Skyleiger, 2021-05-19:
Hi, 
I found out how to turn off the leds of my mss425f (Meross Smart Plug Strip).
You can change the "DNDMode" (LED Mode) via MQTT by sending a request like this:
```
{
   "header":{
      "from":"/appliance/<device-uuid>/publish",
      "messageId":"<message-id>",
      "method":"SET",
      "namespace":"Appliance.System.DNDMode",
      "payloadVersion":1,
      "sign":"<sign>",
      "timestamp":1612261079,
      "triggerSrc":"Android"
   },
   "payload":{
      "DNDMode":{
         "mode":1
      }
   }
}
```

Mode 1 = LEDs off
Mode 0 = LEDs on

Well, it would be cool if something like that could be implemented in the script.

Sincerely,
Dominic

P.s. you can get an overview of all configuration settings by sending a request like this to your device:
```
{
   "header":{
      "from":"/appliance/<device-uuid>/publish",
      "messageId":"<message-id>",
      "method":"GET",
      "namespace":"Appliance.System.Ability",
      "payloadVersion":1,
      "sign":"<sign>",
      "timestamp":1612261079,
      "triggerSrc":"Android"
   },
   "payload":{
   }
}
```

bytespider, 2021-05-19:
Thanks for contrubiting.

I considered doing something in the tool, however, since most people are using MQTT via Home Assistant I felt it a little redundant.

Tools like https://github.com/krahabb/meross_lan would be better suited to handle the control logic.

Skyleiger, 2021-05-19:
Ah, excuse me.
I was not aware of the other project until now.
It might be useful to mention the other project in the documentation or in the readme.

Thank you, have a nice evening!

closed 2021-05-19 by Skyleiger