bytespider/Meross #30: Issues with connecting MSS310

funkenholy, 2021-06-02:
I'm using a a MSS310 `mss210 un mt7682 (hardware:2.0.0 firemwa:2.1.16)`
My HomeAssistant is operating on a Raspberry pi 4 b, along with the MQTT server plugin.

The config for this is:
```yaml
logins: []
customize:
  active: false
  folder: mosquitto
certfile: server.crt
keyfile: server.key
require_certificate: false
allow_anonymous: true
cafile: ca.crt
```
I've added a new user as per your instructions: https://github.com/bytespider/Meross/wiki/Home-Assistant-(HASSIO)

When this starts up it indicates that it can find the SSL certs, so I believe I've put them in the correct location on the pi. I feel my issue is either the certificates have not been generated correctly or perhaps something else is misconfigured.

This is the output of generating the Self Signed Cert:
```
william@fuel-ubuntu:~/ssl$ openssl genrsa -des3 -out ca.key 2048
Generating RSA private key, 2048 bit long modulus (2 primes)
..+++++
.............................................................................................................................................................................................................................................+++++
e is 65537 (0x010001)
Enter pass phrase for ca.key:
Verifying - Enter pass phrase for ca.key:
william@fuel-ubuntu:~/ssl$ openssl req -new -x509 -days 1826 -key ca.key -out ca.crt
Enter pass phrase for ca.key:
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:
State or Province Name (full name) [Some-State]:Western Australia
Locality Name (eg, city) []:Perth
Organization Name (eg, company) [Internet Widgits Pty Ltd]:Home
Organizational Unit Name (eg, section) []:
Common Name (e.g. server FQDN or YOUR name) []:ha
Email Address []:
william@fuel-ubuntu:~/ssl$ openssl genrsa -out server.key 2048
Generating RSA private key, 2048 bit long modulus (2 primes)
............................................................................................+++++
..............+++++
e is 65537 (0x010001)
william@fuel-ubuntu:~/ssl$ openssl req -new -out server.csr -key server.key
You are about to be asked to enter information that will be incorporated
into your certificate request.
What you are about to enter is what is called a Distinguished Name or a DN.
There are quite a few fields but you can leave some blank
For some fields there will be a default value,
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:
State or Province Name (full name) [Some-State]:WA
Locality Name (eg, city) []:Perth
Organization Name (eg, company) [Internet Widgits Pty Ltd]:Server-cert
Organizational Unit Name (eg, section) []:test
Common Name (e.g. server FQDN or YOUR name) []:192.168.2.103
Email Address []:

Please enter the following 'extra' attributes
to be sent with your certificate request
A challenge password []:
An optional company name []:
william@fuel-ubuntu:~/ssl$ openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 360
Signature ok
subject=C = AU, ST = WA, L = Perth, O = Server-cert, OU = test, CN = 192.168.2.103
Getting CA Private Key
Enter pass phrase for ca.key:
william@fuel-ubuntu:~/ssl$ scp ca.crt server.crt server.key root@192.168.2.103:/ssl/
ca.crt                                                                                                                          100% 1289     1.1MB/s   00:00
server.crt                                                                                                                      100% 1192     1.3MB/s   00:00
server.key                                                                                                                      100% 1675     2.1MB/s   00:00
william@fuel-ubuntu:~/ssl$
```

From the output you can see the HA is running on 192.168.2.103, which is what I've used for the CN of the server cert and I've explicitly used a different CN for the CA, as I've seen you've suggested this as a potential issue.
I've also not provided any value for the challenge password.

This is the setup command I'm sending to the Smart Socket:
```
./bin/meross setup --wifi-ssid 'Tribe Of Bruce' --wifi-pass '<some password>' --mqtt mqtt://192.168.2.103
```
This seems to connect to the wifi correctly and tries to establish a connection to the MQTT broker, but this is the output:
```
1622612400: New connection from 192.168.2.48 on port 1883.
1622612400: Client <unknown> disconnected due to protocol error.
1622612408: New connection from 192.168.2.48 on port 1883.
1622612408: Client <unknown> disconnected due to protocol error.
1622612427: New connection from 192.168.2.48 on port 1883.
1622612427: Client <unknown> disconnected due to protocol error.
1622612444: New connection from 192.168.2.48 on port 1883.
1622612444: Client <unknown> disconnected due to protocol error.
1622612508: New connection from 192.168.2.48 on port 1883.
1622612508: Client <unknown> disconnected due to protocol error.
1622612610: New connection from 192.168.2.48 on port 1883.
1622612610: Client <unknown> disconnected due to protocol error.
1622612798: New connection from 192.168.2.48 on port 8883.
401: Unauthorized1622612799: Socket error on client <unknown>, disconnecting.
1622612829: New connection from 192.168.2.48 on port 8883.
1622612829: Socket error on client <unknown>, disconnecting.
1622612876: New connection from 192.168.2.48 on port 1883.
1622612876: Client <unknown> disconnected due to protocol error.
1622612894: New connection from 192.168.2.48 on port 1883.
1622612894: Client <unknown> disconnected due to protocol error.
1622612903: New connection from 192.168.2.48 on port 1883.
1622612903: Client <unknown> disconnected due to protocol error.
1622612923: New connection from 192.168.2.48 on port 1883.
1622612923: Client <unknown> disconnected due to protocol error.
1622612978: New connection from 192.168.2.48 on port 1883.
1622612978: Client <unknown> disconnected due to protocol error.
1622613045: New connection from 192.168.2.48 on port 1883.
1622613045: Client <unknown> disconnected due to protocol error.
1622613188: New connection from 192.168.2.48 on port 1883.
1622613188: Client <unknown> disconnected due to protocol error.
1622613202: New connection from 192.168.2.48 on port 1883.
1622613202: Client <unknown> disconnected due to protocol error.
1622613214: New connection from 192.168.2.48 on port 1883.
```

If I try using `mqtts://192.168.2.103` as the `--mqtt` value, the device simply goes back into configuration mode.

Any help with this would be greatly appreciated, really love the work you've done with your tooling.

bytespider, 2021-06-02:
What version of the Mosquitto add-on are you using?

funkenholy, 2021-06-02:
```1622615578: mosquitto version 1.6.12 starting```

bytespider, 2021-06-02:
The issue is related to the latest version of the add-on. Roll back to 5.1 if you can
https://github.com/home-assistant/addons/issues/2020

And take a look here if you can't roll back https://github.com/bytespider/Meross/issues/28#issuecomment-851590907

funkenholy, 2021-06-02:
I wasn't sure how to roll back but adding you repo was easy enough.

I tried to just add the 5.1.1 version of the MQTT Add-on, but I think removing the 6.0.1 did something funky with the Mosquitto broker integration and the related user account it creates.

To get it all working I ended up removing the MQTT Add-on, the Mosquitto broker integration and the Meross LAN integration, then add them back in the same order with a few restarts in between.

After changing to MQTT 5.1.1 I was able to push the `--mqtt` setting with a mqtts:// address and not have the MSS310 reset back into pairing/configuration mode straight afterward.

Thank you for your help and the great software

closed 2021-06-02 by funkenholy

bytespider, 2021-06-02:
You're welcome. Make sure you update to the official broker add-on when they have resolved the issues.

balbulator, 2021-06-05:
updated to core-2021.6.2 
The problem appeared again

ghiblix35, 2021-06-11:
Thanks finally I was able to get my inverter to work.