bytespider/Meross #33: mss210 us rtl8710cf (hardware:6.0.0 firmware:6.1.12) unable to connect

platofan23, 2021-08-18:
mss210 us rtl8710cf (hardware:6.0.0 firmware:6.1.12) not working with mosquito. The plug goes tries to connect to the internet but fails. I hear to clicks. First when the led changes flashing green and later going back into pairing mode. 

my mosquitto conf is;
`port 1883
listener 8883
#use_username_as_clientid true
allow_anonymous true
capath /etc/ssl/certs

certfile /etc/mosquitto/certs/mqtt_server.pem
keyfile /etc/mosquitto/certs/mqtt_server.pem

tls_version tlsv1.1
`
My command:
`./meross setup --gateway 10.10.10.1 --wifi-ssid XXX --wifi-pass XXX  --mqtt 192.168.2.201 --key XXX --mqtt 192.168.2.206

Setting up device with IP 10.10.10.1
┌────────────────────────────────────────────────┬────────────────────────────────────────────┐
│Primary MQTT broker                             │192.168.2.201:1883                          │
├────────────────────────────────────────────────┼────────────────────────────────────────────┤
│Failover MQTT broker                            │192.168.2.206:1883                          │
└────────────────────────────────────────────────┴────────────────────────────────────────────┘
┌────────────────────┬────────────────────────────────────────────────────────────────────────┐
│Device              │mss210 us rtl8710cf (hardware:6.0.0 firmware:6.1.12)                    │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│UUID                │2012174007893251804648e1e9425940                                        │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Mac address         │48:e1:e9:42:59:40                                                       │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│IP address          │10.10.10.1                                                              │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Current MQTT broker │192.168.2.201:1883                                                      │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Credentials         │User: 48:e1:e9:42:59:40                                                 │
│                    │Password: 0_952716e80f5c54ff4493465156d32aa0                            │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│MQTT topics         │Publishes to: /appliance/2012174007893251804648e1e9425940/publish       │
│                    │Subscribes to: /appliance/2012174007893251804648e1e9425940/subscribe    │
└────────────────────┴────────────────────────────────────────────────────────────────────────┘
`
Form mosquitto when trying to sub the plug (sudo mosquitto_sub -d -t "/#" --cafile /etc/mosquitto/certs/mqtt_server.pem):
Client mosqsub|12810-PlatoPi sending PINGREQ
Client mosqsub|12810-PlatoPi received PINGRESP

This is from the log: (sudo tail -f /var/log/mosquitto/mosquitto.log)
New client connected from 192.168.2.50 as fmware:20040808158569251h3748e1e9197500_uBD0TymGUdnTTzoZ (c1, k30, u'48:e1:e9:19:75:00').
1629319822: Socket error on client mosqsub|12433-PlatoPi, disconnecting.

Thank you for helping =)

bytespider, 2021-08-18:
Hi @platofan23 can you tell me a bit about your broker? Is it running via Home Assistant, and if so what version of the HA broker is it? V6.x is bad and you'll need to find a way to downgrade to 5.x.

platofan23, 2021-08-18:
> 
> 
> Hi @platofan23 can you tell me a bit about your broker? Is it running via Home Assistant, and if so what version of the HA broker is it? V6.x is bad and you'll need to find a way to downgrade to 5.x.

Iam running the broker just over ssh. Later i want to integrate this in homeassistant.

is there a site to download the 5.x firmware from meross for the plug?

bytespider, 2021-08-18:
If you are on using Homeassistant there are a couple of options.
Harder: https://github.com/krahabb/meross_lan/discussions/63#discussion-3525179

Simple: https://github.com/bytespider/Meross/issues/28#issuecomment-851441094

platofan23, 2021-08-19:
> If you are on using Homeassistant there are a couple of options.
> Harder: https://github.com/krahabb/meross_lan/discussions/63#discussion-3525179
> 
> Simple: https://github.com/bytespider/Meross/issues/28#issuecomment-851441094
> 

That worked but i don't want my Raspberry Pi to bei Connect to the WiFi of the Plug all the time...

bytespider, 2021-08-19:
If you followed the commands in the first option then the final step will cause the plug to reboot and connect to the WIFI details you provided.

platofan23, 2021-08-19:
I didnt got that to Work. But i found an Other way. I connected the Plug via the meross app and Just Blocked the Internet Access with my Fritzbox. I still can use the meross LAN component for HA

Ostepanov, 2021-08-20:
> 
> 
> I didnt got that to Work. But i found an Other way. I connected the Plug via the meross app and Just Blocked the Internet Access with my Fritzbox. I still can use the meross LAN component for HA

It's a bit risky, because of blocked connection to meross mqtt, sometimes device will reboot itself.

Check if common name is the same in all your certificates

bytespider, 2021-08-20:
The problem is not related to Meross directly. The device when it cannot find the broker will try the second broker defined before finally failing. You should choose the best option for you. Either your own broker or use the Meross ones.

 We cater for your own broker setup.

platofan23, 2021-09-12:
> 
> 
> The problem is not related to Meross directly. The device when it cannot find the broker will try the second broker defined before finally failing. You should choose the best option for you. Either your own broker or use the Meross ones.
> 
> We cater for your own broker setup.

Later i solved the problem by adding the plugs with your HA-Integrations. That worked great =)

closed 2021-09-12 by platofan23