bytespider/Meross #35: TypeError [ERR_INVALID_ARG_TYPE] at logRequest(), app.js

hnakai0909, 2021-08-23:
**Problem:**
`./meross` throws `TypeError [ERR_INVALID_ARG_TYPE]` and cannot see what the real error is,
when `./meross` with `-v` option and situation that occuring some network error, such as reserving inappropriate IP address to `--gateway` option.

**Log:**
`10.10.10.155` is not meaningful IP address example which not exist Meross device.
```
$ sudo ./bin/meross info --gateway 10.10.10.155 -v
Getting info about device with IP 10.10.10.155
> POST /config
> Host: 10.10.10.155
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '3d09087d7574042037eb832a3a3f8ff8',
    timestamp: 1629745262,
    sign: '3c699b7bae58b4b647e8864e6b1299ac'
  },
  payload: {}
}

node:internal/process/promises:227
          triggerUncaughtException(err, true /* fromPromise */);
          ^

TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:329:5)
    at validateString (node:internal/validators:129:11)
    at Url.parse (node:url:169:3)
    at Object.urlParse [as parse] (node:url:156:13)
    at new URL (/home/pi/Downloads/Meross/lib/api.js:6:35)
    at logRequest (/home/pi/Downloads/Meross/lib/api.js:62:15)
    at handleRequestError (/home/pi/Downloads/Meross/lib/api.js:99:13)
    at API.deviceInformation (/home/pi/Downloads/Meross/lib/api.js:205:13)
    at processTicksAndRejections (node:internal/process/task_queues:94:5)
    at async /home/pi/Downloads/Meross/bin/meross-info:42:5 {
  code: 'ERR_INVALID_ARG_TYPE'
}

```

**My environment:**
Hardware : Raspberry Pi 1 (Yes, it's old)
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
$ node -v
v15.6.0
$ npm -v
7.21.0
```

**Cause?:**
In
https://github.com/bytespider/Meross/blob/7a9ea926d3479b6a90dff256c59547c57f7614aa/lib/api.js#L62
, `logRequest()` fails if argument `request` does not contains `url` .

**My workaround:**
In https://github.com/bytespider/Meross/blob/7a9ea926d3479b6a90dff256c59547c57f7614aa/lib/api.js#L98
,Fix
`} else if (error.request) {`
to
`} else if (error.request && error.request.url) {`

**Log with applying workaround:**
Shows error normally (without TypeError).
```
$ sudo ./bin/meross info --gateway 10.10.10.155 -v
Getting info about device with IP 10.10.10.155
> POST /config
> Host: 10.10.10.155
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '848fbdb9951232b81a9f1d5858406023',
    timestamp: 1629746553,
    sign: '42006e85b09bf0d4da77427515cab7ff'
  },
  payload: {}
}

Error connect EHOSTUNREACH 10.10.10.155:80
```

bytespider, 2021-08-23:
Well the true error is actually that the device you specified at `10.10.10.155` doesn't have the Meross HTTP API or doesn't exist. It's unusual for a Meross device to use anything but `10.10.10.1` except when it's connected to your WiFi.

hnakai0909, 2021-08-23:
I mean `10.10.10.155` is just one of the example to manually cause network-related error, so this address does not have meaning.

What I wanted to say is, when network-related error happens, some of that that does not contains `url` in `error.request`, the TypeError cause, and we cannot get what the real error is.

hnakai0909, 2021-08-23:
Related PR #36 issued.

hnakai0909, 2021-08-23:
Sorry, there are no need to add `sudo`, so I fixed logs.

closed 2021-11-05 by bytespider