bytespider/Meross #43: Question: Is it important to sign the messages?

eduardomarcos, 2021-12-11:
Hi bytespider,

First of all, congrats for this amazing work. I'm a huge fan of meross devices and currently I use a few of them with MQTT and Nodered for IoT stuff.

In my current implementations, I'm always using the same messageId, with the same sign and timestamp, and I would say I haven't had a problem for doing that (of course, those values are not random, are based on a valid verification to be sure the device does not reject the http request. I forgot to mention that all 'input' interactions I do with the devices are based on http requests.

So my question is...what is the different between using a new generated messageId, etc. per call versus using the same again and again?

Best regards!

bytespider, 2021-12-11:
Thank you 

From what I can tell, very little as the devices don't seem to keep track of what message I'd has been used.

It's more likely to do with the two way communication via mqtt as in some cases the cloud broker will respond with the same message I'd so the other knows what it's a response to

eduardomarcos, 2021-12-11:
Ok, makes sense. It can be used as an idempotency mechanism to control the messages consumed using mqtt.

I will continue trying because with switches I have no problem at all, but with the light bulbs, sometimes, I get some device a bit crazy and it stops accepting http requests...still don't know if due to wifi connection issues or some mechanism to block the device some time if it receives a lot of request in some specific time (using http I'm doing polling to check the status of the devices every 5 secs)

Thanks bytespider!

bytespider, 2021-12-11:
If you you feel confident you could reconnect the bulb with the cloud and record the messages for a few hours using https://github.com/bytespider/mqtt-debug.  More Info on the process I used here https://github.com/krahabb/meross_lan/discussions/126

closed 2022-11-28 by bytespider