bytespider/Meross #46: Devices are "lost"  on Meross Lan

hampelo, 2022-02-27:
I rebooted my router and since have "lost"  several meross devices.  It is possible they have changed IP.

meross lan:  2.5.5
HA:  2022.2.9

Can these devices be recovered?  Do I need to remove and read the integration?

bytespider, 2022-02-27:
If the Led on the devices is green then that would suggest they can connect to the network and the broker (cloud or local).

hampelo, 2022-02-27:
All devices work physically and work fine through the meross app so the connection to the network is good.  They have simply become "unavailable" to HA.  Not all of them, just 5 or 6 of the 30.

bytespider, 2022-02-28:
This is an issue for Meross LAN then, since it's unrelated to this tool.

closed 2022-02-28 by bytespider