bytespider/Meross #47: Struggle with Meross light switch disconnecting every few seconds from MQTT broker

agrabren, 2022-04-25:
I'm not sure what I'm doing wrong, but watching the Mosquitto log, I keep seeing the device connect, subscribe, publish a few times, and disconnect. I have over 60 different Meross switches that I will eventually all pair to my home assistant, but only after I can reliably control one. 

Mosquitto version:
```
  mosquitto version 2.0.14 starting
```

```
Config:
port 8883

require_certificate false
use_identity_as_username false
allow_anonymous true

capath /etc/mosquitto/certs

# replace with your CA Root and server certificate and key paths
cafile /etc/mosquitto/certs/mqtt_ca.crt
certfile /etc/mosquitto/certs/mqtt.crt
keyfile /etc/mosquitto/certs/mqtt.key

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log
log_type all

include_dir /etc/mosquitto/conf.d
```

Log:
```
1650919713: New connection from 192.168.1.171:54012 on port 8883.
1650919713: New client connected from 192.168.1.171:54012 as fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (p1, c1, k120, u'48:e1:e9:43:e9:bc').
1650919713: No will message specified.
1650919713: Sending CONNACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (0, 0)
1650919713: Received SUBSCRIBE from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb
1650919713:     /appliance/2012317778118925583748e1e943e9bc/subscribe (QoS 1)
1650919713: fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb 1 /appliance/2012317778118925583748e1e943e9bc/subscribe
1650919713: Sending SUBACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb
1650919713: Received PUBLISH from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (d0, q1, r0, m3, '/appliance/2012317778118925583748e1e943e9bc/publish', ... (347 bytes))
1650919713: Sending PUBACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (m3, rc0)
1650919713: Received PUBLISH from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (d0, q1, r0, m4, '/appliance/2012317778118925583748e1e943e9bc/publish', ... (731 bytes))
1650919713: Sending PUBACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (m4, rc0)
1650919720: Received PUBLISH from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (d0, q1, r0, m5, '/appliance/2012317778118925583748e1e943e9bc/publish', ... (731 bytes))
1650919720: Sending PUBACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (m5, rc0)
1650919726: Received PUBLISH from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (d0, q1, r0, m6, '/appliance/2012317778118925583748e1e943e9bc/publish', ... (731 bytes))
1650919726: Sending PUBACK to fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb (m6, rc0)
1650919732: Received DISCONNECT from fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb
1650919732: Client fmware:2012317778118925583748e1e943e9bc_bVUplGdeu4lGBUtb disconnected.
1650919736: New connection from 192.168.1.171:54013 on port 8883.
```

bytespider, 2022-04-26:
I've not tested against Mosquito 2.

This is interesting behaviour, it's possible the device is expecting a response.

We know that Meross devices are hard coded to contact one of several NTP servers, are your devices able to do so?

bytespider, 2022-04-26:
Update: I tested with a basic Mosquitto setup and my MSS310 connected fine.

lechercheur123, 2022-04-26:
I don't use Home Assistant, that's why I have my own script to handle the MQTT messages. The script listen to the MQTT messages sent by the MSS310 and send the power consumption to an other topic.

agrabren, 2022-04-27:
I'm really sorry I can't get test all this stuff tonight, my work day got the best of me and I haven't had much luck getting an easier-to-debug MQTT server setup. So I lost a little ground on this. I can provide these details about the device I'm using:
  Type:  mss510x
  Version: 3.0.0
  Chip: mt7682
  Firmware ver: 3.2.2
Hopefully tomorrow I'll have better luck with a clean MQTT installation and configuration with new certs and such. Then I will update with my latest, and hopefully some additional debug information.

bytespider, 2022-04-27:
@agrabren Make sure you try the setup on a network with no restriction on out going traffic. Then as @lechercheur123 explains, try configuring Appliance.System.Clock and Appliance.System.Time.

Another option is to use Home Assistant and the [Meross LAN](https://github.com/krahabb/meross_lan) integration. Be aware that you must use version <= 5.1 of the MQTT integration as in later versions compatibility with Meross devices broke and they refuse to fix it stating that they will not support usernames that don't conform with the "Unix password file" convention (Meross devices use MAC addresses as usernames and the colon breaks the password file).

agrabren, 2022-04-29:
Thanks to both of your insights and comments, I've made progress! Thank you so much (although I may update the Custom Pairer app as it could really use some TLC and documentation for us newbies). The problem I was encountering was that I didn't have the correct key. Since I did a full factory reset of the switch and fumbled around with the custom pairing app, I hadn't noticed that the MQTT key was both required and something that could be programmed in via the app. Once I got MQTT Hub going with the correct key, I started getting to register the switch, which led to everything appearing to work. So I'm going to close this with comment, and I'll suggest some edits and make a pull request on some fixes for the custom pairer (I'm an Android dev by day, home automation newbie by night)

closed 2022-04-29 by agrabren