bytespider/Meross #48: Unable to Setup Meross MSS510

timnolte, 2022-05-16:
So, today I tried setting up a new Meross MSS510 switch using the local only MQTT setup. I first attempted running the npx meross info command after connecting to the switches WiFi. While it was attempting to pull the switch information it toggled the switch on/off a couple of times and then just timed out. I ran with the verbose option and got this output:
```shell
Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '483cdbd7553e6dacfd6fe1ecb2de5e8f',
    timestamp: 1652554693,
    sign: '2df1f5d866c518eb3de9171ddb614132'
  },
  payload: {}
}

(node:3996662) UnhandledPromiseRejectionWarning: TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at validateString (internal/validators.js:120:11)
    at Url.parse (url.js:159:3)
    at Object.urlParse [as parse] (url.js:154:13)
    at new URL (/home/tnolte/.npm/_npx/3996214/lib/node_modules/meross/lib/api.js:6:35)
    at logRequest (/home/tnolte/.npm/_npx/3996214/lib/node_modules/meross/lib/api.js:62:15)
    at handleRequestError (/home/tnolte/.npm/_npx/3996214/lib/node_modules/meross/lib/api.js:99:13)
    at API.deviceInformation (/home/tnolte/.npm/_npx/3996214/lib/node_modules/meross/lib/api.js:204:13)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
(node:3996662) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:3996662) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
I didn't having any problem with the Meross Smart Plug (MSS110) that I was initially testing with. I ended up just setting up the switch through the Meross app and got it added via Meross LAN the normal way. Is this switch not supported for some reason?

From the Meros app this is the details for the device:

- Model Number: mss510x
- Firmware Version: 3.2.2
- Hardware Version: 3.0.0

timnolte, 2022-05-17:
Just a follow-up that I was actually able to get an mss510 switch paired with Meross LAN using MQTT with the pairer mobile app, that is mentioned in the linked issue. The fact that the mobile app was able to get the switch paired but this Node Setup Utility can't without error is a bit baffling. Would the error I was receiving have anything to do with the version of Node I was running under? I do have NVM so can switch between Node versions. I quite often switch between Node v12 & v14.

timnolte, 2022-05-24:
Just noting that I have tried both Node 12 & Node 14, installed the app via NPM, also tried just using NPX but that results in a lot of delay when connected to the device and there is no internet. I'm not sure if these devices are actually supported or not. Basically when I attempt to run `meross info` I'm disconnected from the device's WiFi network and I get no response. I have been able to use the Custom Pairer Android app, however, some of my devices only manage to get paired with Home Assistant via IP and not MQTT.

timnolte, 2022-05-26:
So a bit of additional information is that I think this is partially related to newer HomeKit devices, this doesn't explain the challenges with the mss510 smart switch so there might be more to it, however there is definitely challenges with pairing the newer HomeKit devices as mentioned in [this thread](https://github.com/krahabb/meross_lan/discussions/63#discussioncomment-2824237).

bytespider, 2023-01-15:
@timnolte this tool requires node 12+

In general this tool works when the device has been factory reset and you have connected to it's hotspot.

I do not own any of these devices so I'm unable to assist.

closed 2023-01-15 by bytespider

timnolte, 2023-01-15:
@bytespider as I noted I tried with both Node 12 & Node 14. The Android custom pairer app seems to have the same problem so it seems that neither this tool or the custom pairer app are updated to properly work with HomeKit compatible devices. At this point it seems that setting some of these devices up via the Home Assistant HomeKit integration is more reliable or is really the only compatible way to have them setup. It would be nice if they could be setup via MQTT but I have pretty much given up on that solution with these newer devices for now.