bytespider/Meross #52: MSS310 keeps blinking

ichmachnixichgucknur, 2022-09-10:
Hi,
I just bought a new MSS310 and followed your instructions to get this device working in LAN without pairing to the cloud. Your tools worked fine and the device is also usable by HA. I can switch and get usage Information. The only issue is that the LED keeps blinking green (fast) if the switch is on. When it is turned off, the LED is off, too.
According to the manual, the LED should be simply green.

Hardware: 2.0.0
Firmware: 2.1.16

Do you have any hint on this?

lechercheur123, 2022-09-10:
Hi,
my MSS310 is blinking green only at startup or when my Wi-Fi Router reboots. Maybe the Wi-Fi signal is too weak ?

ichmachnixichgucknur, 2022-09-10:
According to HA ... signal strength 100% and the AP is approx 3m away. So this should be ok.

lechercheur123, 2022-09-10:
Do your MSS310 have access to the Internet ?

ichmachnixichgucknur, 2022-09-10:
Yes, they have.
But they actually do not need internet access in this setup. Or?
Would they do an automatic Firmware update then?

lechercheur123, 2022-09-10:
They need to have access to some NTP servers. You can simulate these NTP servers, or let the MSS310 access them through internet.

ichmachnixichgucknur, 2022-09-10:
Do they use the NTP servers provided by DHCP or do the need access to their own?

lechercheur123, 2022-09-10:
Their own. At least 3 different one if I remenber correctly. You can determine the NTP servers IPv4 addresses via a Wireshark capture.

ichmachnixichgucknur, 2022-09-10:
But ... they have internet access. So, this should not be the problem 🤔

ichmachnixichgucknur, 2022-09-10:
What firmware do you use?

lechercheur123, 2022-09-10:
```
{
    "header":{
        "messageId":"47e87acd7ced4519517df8916942f1a1",
        "namespace":"Appliance.System.All",
        "method":"GETACK",
        "payloadVersion":1,
        "from":"/appliance/myuuid/publish",
        "timestamp":1662839986,
        "timestampMs":107,
        "sign":"1f6f6ebac1172344cb65448eaade5020"
    },
    "payload":{
        "all":{
            "system":{
                "hardware":{
                    "type":"mss310",
                    "subType":"un",
                    "version":"6.0.0",
                    "chipType":"rtl8710cf",
                    "uuid":"myuuid",
                    "macAddress":"48:e1:e9:00:00:00"
                },
                "firmware":{
                    "version":"6.1.10",
                    "compileTime":"2021/08/17-14:35:59",
                    "wifiMac":"10:7b:44:00:00:00",
                    "innerIp":"192.168.2.246",
                    "server":"mqtts.pommetelecom.fr",
                    "port":8883,
                    "userId":0
                },
                "time":{
                    "timestamp":1662839986,
                    "timezone":"Europe/Paris",
                    "timeRule":[
                        [
                            1648342800,
                            7200,
                            1
                        ],
                        [
                            1667091600,
                            3600,
                            0
                        ],
                        [
                            1679792400,
                            7200,
                            1
                        ]
                    ]
                },
                "online":{
                    "status":1
                }
            },
            "digest":{
                "togglex":[
                    {
                        "channel":0,
                        "onoff":1,
                        "lmTime":92191
                    }
                ],
                "triggerx":[
                ],
                "timerx":[
                ]
            }
        }
    }

}
```

ichmachnixichgucknur, 2022-09-10:
Ok, you are using completely other HW and Firmware.
I am using the "hack mode". Could that lead to such problems? Thought it would just an option to set it up without a key, but the readme complains about "random issues".

lechercheur123, 2022-09-10:
The key is used in the signature of MQTT messages. I guess it could be an issue to have a blank key.

I used an app to pair my MSS310 : https://github.com/albertogeniola/Custom-Meross-Pairer

ichmachnixichgucknur, 2022-09-10:
Ok, I will try to reconfigure using a key.

lechercheur123, 2022-09-10:
Also, if you just want to disable the LED, you can take a look at this setting : https://github.com/arandall/meross/blob/master/doc/protocol.md#appliancesystemdndmode

bytespider, 2023-01-15:
@ichmachnixichgucknur did you get this issue resolved?

Closing due to inactivity.

closed 2023-01-15 by bytespider