bytespider/Meross #55: Unable to setup device MSS210

zisco67, 2023-01-08:
I'm trying to setup my MSS210 - but I'm stuck in a connection error. 

```
bodhi@bodhi:~$ npx meross setup --wifi-ssid '<SSID>' --wifi-pass '<PASS>' --mqtt mqtts://zshq-ohab:8883
npx: installed 27 in 72.512s
Setting up device with IP 10.10.10.1
┌──────────────────────────────────────────────────────┬────────────────────────
│Primary MQTT broker                                   │zshq-ohab:8883
├──────────────────────────────────────────────────────┼────────────────────────
│Failover MQTT broker                                  │zshq-ohab:8883
└──────────────────────────────────────────────────────┴────────────────────────
Error Unable to connect to device
Error Unable to connect to device
Error Unable to connect to device
Device will reboot...
```
Is there a way to get more info about the connection problem?

simonporter007, 2023-01-10:
Finally managed to connect my MSS210 after a few days of messing and debugging and trial and error using these tools. When you tried to run the setup command, was the plug on the right network? or was your machine on the right network?

If it's not yet connected to your WiFi, you need to connect to the AP that the plug provides, `Meross_AP_XXXX` and then run the above command. If you've already used the tool successfully before, or used the official app, you will need to provide the IP address of the plug as the "gateway" setting with the `-g 192.0.0.0.0` parameter to the tool.

zisco67, 2023-01-11:
When connected to the `Meross_AP_XXXX` I didn't use the `-g` Parameter; I tried now with `-g 192.168.1.213` unfortunately unsuccessful as well.

simonporter007, 2023-01-11:
@zisco67 sorry for not being clear! The `-g` would only be required if the device is already connected to your home WiFi network (in case you ran the tool previously). If you're connecting to the Meross AP instead when running the command, then the default gateway of 10.10.10.1 should be fine.

I guess you've tried resetting it just in case by holding the power button for 5 seconds and letting it reboot itself into pairing mode again? When you connect to the Meross AP, can you navigate to http:10.10.10.1 just to verify the IP address is the correct one or rather, can you get a valid output from `./bin/meros info` ?

zisco67, 2023-01-11:
@simonporter007 sorry for being unclear, too!
Scenario: Connected to Meross-WIFI in paring-mode (host IP: 10.10.10.2):
```
markus@bangui:~/Downloads/Meross-master/bin$ sudo npx meross info
[sudo] password for markus:
npx: installed 27 in 133.483s
Getting info about device with IP 10.10.10.1
Error Unable to connect to device
```

simonporter007, 2023-01-11:
I don't think the tool has much more logging than that. It's weird as it should just be making a http request to the default ip address. If you've reset it and connected to the AP, I don't know why that wouldn't work.

If you try navigating to http://10.10.10.1/config in your browser after connecting to the Meross AP, does it load?

simonporter007, 2023-01-11:
I think the only reset is to hold down the button the plug for 5 seconds. This seems to either:
- flash red/orange/green quickly before rebooting (you'll hear the click of it going off)
- flash orange/green slowly which signals it's reset back to pairing mode and the Meross AP should be active

I also forgot there's the verbose option `-v` which might print some additional details. Can you try resetting again into pair mode, and then try meross info with the -v flag?

zisco67, 2023-01-11:
```
markus@bangui:~/Downloads/Meross-master/bin$ sudo npx meross info -v
npx: installed 27 in 8.814s
Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
Error Object.fromEntries is not a function
```

simonporter007, 2023-01-11:
ah! What version of node are you running? `node -v` I don't think it's written anywhere, but I think you'll need node v12+ (at least from that error)

zisco67, 2023-01-11:
Thank you so much, that was the problem...

bytespider, 2023-01-14:
Thank you @simonporter007 for helping @zisco67

zisco67, 2023-01-15:
There is still an issue, when I try to change the state of the power outlet:
```
1673718921: New connection from 192.168.1.213:49260 on port 8883.
1673718922: Client fmware:2009072623298651802748e1e92fbcb9_Mdpk5bcXwO6q4oFr already connected,
ng old connection.
1673718922: New client connected from 192.168.1.213:49260 as fmware:2009072623298651802748e1e92
_Mdpk5bcXwO6q4oFr (p1, c1, k30, u'48:e1:e9:2f:bc:b9').
```
It reboots and reconnects, but doesn’t change the state - any ideas left? Thank you!

bytespider, 2023-01-21:
@zisco67 what's the message you are sending in?

zisco67, 2023-01-24:
Good question, I need to log MQTT - but I think, one of the problems is (as mentoined here in step 3 https://github.com/arandall/meross/blob/main/doc/provisioning.md), that the device is waiting for a reponse from MQTT (even tough my device's indicator led is already in steady green state): _In my testing the device sends an Appliance.System.Clock to the /appliance/<UUID>/publish topic until it gets a response. From testing the response must be sent via MQTT to the topic /appliance/<UUID>/subscribe. If a response is not received within 10 attempts the device will restart and go back to the beginning of Step 2._

bytespider, 2023-11-01:
Messages sent in response to MQTT needs to be implemented by you I'm afraid. This tool only facilitates pairing the device with your WIFI and setting the MQTT broker details.

closed 2023-11-01 by bytespider