bytespider/Meross #57: Setting MQTT and Wifi for MSS210

simonporter007, 2023-01-11:
Hi there!

Been fiddling with the MSS210 plug I picked up from Amazon recently and looking to get it speaking offline to my local MQTT broker. I've gotten it working twice now (I redid it the second time to make sure the steps were accurate) and was looking to push a PR to allow switching MQTT settings separate from Wifi.

Except, it seems like I have to allow it to connect to Meross IOT broker at least once, before switching over the MQTT details to my local server. Perhaps someone knows more about the protocol can let me know why this might be?

The steps I took:
- reset the device back into pairing mode by holding the button for 5 seconds
- connected to Meross AP and was able to call `./bin/meross info`
- set up Wifi and dummy MQTT (for now) with `./bin/meross setup` making sure to specify both `user` and `key` (I'm not sure if these are required specifically, but if it has to talk to Meross MQTT once, then presumably so) and `mqtt` with `127.0.0.1:8883`

The device should reboot and reconnect to the local Wifi network now correctly. I could verify this with `./bin/meross info` and specifying the same `key`. MQTT connection is obviously broken/unconfigured at this point!

Here's where I go off track :sweat_smile: I wanted to test setting MQTT only, so was playing with modified code, specifically I just didn't call the `configureWifiCredentials` function, and called `./bin/meross setup` with the correct MQTT local broker. This works, and the device doesn't reboot, it tries to connect to the new broker immediately. I can subscribe to the `/appliance/....../publish` topic and see the plug publishing messages. This is where it looks to get stuck in a loop.

It keeps publishing _Appliance.Control.Bind_ and then _Appliance.Control.Report_ messages before disconnecting, reconnecting and continuing the process. It seems like it's waiting for a response from Meross?

I run the same code as above, but this time with the official Meross MQTT broker details (found from pcap/debugging elsewhere) and let it sit for a minute. I then run the same code again but with my local MQTT broker details, and now it connects fine and happy, and works as expected. I get an initial _Appliance.Control.Report_ message and then it responds to button presses with _ToggleX_ etc.

I've tried mqtt-debug with certs, but it doesn't seem to like proxying the connection on to Meross MQTT broker for some reason, I see the CONNECT but nothing else. As I can't see the packets received when it's connecting to the official broker, I'm not sure what other magic is happening on that initial MQTT connection with the _bind_/_report_.

Anyone able to shed some light on that part?

bytespider, 2023-01-15:
Key and User ID: These aren't strictly required. Key is there to sign the messages, I defaulted to a blank string. The user ID isn't used anywhere and is sent by the device when logging into the Meross Cloud API

The only communication from the device to the internet for devices I've tested was for NTP, and those urls seem to be hardcoded in the device firmware.

From memory, the correct response to Appliance.Control.Bind SET is SETACK with the message message Id you received.
I did just find there is a repo with the [API well documented](https://github.com/arandall/meross/blob/main/doc/protocol.md#appliancecontrolbind)

I'm curious that mqtt-debug didnt work for you, did you use the docker certs or create your own?

simonporter007, 2023-01-15:
Ah, yes, I found that repo too last week and was going to give their suggestion in provisioning a go, but haven't had a chance to play around with it just yet. In the provisioning doc it mentions:

> In the "Official" scenario Meross performs a number of requests to the device to configure the device fully. Including but not limited to a get of Appliance.System.All and a setting of Appliance.System.Time.

So wondered if that explained what I was seeing. No mqtt set time, so it restarts.

Re:mqtt-debug, I was using the same self signed certs I was using for local broker. I'll give the docker route a go though and see if that does the trick!

simonporter007, 2023-02-04:
@bytespider sorry for the delay! I was able to do a bit more tinkering, and once I used the docker image for mqtt-debug, it seemed to work. Must be something wrong with the certs I generated I guess. Thanks for that!

From what I can see, it needs a userId and/or key now? I'm not sure if both are required or just one. If I leave those off, I don't get a proper MQTT connection to my local broker.

After using the docker image for mqtt-debug, it seems the final piece it needs is a SETACK message in response to the `Appliance.Control.Bind` message it publishes when it connects to the broker. It can use the same values that are given in the header as the arandall repo mentions. I was able to set up some node red flow that listens for the `/publish` messages, if _Appliance.Control.Bind_ generate a new message to publish with the values taken from the payload, and publish it back out to the /subscribe topic. That seemed to do the trick and it connects every time now.

The message format to be published is:
```
/appliance/<appliance_id>/subscribe {
  header: {
    messageId: <same as the one received in the 'SET' incoming message>,
    namespace: 'Appliance.Control.Bind',
    timestamp: <same as the one received in the 'SET' incoming message>,
    method: 'SETACK',
    sign: <same as the one received in the 'SET' incoming message>,
    from: '/cloud/hook/subscribe' // don't know if required, kept it just in case
  },
  payload: {}
}
```

Not sure if the same /subcribe can be called over HTTP but this'll keep me going for now :sweat_smile: 

Thanks again!

closed 2023-02-04 by simonporter007