bytespider/Meross #65: mss110 power switch with firmware 7.3.19 requires user and key

amair, 2023-09-25:
After upgrading an ms110 switch to firmware 7.3.19 using meross-setup without a user and key setup results in an error.

The user (as returned from meross-info is the MAC address of the device.  Trying to set this as the user in meross-setup results in the app trying to interpret the string as an integer which then fails validation.

amair, 2023-09-26:
OK, some more details of the issue using verbose logging option

**Initial Attempt: No user or key**
user@host bin % sudo ./meross-setup --verbose --gateway 10.10.10.1 --wifi-ssid '<WIFI>' --wifi-pass '<WIFI_PASS>' --mqtt mqtts://mqtt.local:8883
Setting up device with IP 10.10.10.1
│Primary MQTT broker                               │mqtt.local:8883                         │
│Failover MQTT broker                              │mqtt.local:8883                         │
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'SET',
    namespace: 'Appliance.Config.Key',
    messageId: 'bc7d4be83a18449f4112e8ff7ddb85c0',
    timestamp: 1695711098,
    sign: '8bacc21ffe154410dadf9e6db34b1de8'
  },
  payload: {
    key: {
      userId: '0',
      key: '',
      gateway: {
        host: 'mqtt.local',
        port: '8883',
        secondHost: 'mqtt.local',
        secondPort: '8883',
        redirect: 1
      }
    }
  }
}

node:internal/errors:496
    ErrorCaptureStackTrace(err);
    ^

TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:405:5)
    at validateString (node:internal/validators:162:11)
    at Url.parse (node:url:170:3)
    at Object.urlParse [as parse] (node:url:141:13)
    at new URL (/Users/alastair/Programming/openhab/Meross/lib/api.js:6:35)
    at logRequest (/Users/alastair/Programming/openhab/Meross/lib/api.js:62:15)
    at handleRequestError (/Users/alastair/Programming/openhab/Meross/lib/api.js:99:13)
    at API.configureMqttServers (/Users/alastair/Programming/openhab/Meross/lib/api.js:341:13)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /Users/alastair/Programming/openhab/Meross/bin/meross-setup:89:9 {
  code: 'ERR_INVALID_ARG_TYPE'
}

**Query Device for key and password**
user@host bin % sudo ./meross-info -g 10.10.10.1
Getting info about device with IP 10.10.10.1
│Device       │mss110 us rtl8710cm (hardware:7.0.0 firmware:7.3.19)                           │
│UUID         │<UUID>                                               │
│Mac address  │<MAC_ADDRESS>                                                              │
│IP address   │10.10.10.1                                                                  │
│Credentials  │User: <MAC_ADDRESS>                                                        │
│             │Password: <KEY>                                   │
│MQTT topics  │Publishes to: /appliance/<UUID>/publish              │
│             │Subscribes to: /appliance/<UUID>/subscribe           │

**Successful attempt using user and key (with code modification to remove conversion to int for user) **
user@host bin % sudo ./meross-setup --user '<MAC_ADDRESS>' --key '<KEY>' --verbose --gateway 10.10.10.1 --wifi-ssid '<WIFI>' --wifi-pass '<WIFI_PASS>' --mqtt mqtts://mqtt.local:8883
Setting up device with IP 10.10.10.1
│Primary MQTT broker                               │mqtt.local:8883                         │
[... 139 lines cut ...]
  payload: {
    wifi: { ssid: '<HIDDEN>', password: '<HIDDEN>' }
  }
}

< 200 OK
< content-type: application/json
< connection: close
<
{
  header: {
    messageId: '9dd06cd2a64f4990ea11c872b2d076e9',
    namespace: 'Appliance.Config.Wifi',
    method: 'SETACK',
    payloadVersion: 1,
    from: '/appliance/<UUID>/publish',
    uuid: '<UUID>',
    timestamp: 33,
    timestampMs: 460,
    sign: 'c82c8620f4443780167834e7f4082df1'
  },
  payload: {}
}

Device will reboot...

bytespider, 2023-09-26:
Have you tried passing a user id that isn't 0. It's possible they have changed the firmware so that 0 is no longer valid.

bytespider, 2023-11-01:
Closing due to inactivity

closed 2023-11-01 by bytespider