bytespider/Meross #67: [QUESTION] What exactly is --gateway?

wsw70, 2023-10-31:
> Before you can use the tool to setup your device you need to put it into paring mode and connect to it's Access Point. It's IP address is known as the `--gateway` parameter and is typically 10.10.10.1.

I recently paired a device and was confused by `--gateway` (it's been a long time since I paired something). The README mentions that this parameter is its IP address.

Which IP address, exactly? 
- is this the address of the access point, and I will get an appropriate IP upon joining the AP SSID? (in which case I should never need it)
- or is this the IP address of the **paired** device? (once it has rebooted and joined my own network). I used it in that context to get information about an existing and running device:

```text
PS > node .\bin\meross-info --gateway 192.168.10.183

Getting info about device with IP 192.168.10.183
(then comes the table with the device info)
```

bytespider, 2023-10-31:
`--gateway` is the IP address of the device at any point in time.

During pairing the IP will be typically 10.10.10.1 and does correspond to the gateway IP that you see when connecting to the WIFI access point the device broadcasts.
Later the IP corresponds to the IP the device is given by your network, and in theory the tool could be used to change the MQTT data of an already configured device.

I do understand your confusion, perhaps better documentation is required.

bytespider, 2023-11-03:
I have updated referenced to `--gateway` with `--ip` in the rewrite i'm doing
https://github.com/bytespider/Meross/blob/be2a70ba759278e4956478b95419df3de219c625/bin/meross-setup.js#L40-L44

closed 2023-11-03 by bytespider