bytespider/Meross #71: Error Unable to connect to device - TypeError: Cannot read properties of undefined (reading 'system')

fireheadman, 2023-11-01:
Getting this error on 3 new Meross garage openers (MSG100 - non Homekit models)
Running nodejs/npm w/commander w/uuid installed on a macbook pro

Any clue what this might be result of?
```
[fireheadman@minion bin 313]$ sudo ./meross-info
Getting info about device with IP 10.10.10.1
Error Unable to connect to device
/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:155
        const system = data.system
                            ^

TypeError: Cannot read properties of undefined (reading 'system')
    at API.deviceInformation (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:155:29)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /Users/fireheadman/meross/Meross-1.0.14/bin/meross-info:42:5

Node.js v21.1.0
```

fireheadman, 2023-11-01:
saw you are also using npx, so tried that also, same error.

```
[fireheadman@minion bin 317]$ npx meross info
Getting info about device with IP 10.10.10.1
Error Unable to connect to device
/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:155
        const system = data.system
                            ^

TypeError: Cannot read properties of undefined (reading 'system')
    at API.deviceInformation (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:155:29)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async /Users/fireheadman/meross/Meross-1.0.14/bin/meross-info:42:5

Node.js v21.1.0

[fireheadman@minion bin 318]$ ping 10.10.10.1
PING 10.10.10.1 (10.10.10.1): 56 data bytes
64 bytes from 10.10.10.1: icmp_seq=0 ttl=255 time=8.212 ms
64 bytes from 10.10.10.1: icmp_seq=1 ttl=255 time=39.465 ms
^C
--- 10.10.10.1 ping statistics ---
2 packets transmitted, 2 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 8.212/23.839/39.465/15.627 ms
```

bytespider, 2023-11-01:
could you try with `--verbose` that should log the response

fireheadman, 2023-11-01:
Also noticed when I run this command while running a live ping to 10.10.10.1 the ping will stop.   Looks like it causes a reboot of the device?

```
[fireheadman@minion bin 367]$ pwd
/Users/fireheadman/meross/Meross-1.0.14/bin
[fireheadman@minion bin 367]$ sudo ./meross-info -v -g 10.10.10.1 --include-wifi
Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '1f57d5ee57b31014bb7705c4687eca22',
    timestamp: 1698869249,
    sign: 'd85bf65534a9d5e12ea5f281bd7522fa'
  },
  payload: {}
}

node:internal/validators:162
    throw new ERR_INVALID_ARG_TYPE(name, 'string', value);
          ^

TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at validateString (node:internal/validators:162:11)
    at Url.parse (node:url:170:3)
    at Object.urlParse [as parse] (node:url:141:13)
    at new URL (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:6:35)
    at logRequest (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:66:15)
    at handleRequestError (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:109:13)
    at API.deviceInformationData (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:221:13)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async API.deviceInformation (/Users/fireheadman/meross/Meross-1.0.14/lib/api.js:153:22)
    at async /Users/fireheadman/meross/Meross-1.0.14/bin/meross-info:42:5 {
  code: 'ERR_INVALID_ARG_TYPE'
}

Node.js v21.1.0
```

fireheadman, 2023-11-01:
trying this same command with the previous version (Meross-1.0.12) produces this output...
[image]

fireheadman, 2023-11-01:
trying this on my macmini also... but I have an issue on this machine since I can't update npm to latest due to brew not being support on macOS Sonoma...  so using the previous version that allows me to run it.

BTW:  I do see the ping comes back eventually

```
[fireheadman@templar bin 4958]$ sudo ./meross-info -v -g 10.10.10.1 --include-wifi
Getting info about device with IP 10.10.10.1
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '6702a983546c47d3ed93723e82fc88d9',
    timestamp: 1698870081,
    sign: '73e690dece749a0016a66a1d7dc1603f'
  },
  payload: {}
}

node:internal/errors:464
    ErrorCaptureStackTrace(err);
    ^

TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:371:5)
    at validateString (node:internal/validators:119:11)
    at Url.parse (node:url:169:3)
    at Object.urlParse [as parse] (node:url:156:13)
    at new URL (/Users/fireheadman/meross/Meross-1.0.12/lib/api.js:6:35)
    at logRequest (/Users/fireheadman/meross/Meross-1.0.12/lib/api.js:62:15)
    at handleRequestError (/Users/fireheadman/meross/Meross-1.0.12/lib/api.js:99:13)
    at API.deviceInformation (/Users/fireheadman/meross/Meross-1.0.12/lib/api.js:204:13)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async /Users/fireheadman/meross/Meross-1.0.12/bin/meross-info:42:5 {
  code: 'ERR_INVALID_ARG_TYPE'
}
```

fireheadman, 2023-11-01:
ha... success this time using npx
I'm unix savvy, but not so much with npm/npx.   unsure of the differences in these commands

[image]

bytespider, 2023-11-01:
Glad you got it working. Sounds like the device was being temperamental.

npm is the package manager and installs dependencies.
npx executes a node script/library like it's binary

I updated the requirement to node 18 since 12 is no longer supported.

fireheadman, 2023-11-01:
I'm upgrading my node to version 18 now to retry this on my macmini... will let you know how it goes.

bytespider, 2023-11-01:
you could also use NVM if that doesn't work https://github.com/nvm-sh/nvm

fireheadman, 2023-11-01:
well I have broken npm by having homebrew upgraded and npm at v21.1.0
it was working with the previous version... bummer

fireheadman, 2023-11-01:
think I am back on track.... 
I have get data from the device using nodejs 16
```
[fireheadman@templar bin 5064]$ node -v
v16.20.2
```
but it will fail using anything greater than this... such as v18.x, v20.x, v21.x

fireheadman, 2023-11-01:
[image]

bytespider, 2023-11-01:
Good luck

fireheadman, 2023-11-02:
its working...
REF:  [https://github.com/krahabb/meross_lan/discussions/63#discussioncomment-7452413](url)

closed 2023-11-02 by fireheadman

bytespider, 2023-11-02:
@fireheadman I don't know if this is still true but the  HA Mosquitto add-on gave so many of us headaches. HA decided to stick with out dated technology and as such Meross device usernames weren't allowed.

I've had much better success with EMQX which is also in the Add-on store.

harphere, 2023-11-20:
Thanks @bytespider and @fireheadman. By reading this issue, I was able to narrow in on my problem and get 1 bulb and two switches working and integrated. Details like this help a lot.