bytespider/Meross #73: Device mss425e won't connect to AP, falls back to pairing mode

darthdule, 2023-11-13:
Hi, 

I have a mss425e switch. I can read the information with 

`/meross-info --verbose --include-wifi
`
 The setup command will restart the device twice and is still not connected to the wifi, remaining in paring mode

`/meross-setup --gateway 10.10.10.1 --wifi-ssid XXXXX --wifi-pass XXXXX--mqtt 192.168.XXX.XXX:8883 `

also tried

`/meross-setup --gateway 10.10.10.1 --wifi-ssid XXXXX --wifi-pass XXXXX--mqtt 192.168.XXX.XXX:8883 --wifi-cipher 6 --wifi-encryption 7 --wifi-channel 6 --verbose
`

My SSID has space characters, tried to put the ssid in "" or  '', but it is not the problem.

Two hours ago I paired it succesfully with the meross paring app, but was giving this way a try with meross LAN, because I did not get the other implementation "meross local broker" to run.

best greetings

bytespider, 2023-11-13:
@darthdule, my WIFI SSID also has spaces so I agree it's unlikely that.

Can you try without `--wifi-cipher 6 --wifi-encryption 7 --wifi-channel 6`

darthdule, 2023-11-13:
I did this already when I started, tried those parameters later, even with bssid.
Had approx. 20 tries. BUT, it happened one time that my router saw it and assigned an IP for like 1s then it disappeared and the device restarted again. There are always two restarts of the device with ~20s in between.

bytespider, 2023-11-13:
> I did this already when I started, tried those parameters later, even with bssid.

You did say that, sorry, i read your message quickly.

If you press and hold the button for approx 10s, that puts it in pairing mode which is effectively factory reset.

The device reboots after the `Appliance.Config.Wifi` command, the second resetting would be the built in failsafe to go back to pairing if it cannot connect.

Which version of the tools are you using, if your device is new enough it may need the latest version which send the `Appliance.Config.WifiX` message instead

darthdule, 2023-11-13:
I bought the meross device 3 years ago. I guess it is not part of the newer generation device just because of its age, but I don't know...

I tried to connect to a different router with the same result. 

What I don't understand correctly. 
If the MQTT broker in HA cannot be reached by the device, will it go to failsave, too? 

I checked my MQTT config in HA carefully and it seems OK. I use ZigBee2MQTT, so mosquito MQTT is working. But non-SSL port 1883 is used by ZigBee2MQTT, not the 8883 SSL, so it is not 100% proof for working mosquito setup.

darthdule, 2023-11-14:
Hi, thanks for your answer. I will look into emqx tomorrow. I fear that I will break something else, like Zigbee2MQTT. I  am not sure if it is worth the effort. 
I cant find the link right now, but I swear I read somewhere that my device is working, but dont know the mqtt setup.

bytespider, 2023-11-15:
I also have it set up for Zigbee2MQTT. I used this guide to let me know what areas needed to be reconfigured after installing EMQX. https://smarthomescene.com/guides/setting-up-emqx-mqtt-broker-in-home-assistant/

darthdule, 2023-11-15:
Thanks for the link, I took a quick look and if my headache goes away in the evening, I will try setting it up :-)

bytespider, 2023-11-15:
I dont think it's mentioned in the article, but make sure you stop and disable the mosquito add-on before your start.

darthdule, 2023-11-15:
Thank for the info, but I got stuck at the first step, basicallyy. EMQX is only available for x86 platforms and no ARM, hence no RPI support :(

bytespider, 2023-11-16:
I'm confused as I have it on my RPI 4 using the HA EMQX addon

darthdule, 2023-11-16:
> I'm confused as I have it on my RPI 4 using the HA EMQX addon

I am running 32-bit Buster with HA supervised on my RPI4, but EMQX arm build is 64-bit.
That motivated me even more to set up a new system, but I would like to use a hypervised system like proxmox, so I may switch to some NUC and run HA OS as VM.

darthdule, 2023-11-16:
I am running debian Buster 32-bit with HA supervised, but EMQX ARM builds are 64-bit.

This raises my motivation to set up a new system. I would like to go with some Hypervisor like proxmox and run HA OS as VM, but RPI is too weak, so I may switch to some NUC.

bytespider, 2023-11-16:
Gotcha, I assumed you were using HA OS.

Since you're running your own OS you could set up your own Mosquitto broker and test the devices against that, and either migrate HA to use that broker. Instructions to set up can be found here, though I'm not sure how relevant they are as there have been new versions of Mosquitto https://github.com/bytespider/Meross/wiki/MQTT

Lenovo M900's go for pretty cheap on EBay. I've been meaning to move HA over to it but I need a proper full day with no family to complain XYZ isn't working.

While this article refers to x86, you may be able to use it as reference https://wiki.debian.org/Migrate32To64Bit.

darthdule, 2023-11-16:
Thanks for the tip, but I just ordered a Beelink s12 pro with n100 CPU :-) Performance is really comparable with the old 6500T for example which I found in a lot of those M900s, but has 16GB RAM and 500GB SSD.
So I will just move on and use HA OS in a VM on the NUC.  With a little bit of luck I can try EMBX this WE.

darthdule, 2023-11-16:
Thanks for all the useful links, but I just ordered a Beelink s12 pro with n100 CPU like a minute ago :-) Performance is really comparable with the old 6500T for example which I found in a lot of those M900s, but has 16GB RAM and 500GB SSD. While searching for reviews for the mini PC came across a lot of posts where people are using it with the exact same setup I am planning
You can even replace the Wifi card with a Google coral m.2. Since I will connect to Ethernet and have a separate BT Antenna Dongle, it is great 👍

So I will just move on and use HA OS in a VM on the NUC. With a little bit of luck I can try EMQX this WE.

bytespider, 2023-12-24:
@darthdule if you have any new information please feel free to reopen this issue.

closed 2023-12-24 by bytespider

darthdule, 2023-12-24:
Will get back, soon. I recreated my whole server with a new Intel NUC and proxmox. Running EMQX, since a few days...