bytespider/Meross #77: Issues with ssl after upgrading from Debian 11 to Debian 12

override80, 2024-01-01:
Hi @bytespider ,

I've just upgraded from Debian 11 to Debian 12 as a base OS for Home Assistant. Everything used to work perfectly on Debian 11. I have a local mqtt broker, I'm not using the docker plugin but installed it via apt-get install mosquitto on the base system.

These are the versions involved:

```
root@hassio:/etc/mosquitto/conf.d# apt-cache policy openssl
openssl:
  Installed: 1.1.1n-0+deb10u3
  Candidate: 3.0.11-1~deb12u2
  Version table:
     3.0.11-1~deb12u2 500
        500 http://deb.debian.org/debian bookworm/main amd64 Packages
        500 http://deb.debian.org/debian-security bookworm-security/main amd64 Packages
 *** 1.1.1n-0+deb10u3 500
        500 http://ftp.de.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status

root@hassio:/etc/mosquitto/conf.d# apt-cache policy mosquitto
mosquitto:
  Installed: 1.5.7-1+deb10u1
  Candidate: 2.0.11-1.2+deb12u1
  Version table:
     2.0.11-1.2+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main amd64 Packages
        500 http://deb.debian.org/debian-security bookworm-security/main amd64 Packages
 *** 1.5.7-1+deb10u1 500
        500 http://ftp.de.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status
```

FWIW, I've tried upgrading openssl and mosquitto to 3.0 and 2.0 respectively, with no big differences.

The problem is that all of my 15 Meross plugs blink green and at mqtt level i get a bunch of these in the mosquitto logs:

```
1704134152: New connection from 10.11.12.166 on port 8883.
1704134152: OpenSSL Error: error:1417A0C1:SSL routines:tls_post_process_client_hello:no shared cipher
1704134152: Socket error on client <unknown>, disconnecting.
```

or, with the latest version of mosquitto/openssl: 

```
1704134218: New connection from 10.11.12.171:55436 on port 8883.
1704134218: OpenSSL Error[0]: error:0A0000C1:SSL routines::no shared cipher
1704134218: Client <unknown> disconnected: Protocol error.
```

I tried stopping the mqtt server and using the one you provided as an addon, but no success as well.

What am I missing here?

override80, 2024-01-01:
For whoever lands here... the issue was due to the private key type format. Do not use ECDSA, stick to RSA... the openssl error was really misleading TBH!

bytespider, 2024-01-01:
Thanks for the update. That makes total sense. Glad you were able to figure that out. it actually solves a couple other users issue too I suspect.

closed 2024-01-01 by bytespider