bytespider/Meross #78: Cant connect MSS710

javico88, 2024-01-12:
Hi, I'm desperate trying to connect MSS710 to my local MQTT broker. I'm going to relate what I have done until now.
I have Home Assistant Core installed in an old computer, because of that I can't use MQTT addon so I decided to install mosquitto in the system (Linux MX). I have running and working, I can connect to it locally and remotely. This is my mosquitto.conf:
``` 
#mosquitto.conf
pid_file /run/mosquitto/mosquitto.pid

persistence true
persistence_location /var/lib/mosquitto/

log_dest file /var/log/mosquitto/mosquitto.log
log_type debug
log_type error
log_type warning
log_type notice
log_type information

include_dir /etc/mosquitto/conf.d

listener 8883

allow_anonymous true
password_file /etc/mosquitto/passwd

require_certificate false
cafile /etc/mosquitto/certs/ca-root-cert.crt
certfile /etc/mosquitto/certs/server.crt
keyfile /etc/mosquitto/certs/server.key
``` 
And if I try to connect to my broker:
```
~# mosquitto_sub -L mqtts://192.168.1.119/test --cafile /etc/mosquitto/certs/ca-root-cert.crt -d
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending SUBSCRIBE (Mid: 1, Topic: test, QoS: 0, Options: 0x00)
Client (null) received SUBACK
Subscribed (mid: 1): 0
```
Then I have setup my Meross device with my computer:
```
./meross-info > device-info.txt
cat device-info.txt
Device                mss710 us rt18710cf (hardware:6.0.0 firmware:6.3.2)
UUID                 2301....64d8
Mac address       xx:xx:xx:xx:xx:xx
IP address          10.10.10.1
Credentials         User:  xx:xx:xx:xx:xx:xx
                          Password:  <Device Password>
MQTT topics       Publishes to: /appliance//publish
                          Subscribes to: /appliance//subscribe

./meross-setup -g 10.10.10.1 --wifi-ssid <myWifiSSID> --wifi-pass <myWifiPass> --mqtts://192.168.1.119
```
The device successful connect to my wifi and tries to subscribe to the MQTT broker but I get error logs and no messages is getting to the MQTT. This is the error log of mosquitto:
```
# tail /var/log/mosquitto/mosquitto.log
1705044893: Client <unknown> disconnected, not authorised.
1705045028: New connection from 192.168.1.118:53692 on port 8883.
1705045029: Sending CONNACK to 192.168.1.118 (0, 5)
1705045029: Client <unknown> disconnected, not authorised.
1705045167: New connection from 192.168.1.118:53693 on port 8883.
1705045167: Sending CONNACK to 192.168.1.118 (0, 5)
1705045167: Client <unknown> disconnected, not authorised.
1705045307: New connection from 192.168.1.118:53694 on port 8883.
1705045307: Sending CONNACK to 192.168.1.118 (0, 5)
1705045307: Client <unknown> disconnected, not authorised.
```
I have also tried to disable the certificates and connect with mqtt but I get "protocol error" message. Is this an issue os am I missing something?

Thanks for your help

bytespider, 2024-01-13:
This my just be a typo in this issue but `--mqtts://192.168.1.119` is incorrect, it should be `--mqtt mqtts://192.168.1.119`

javico88, 2024-01-13:
> This my just be a typo in this issue but `--mqtts://192.168.1.119` is incorrect, it should be `--mqtt mqtts://192.168.1.119`

Yes, it's a typo error. Sorry
In other case I would have got an unrecognized error message in that command

bytespider, 2024-01-14:
I think you're missing `per_listener_settings true`
http://www.steves-internet-guide.com/mqtt-username-password-example/

javico88, 2024-01-14:
That was not the problem but the link was very useful. I didn't know that if you set a password file the MQTT server check it even if you set anonymous login. I remove the password file and it connected it without problem.
You can set it as solved. Thank you very much

closed 2024-01-15 by bytespider