bytespider/Meross #82: Unable to setup MSS310

rolfmcduff, 2024-02-11:
I can successfully connect to the device and get the device information by the meross-info script. But when I want to setup the device I get the following log output and I can see no logs in the Mosquitto broker and the dece is not connected to the broker. What am I doing wrong?

PS: I marked secrets with <!.

./meross setup --gateway 10.10.10.1 --wifi-ssid 'FRITZ!Box Fon WLAN 7360 SL' --wifi-pass <!pass!> --mqtt mqtt://192.168.178.76:8883 --wifi-cipher 3 --wifi-encryption 6 --wifi-channel 1
Setting up device with IP 10.10.10.1
┌───────────────────────────────────────────────┬─────────────────────────────────────────────┐
│Primary MQTT broker                            │192.168.178.76:8883                          │
├───────────────────────────────────────────────┼─────────────────────────────────────────────┤
│Failover MQTT broker                           │192.168.178.76:8883                          │
└───────────────────────────────────────────────┴─────────────────────────────────────────────┘
Error Unable to connect to device
┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mss310 un rtl8710cf (hardware:6.0.0 firmware:6.3.22)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │2208025800992751080248e1e99d5309                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │48:e1:e9:9d:53:09                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: <!Mac!>                                                        │
│             │Password: <!pass!>                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/2208025800992751080248e1e99d5309/publish              │
│             │Subscribes to: /appliance/2208025800992751080248e1e99d5309/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
Device will reboot...

Smilebags, 2024-03-11:
I've got the same issue, with verbose logging enabled the SETACK response seems to have an empty `payload`.

```

< 200 OK
< content-type: application/json
< connection: close
<
{
  header: {
    messageId: '564ae69e20c42d0fc5c4cbe6627a654d',
    namespace: 'Appliance.Config.Wifi',
    method: 'SETACK',
    payloadVersion: 1,
    from: '/appliance/2208100709673951070348e1e9a09a88/publish',
    uuid: '2208100709673951070348e1e9a09a88',
    timestamp: 151,
    timestampMs: 156,
    sign: '112e0478ef49254c2f6616f3dcef58d5'
  },
  payload: {}
}
```

bytespider, 2024-03-11:
@Smilebags @rolfmcduff have you tried `--use-wifi-x`? alternatively there's a beta version that knows to apply it automatically.  https://www.npmjs.com/package/meross/v/2.0.0-beta-1

closed 2024-11-11 by bytespider