bytespider/Meross #84: Could you add specific credentials to MQTT broker ?

nerux1, 2024-06-12:
Hello,

First, thanks for your work !

I have a request, could you, or is it possible to add specific credentials to use with a mosquitto MQTT server ?
mosquitto cannot accept user with semi-colon and Meross device use the mac address as user :(
It works fine in an anonymous mode but it is the only device with this mode.

Regards,

lechercheur123, 2024-06-12:
Or you can use the method describ by the author in the wiki : https://github.com/bytespider/Meross/wiki/MQTT#authenticated-users

nerux1, 2024-06-12:
Thanks for your reply, ok, I will try this plugin.

lechercheur123, 2024-06-12:
A simple tutorial:

First add this to your mosquitto config file:
```
plugin /usr/lib/x86_64-linux-gnu/mosquitto_dynamic_security.so
plugin_opt_config_file /etc/mosquitto/dynamic-security.json
```

Then restart your mosquitto broker.

In a terminal, initialise the dynamic-security.json file with this command:
`mosquitto_ctrl dynsec init path/to/dynamic-security.json chooseyourownadminusername`

You can rename chooseyourownadminusername with admin if you want. 
You will be asked to choose an admin password in this step.

The commands used in the next steps will depend if you use MQTT with or without TLS.

Without TLS, the beginning of the command looks like this:
`mosquitto_ctrl -u chooseyourownadminusername -h mqtt.server.fr -P adminpassword -p 1883 dynsec ...`

With TLS, the beginning of the command looks like this:
`mosquitto_ctrl -u chooseyourownadminusername -h mqtts.server.fr -P adminpassword -p 8883 --capath /etc/ssl/certs dynsec ...`

For clarity I will only use the command without TLS for this tutorial.

So, next you have to create a Role for your plugs:
`mosquitto_ctrl -u chooseyourownadminusername -h mqtt.server.fr -P adminpassword -p 1883 dynsec createRole meross-role`

Then you have to add ACL to allow the plug to send and receive messages:
`mosquitto_ctrl -u chooseyourownadminusername -h mqtt.server.fr -P adminpassword -p 1883 dynsec addRoleACL meross-role publishClientSend app/# allow`
`mosquitto_ctrl -u chooseyourownadminusername -h mqtt.server.fr -P adminpassword -p 1883 dynsec addRoleACL meross-role subscribePattern appliance/# allow`

Finally you can add your plug user:
`mosquitto_ctrl -u admin-user -h mqtts.pommetelecom.fr -P HBIgcNxZKuGDHKZhrbJS8DTg0wTgDJRp17E9UAAP -p 8883 --capath /etc/ssl/certs dynsec createClient 48:e1:e9:00:00:00`
You will be asked to enter the password for this user.

And you have to assign the Role to your user:
`mosquitto_ctrl -u admin-user -h mqtts.pommetelecom.fr -P HBIgcNxZKuGDHKZhrbJS8DTg0wTgDJRp17E9UAAP -p 8883 --capath /etc/ssl/certs dynsec addClientRole 48:e1:e9:00:00:00 meross-role`

I think that's all (of course you have to do something similar for the system that will talk to the meross plug through the MQTT broker).

nerux1, 2024-06-13:
It works fine.
Thanks

closed 2024-06-13 by bytespider