bytespider/Meross #85: what is wrong

userMak, 2024-11-10:
Hi
3 years ago you helped me to setup my meross devices with mqtt. Since then frequently the devices would disconnect and I was doing the following in my windows laptop after I was connected to the device APP and I could connected it back to home assistant.

Today I got the following error which I don't even know what it means.
Can you help me please? 

`C:\>cd C:\Meross-1.0.10\Meross-1.0.10

C:\Meross-1.0.10\Meross-1.0.10>npx meross setup --wifi-ssid xxxxxx --wifi-pass xxxxxxxxxx --mqtt mqtts://192.168.1.15

npx: installed 49 in 3.335s
C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\tree-kit\lib\extend.js:277
                        if ( ! targetValueIsObject || ! Object.hasOwn( target , targetKey ) ) {
                                                               ^

TypeError: Object.hasOwn is not a function
    at extendOneKV (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\tree-kit\lib\extend.js:277:43)
    at extendOne (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\tree-kit\lib\extend.js:189:4)
    at Object.extend (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\tree-kit\lib\extend.js:142:3)
    at Object.Terminal.create [as createTerminal] (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\terminal-kit\lib\Terminal.js:188:28)
    at terminal (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\terminal-kit\lib\termkit.js:124:18)
    at Object.get [as terminal] (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\node_modules\lazyness\lib\Lazyness.js:48:16)
    at Object.<anonymous> (C:\Users\Makis\AppData\Roaming\npm-cache\_npx\16952\node_modules\meross\lib\api.js:14:37)
    at Module._compile (internal/modules/cjs/loader.js:1068:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1097:10)
    at Module.load (internal/modules/cjs/loader.js:933:32)`

bytespider, 2024-11-11:
Do you have node >= 18 installed?

userMak, 2024-11-19:
Hi, I am not sure how but I managed to fix it (I hope). I have only windows and limited knowledge so I am not sure what make it work. ( I installed various node - npx etc :) ) I hope the next time will be solved.

closed 2024-11-22 by bytespider