bytespider/Meross #86: Cannot connect to MQTT broker (mosquitto)

crushjz, 2024-11-19:
Hello 👋 

I've the `mss310x` un mt7682 (hardware:2.0.0 firmware:2.1.2) smart plug. I've followed your guide and setup it with the following command:

```bash
./bin/meross setup --wifi-ssid <SSID> --wifi-pass <PSW> --mqtt mqtt://192.168.1.102:1883
```

The smart plug successfully connects to the WiFi, but it cannot connect to mosquitto. In the mosquitto logs, I see the following:
```
1731973176: New connection from 192.168.1.15:51314 on port 1883.
1731973176: Client <unknown> disconnected due to protocol error.
```

This is the mosquitto config that I run through docker on another machine:
```
persistence true
persistence_location /mosquitto/data/
log_dest file /mosquitto/log/log.txt
allow_anonymous true

protocol mqtt
listener 1883
```

Mosquitto works fine with HomeAssistant and Zigbee2MQTT.

Do you have any suggestion?

lechercheur123, 2024-11-19:
Hello,

The MSS310 require a MQTT_ S_ server. You can find some information in the wiki : https://github.com/bytespider/Meross/wiki/MQTT

Best regards,

lechercheur123

bytespider, 2024-11-22:
@crushjz As per https://github.com/bytespider/Meross/issues/86#issuecomment-2485208062 you need to set up a secured TLS MQTT server.

closed 2024-11-22 by bytespider

crushjz, 2024-12-22:
I made progress, and followed the guide to configure mosquitto to handle MQTTS.

In the mosquitto log, the Meross device with IP 192.168.1.15, keeps connecting and disconnecting in loop. The blue led is always flashing.

Logs:
```
1734882606: New connection from 192.168.1.15:58670 on port 8883.
1734882606: New client connected from 192.168.1.15:58670 as fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (p1, c1, k120, u'48:e1:e9:ae:a3:51').
1734882625: Client fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a disconnected.
```

debug logs:
```
1734882963: New connection from 192.168.1.15:58686 on port 8883.
1734882963: New client connected from 192.168.1.15:58686 as fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (p1, c1, k120, u'48:e1:e9:ae:a3:51').
1734882963: No will message specified.
1734882963: Sending CONNACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (0, 0)
1734882963: Received SUBSCRIBE from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a
1734882963:     /appliance/2211015244580451080648e1e9aea351/subscribe (QoS 1)
1734882963: fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a 1 /appliance/2211015244580451080648e1e9aea351/subscribe
1734882963: Sending SUBACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a
1734882963: Received PUBLISH from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (d0, q1, r0, m3, '/appliance/2211015244580451080648e1e9aea351/publish', ... (347 bytes))
1734882963: Sending PUBACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (m3, rc0)
1734882963: Received PUBLISH from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (d0, q1, r0, m4, '/appliance/2211015244580451080648e1e9aea351/publish', ... (730 bytes))
1734882963: Sending PUBACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (m4, rc0)
1734882969: Received PUBLISH from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (d0, q1, r0, m5, '/appliance/2211015244580451080648e1e9aea351/publish', ... (730 bytes))
1734882969: Sending PUBACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (m5, rc0)
1734882975: Received PUBLISH from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (d0, q1, r0, m6, '/appliance/2211015244580451080648e1e9aea351/publish', ... (730 bytes))
1734882975: Sending PUBACK to fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a (m6, rc0)
1734882981: Received DISCONNECT from fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a
1734882981: Client fmware:2211015244580451080648e1e9aea351_BlrupMtzk7LxQi4a disconnected.
```

bytespider, 2024-12-22:
It looks like you are not responding to the messages coming from the device. 

Most likely the message is `Appliance.Control.Bind`. You can see how to respond in this comment https://github.com/bytespider/Meross/issues/57.

If you're integrating the device into your Smart Home powered by Home Assistant you could use the excellent [meross_lan](https://github.com/krahabb/meross_lan) integration.

crushjz, 2024-12-23:
Indeed! That was it, installed [meross_lan](https://github.com/krahabb/meross_lan) and worked perfectly! Thanks again for your help 🙇