bytespider/Meross #87: Meross 310 hardware version 2.1.17 plugs?

HotelAlpha4, 2024-12-13:
I have successfully used this software to program Meross 310 Plugs that are hardware versions 6.x and 8.x, and that works fantastically. A long time ago, I programmed some hardware version 2.1.17 plugs, but it was so long ago I am not even sure I used this software. Either way, I need to re-pair the 2.1.17 plugs with my HA instance and I cannot get this software to connect to these plugs.

meross-info --verbose --gateway 10.10.10.1
  = "ERR_INVALID_ARG_TYPE" The 'url' argument must be of type string. Received undefined

meross-info --gateway 10.10.10.1
  = Getting info about device with IP 10.10.10.1. Error unable to connect to device. Cannot read properties of undefined (reading 'system')
  
 Same errors when trying to use meross-setup.

Is this software ok to use to program the 2.1.17 hardware Meross 310 plugs? 

Appreciate your help.

bytespider, 2024-12-13:
@HotelAlpha4 Thank you for providing detailed information about the steps you took.

To clarify, were you connected to the plug's hotspot and it had the IP 10.10.10.1 when you attempted to run the `meross-info` and `meross-setup` commands? Ensuring this connection is critical for the software to interact with the device.

Additionally, if you haven't already, consider the following steps:

Verify that the device is in pairing mode.

If the issue persists after these steps, please share any additional details or logs, and I'll do my best to assist further.

HotelAlpha4, 2024-12-13:
Thank you for your quick response. Sorry, I should have mentioned that I have put the plug into pairing mode, it flashes green and orange, I connected to its wifi AP, and doing " apr-n " on terminal shows the device ad 10.10.10.1, and " ip addr show " shows my address as 10.10.10.2.

Performing exactly the same steps for a v 6.x or v 8.0 plug works fine. But v8 I need to supply that encryption key. They work fine, just the v2.x hardware that is not connecting.

Is there a particular log that I can include that can help diagnose this?

HotelAlpha4, 2024-12-13:
I think that I actually used the pairer app @albertogeniola at https://github.com/albertogeniola/meross_pair) when I first paired these older v2.x Meross 310 plugs to my HA MQTT endpoint. I will try to give that a go this weekend. 

Happy to help with any debugging to get your pairer app working on v2.x hardware if you want to give that a go.

HotelAlpha4, 2024-12-13:
I tried another option which failed in the same way.

1. Android device with Meross app installed
2. Put plug into pairing mode, flashing orange and green
3. In Meross app on Android, add device
4. It will find the MSS310, select it
5. Supply wifi credentials
6. The MSS310 is now on my wifi network
7. I have the IP address for the plug, and pinged it and it responded (if plug is unplugged, no response)
8. I then ran:

`
./Meross/bin/meross-info  --gateway 192.168.0.30
./Meross/bin/meross-info  --gateway 192.168.0.30 --key <my key here that works for all v8 plugs>
`

Result:
`
Getting info about device with IP 192.168.0.30
Error Unable to connect to device
/home/myuser/Meross/lib/api.js:155
        const system = data.system
                            ^
TypeError: Cannot read properties of undefined (reading 'system')
    at API.deviceInformation (/home/myuser/Meross/lib/api.js:155:29)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async /home/myuser/Meross/bin/meross-info:42:5
`

if I add the --verbose:
`
Getting info about device with IP 192.168.0.30
> POST /config
> Host: 192.168.0.30
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: 'redacted',
    timestamp: 1734090394,
    sign: 'redacted'
  },
  payload: {}
}

node:url:171
  validateString(url, 'url');
  ^

TypeError [ERR_INVALID_ARG_TYPE]: The "url" argument must be of type string. Received undefined
    at Url.parse (node:url:171:3)
    at Object.urlParse [as parse] (node:url:142:13)
    at new URL (/home/myuser/Meross/lib/api.js:6:35)
    at logRequest (/home/myuser/Meross/lib/api.js:66:15)
    at handleRequestError (/home/myuser/Meross/lib/api.js:109:13)
    at API.deviceInformationData (/home/myuser/Meross/lib/api.js:221:13)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async API.deviceInformation (/home/myuser/Meross/lib/api.js:153:22)
    at async /home/myuser/Meross/bin/meross-info:42:5 {
  code: 'ERR_INVALID_ARG_TYPE'
}

Node.js v22.11.0
`

bytespider, 2024-12-13:
My error handling is clearly awful.

Could you try using the beta version? `npx meross@beta info -a 10.10.10.1` you can get more information with `npx meross@beta info --help` and `npx meross@beta setup --help`

HotelAlpha4, 2024-12-13:
I tried all of the above commands and got this:

```
/usr/bin/env: ‘node --no-warnings’: No such file or directory
/usr/bin/env: use -[v]S to pass options in shebang lines
```
This below works fine, and returns all details of the MSS310 plug at this address.
```
npx meross info -g <ip address to MSS310 hardware v6 device on my network>
```

Node v22.11.0 is installed.

Appreciate your help with this.

bytespider, 2024-12-13:
Having tried the code with a version 2 device, I can see that the issue is related to changes to how **Node 22** expects HTTP responses to be terminated with a Newline and Carriage Return, and it looks like the v2 devices are only using Newline.

Please try node 20, either the 2.0.0-beta or the 1.0.15 should work.

HotelAlpha4, 2024-12-14:
That has worked perfectly. Thank you for your help and this awesome product!

I installed NVM, then did this:
```
# Install and use 20.0.0
nvm install v20.0.0
nvm use v20.0.0
nvm ls
```

Then, the Meross pairing app worked perfectly.

closed 2024-12-14 by bytespider