bytespider/Meross #88: Cannot get MSS110 (hardware:7.0.0 firmware:7.3.46) to switch from Meross cloud to local MQTT server

lcwsoft, 2025-01-01:
I am in the process of getting my MSS110 plugs to connect to my local MQTT server and removing them from the Meross cloud, with the goal of going fully localized. Your utility works fine with older versions of the plugs (hardware version 2 and firmware version 2.x), however I got 4 of a newer version (hardware:7.0.0 firmware:7.3.46) that I cannot get to connect no matter what (using the same steps as you've documented).

To reproduce, I do the following under Ubuntu 24.04 LTS with npx installed via npm and NodeJS being managed via nvm (I tried several versions, including 22 and 23.5):

1) Remove the plug from Meross cloud using the app
2) Connect to its wifi ap
3) Run meross info to get the MQTT user and password details so I can add the needed user to my MQTT server (running version 6 provided with HomeAssistant).
4) Run Meross setup with my wifi ap details and MQTT server address, to no avail.

I ran both version 1.0.15 and the newest beta from npm, and in both cases, it seems that the requests fail with connection errors. I've attached my terminal output, with verbosity and only necessary redaction of sensitive details, to show what happens.

The Meross app has no issues setting up this plug, and your utility has no issue with older version MSS110 plugs that I have, so I'm guessing that newer versions of the device use different endpoints or require different requests to them - unfortunately that's a bit beyond my knowledge. However I will gladly try any suggestions that you may have or provide any details that you may need.

Please note that I did have to edit the shebang lines in your utilities, otherwise they would fail with "no such file" errors mentioning "nodejs --no-warnings"). Apparently that's a common issue and not specific to them.
[mss110 version 700 debug data.txt](https://github.com/user-attachments/files/18286664/mss110.version.700.debug.data.txt)

lechercheur123, 2025-01-01:
Hello,

Did you try the option '--use-wifi-x' when using meross-setup ? For my recent MSS310 plugs this is needed, but not for the older hardware revision.

lechercheur123, 2025-01-01:
Also, are you sure you are using the latest version of meross ? I know your folder is named with the latest version, but some options seem to be missing.

Can you show us the output of `meross-setup -V` ?

lcwsoft, 2025-01-01:
Tried --use-wifi-x but no change unfortunately.

lcwsoft, 2025-01-01:
lawrence@lawrence-PC:~/Downloads/meross-bak/Meross-1.0.15/bin$ ./meross-setup -V
1.0.15
lawrence@lawrence-PC:~/Downloads/meross-bak/Meross-1.0.15/bin$ 

I appreciate the thoroughness. I set up this Ubuntu install only a few days ago, and made sure I was using the latest version of meross-info/meross-setup from this repo. As you can see from the log I attached to my first message, I also tried the 2.0 beta to see if that worked.

So far, I've managed to get a bunch of older revision mss110 plugs to work fine - along with a variety of switches - it's just this newer one (and a mss510xr light switch, but that will be a separate case) that are giving me any problems.

lechercheur123, 2025-01-01:
Sorry, the beta deprecated the option `--use-wifi-x` so its normal to don't have it.

Why do you use the option `--ip 192.168.4.227` when using the beta version of meross ?

lcwsoft, 2025-01-01:
At that point, I was basically just trying anything I could think of - including if it worked after the device was connected (hence giving it the IP of the device on my AP). This is my first time using this utility and modding Meross stuff, so I didn't want to make any assumptions about what would work and what wouldn't (in case I overlooked something silly).

lechercheur123, 2025-01-01:
So you successfully connected the plug to your Wi-Fi using this utility or the official Meross app ?

lcwsoft, 2025-01-01:
For now, it's connected to the Meross cloud using their app. Hoping to get this one bound to my local MQTT server, as I got everything else on it. All I got left are just 4 of this particular version of the mss110 and 1 of the mss510xr light switches that are still connected to the Meross cloud.

lechercheur123, 2025-01-01:
Can you show me the output of `npx meross setup --wifi-ssid '<XX-iot>' --wifi-pass '111111' --mqtt 'mqtt://192.168.4.144:8883' --verbose` please ?

lcwsoft, 2025-01-01:
Sure. I just quickly removed it from Meross app, connected to the wifi ap that appeared (Meross_SW_4534) and ran that command (substituting in the real wifi details, of course):

Setting up device with IP 10.10.10.1
┌────────────────────────────────────────────────┬────────────────────────────────────────────┐
│Primary MQTT broker                             │192.168.4.144:8883                          │
├────────────────────────────────────────────────┼────────────────────────────────────────────┤
│Failover MQTT broker                            │192.168.4.144:8883                          │
└────────────────────────────────────────────────┴────────────────────────────────────────────┘
> POST /config
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'SET',
    namespace: 'Appliance.Config.Key',
    messageId: 'd5c59aa376762b077c590059fcd07681',
    timestamp: 1735771967,
    sign: 'd7fe58de3ef8813c8ab8ab873485bb6b'
  },
  payload: {
    key: {
      userId: '0',
      key: '',
      gateway: {
        host: '192.168.4.144',
        port: '8883',
        secondHost: '192.168.4.144',
        secondPort: '8883',
        redirect: 1
      }
    }
  }
}

Error Unable to connect to device
Are you connected to the device's Access Point?

lechercheur123, 2025-01-01:
Sorry, I will have to go to bed, but here are some leads:
- Try this if you have home assistant : https://github.com/bytespider/Meross/issues/72#issuecomment-1859218764 (do not implement step 4, and use option `--nowifi` instead if you are working with version 1.0.15. With the beta version, this option is not needed). Also, replace the command on step 6 by `npx meross setup --gateway 'IP' --mqtt mqtts://<IP or HOSTNAME of MQTT> --verbose --key 'KEY --nowifi'`
- You can also try this app: https://play.google.com/store/apps/details?id=com.albertogeniola.merossconf

lcwsoft, 2025-01-02:
Unfortunately neither of those options worked. However I did discover something potentially interesting. I happened to have another MSS110 in a box here, and meross-info confirms it is the same hardware version (7.0.0) but a slightly older firmware (7.3.19 instead of 7.3.46). meross-setup still fails with the same error when trying to get it to connect to my MQTT server *but* I can use meross-setup to connect to my wifi AP.

This leads me to believe that Meross has probably changed something in a recent firmware that is breaking things.

lcwsoft, 2025-01-02:
I wonder if this may be something related to the HomeKit version of these devices, as these newer MSS110 support Homekit. Whereas the ones I have that work, don't support it. I've ordered a 4 pack of both from Amazon, and am expecting them to be delivered on the 13th. Unless anyone has any other ideas, I will follow up next week and let you know my findings.

bytespider, 2025-01-02:
@lcwsoft Can you please try again with Node 20?

lcwsoft, 2025-01-02:
unfortunately no dice with node 20 either (I used the latest LTS version available via nvm, 20.18.1). meross-setup 1.0.15 fails with:

Error Unable to connect to device
Are you connected to the device's Access Point?

The 2.0 beta fails with a "socket hang up" error it seems. I'm not familiar with Node, so I'm not sure if that's just a different error for the same underlying issue. The output is a bit more verbose than 1.0.15 and may provide more info:

lawrence@lawrence-PC:~/Downloads/meross-bak/Meross-1.0.15/bin$ npx meross@beta setup --wifi-ssid 'XXXX-iot' --wifi-pass 'XXXX' --verbose
(node:3700) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
(node:3711) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
⠇ Configuring WIFI…node:internal/process/promises:391
    triggerUncaughtException(err, true /* fromPromise */);
    ^

RequestError: socket hang up
    at ClientRequest.<anonymous> (file:///home/lawrence/.npm/_npx/6f0a00686670cad7/node_modules/got/dist/source/core/index.js:792:107)
    at Object.onceWrapper (node:events:633:26)
    at ClientRequest.emit (node:events:530:35)
    at emitErrorEvent (node:_http_client:101:11)
    at Socket.socketOnEnd (node:_http_client:528:5)
    at Socket.emit (node:events:530:35)
    at endReadableNT (node:internal/streams/readable:1698:12)
    at Socket.socketOnEnd (node:_http_client:528:25)
    at Socket.emit (node:events:530:35)
    at endReadableNT (node:internal/streams/readable:1698:12)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  input: undefined,
  code: 'ECONNRESET',
  timings: {
    start: 1735855530160,
    socket: 1735855530163,
    lookup: 1735855530169,
    connect: 1735855530169,
    secureConnect: undefined,
    upload: 1735855530169,
    response: undefined,
    end: undefined,
    error: 1735855530197,
    abort: undefined,
    phases: {
      wait: 3,
      dns: 6,
      tcp: 0,
      tls: undefined,
      request: 0,
      firstByte: undefined,
      download: undefined,
      total: 37
    }
  },
  options: Options {
    _unixOptions: undefined,
    _internals: {
      request: undefined,
      agent: { http: undefined, https: undefined, http2: undefined },
      h2session: undefined,
      decompress: true,
      timeout: {
        connect: undefined,
        lookup: undefined,
        read: undefined,
        request: 10000,
        response: undefined,
        secureConnect: undefined,
        send: undefined,
        socket: undefined
      },
      prefixUrl: '',
[... 106 lines cut ...]
    },
    _merging: false,
    _init: [
      {
        timeout: { request: 10000 },
        json: QuerySystemAbilityMessage {
          header: Header {
            method: 'GET',
            namespace: 'Appliance.System.Ability',
            from: undefined,
            messageId: '091cc2d1-ef61-4123-aa6c-9b512a01218a',
            timestamp: 1735855530,
            payloadVersion: 1,
            sign: 'f423ce01b91e1482a02dc2b3739656a1'
          },
          payload: {}
        },
        method: 'post'
      }
    ]
  }
}

Node.js v20.18.1
lawrence@lawrence-PC:~/Downloads/meross-bak/Meross-1.0.15/bin$

lcwsoft, 2025-01-02:
If you need any additional information, please feel free to ask. While I'm not familiar with Node, I am a pretty quick learner and eager to help.

bytespider, 2025-01-03:
Hi sorry for the delay. We have had some trouble with HomeKit and Bluetooth devices in the past, so I wouldn't be surprised if that's related.

Based on your previous messages, I will assume that the device can be pared via WIFI as it previously responded to `npx meross info`

Could you try the following steps?
1. Resetting the device. This is typically holding the button for a few seconds until the led flashes alternately between 2 colours.
2. Connecting to the WIFI AP from the machine you are executing commands on. I would avoid using a VM if possible. If you are using one, let us know.
3. In your terminal run `npx meross@beta info --include-ability`
4. Report back

lcwsoft, 2025-01-05:
No worries :) The device that could be paired with Wifi is an MSS110 with the same hardware version, but older firmware.

I am using Ubuntu 24.04 LTS (not in a VM) with Node 20 as you asked earlier, and followed those steps. Here's the result:

lawrence@lawrence-PC:~$ nvm use 20.18.1
Now using node v20.18.1 (npm v10.8.2)
lawrence@lawrence-PC:~$ npx meross@beta info --include-ability
(node:10777) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
(node:10788) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
Getting info about device with IP 10.10.10.1

That's all I got .

bytespider, 2025-01-09:
In Ubuntu what is the output of `ip addr show` while connected to the device?

lcwsoft, 2025-01-14:
I'll follow up on this later today or tomorrow.

lcwsoft, 2025-01-15:
Updated to the new 1.0.16 version and tried again, but no luck. I connected to the wifi offered by the mss110, ran meross-setup --version, then the ip addr show, then meross-setup with both NODE_OPTIONS set and not set, as well as --use-wifi-x (mostly just to rule it out).

lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ ./meross-setup --version
1.0.16
lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: enp5s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 30:85:a9:8f:49:7e brd ff:ff:ff:ff:ff:ff
3: wlp4s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether ac:5a:fc:55:fe:92 brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.100/24 brd 10.10.10.255 scope global dynamic noprefixroute wlp4s0
       valid_lft 7146sec preferred_lft 7146sec
    inet6 fe80::794e:9f94:c4a4:fb7f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ ./meross-setup --wifi-ssid '<XXX>-iot' --wifi-pass 'XXXXX' --verbose
Setting up device with IP 10.10.10.1
> POST undefined
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: 'c0f2c368ce1e5b4460a00cba42442931',
    timestamp: 1736948201,
    sign: '27bd0bdea9ad722a686439fb7e723999'
  },
  payload: {}
}

< 200 OK
< content-type: application/json
< connection: close
<
{
  header: {
    messageId: 'c0f2c368ce1e5b4460a00cba42442931',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/2307069048441951200148e1e9cf4546/publish',
    uuid: '2307069048441951200148e1e9cf4546',
    timestamp: 425,
    timestampMs: 625,
    sign: 'a28a56be11e48e33957251c7947e394b'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss110',
          subType: 'un',
          version: '7.0.0',
          chipType: 'rtl8710cm',
          uuid: '2307069048441951200148e1e9cf4546',
          macAddress: '48:e1:e9:cf:45:46'
        },
        firmware: {
          version: '7.3.46',
          homekitVersion: '6.3',
          compileTime: '2024/08/21-10:15:09',
          encrypt: 1,
[... 310 lines cut ...]
> POST undefined
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'SET',
    namespace: 'Appliance.Config.WifiX',
    messageId: '95f5e3ccc881d29caf40c715986c7c5b',
    timestamp: 1736948227,
    sign: '89a5c946eb3a3c4cf0465558ce1bd5ba'
  },
  payload: {
    wifi: {
      ssid: 'bGFyMjQtaW90',
      password: 'UYdGb1ayKCnSM4ch+zZSYkMi9s+jZ1CZzNB3IMl8y9Y='
    }
  }
}

Error Unable to connect to device
Are you connected to the device's Access Point?
lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$

lcwsoft, 2025-01-16:
Sorry about the formatting :)

To answer your thoughts:

1) I tried meross-setup with --mqtt too, and no difference.

`lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ NODE_OPTIONS='--insecure-http-parser' ./meross-setup --wifi-ssid '<XXX>-iot' --wifi-pass 'XXXXX' --verbose --mqtt 'mqtt://192.168.4.144:8883'
Setting up device with IP 10.10.10.1
┌────────────────────────────────────────────────┬────────────────────────────────────────────┐
│Primary MQTT broker                             │192.168.4.144:8883                          │
├────────────────────────────────────────────────┼────────────────────────────────────────────┤
│Failover MQTT broker                            │192.168.4.144:8883                          │
└────────────────────────────────────────────────┴────────────────────────────────────────────┘
> POST undefined
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'SET',
    namespace: 'Appliance.Config.Key',
    messageId: 'f90c980bc821b6eb09cf37bf284aa9ef',
    timestamp: 1736989216,
    sign: '01e44a72c7f9ea123157e2479790dc59'
  },
  payload: {
    key: {
      userId: '0',
      key: '',
      gateway: {
        host: '192.168.4.144',
        port: '8883',
        secondHost: '192.168.4.144',
        secondPort: '8883',
        redirect: 1
      }
    }
  }
}

(node:2357) Warning: Using insecure HTTP parsing
(Use `node --trace-warnings ...` to show where the warning was created)
Error Unable to connect to device
Are you connected to the device's Access Point?
`

--include-wifi doesn't work on the current firmware either:

`

lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ NODE_OPTIONS='--insecure-http-parser' ./meross-info --include-wifi
Getting info about device with IP 10.10.10.1
(node:2380) Warning: Using insecure HTTP parsing
(Use `node --trace-warnings ...` to show where the warning was created)
┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.46)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │2307069048441951200148e1e9cf4546                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │48:e1:e9:cf:45:46                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: 48:e1:e9:cf:45:46                                                        │
│             │Password: 0_8737bd6ae86640ba6e396e20b0bf4bbd                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/2307069048441951200148e1e9cf4546/publish              │
│             │Subscribes to: /appliance/2307069048441951200148e1e9cf4546/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
⠇ Getting WIFI list…
Error Unable to connect to device
Are you connected to the device's Access Point?
lawrence@lawrence-PC:~/Downloads/meross-1.0.16/Meross-1.0.16/bin$ 
`
I do seem to recall that --include-wifi worked on the original firmware for the version 7.0.0 of the MSS110 (7.3.19). I've since managed to get a new in box one (with those specs) to connect to wifi using the earlier meross-setup command on 1.0.15 with just wifi details, and then link up with HomeAssistant using their HomeKit integration. It's not what I hoped, but it's a decent enough compromise to accomplish my original goal (which was to get off Meross cloud). That leads me to think that they've changed something in a firmware update that prevents meross-setup from working as intended.

I've spent a fair bit of time tinkering around with these plugs and don't know if there's anything else I can do. Unfortunately between work and life, it might be quite some time before I can get deeper. In the meantime, I'm going to reach out to Meross as I've heard that they will remotely downgrade firmware and see if I can get my remaining plugs down to an older firmware and get up on Homekit to at least resolve this for me.

If I find another of these plugs laying around and don't have a need for it, would you be open to having me send it to you? I tried ordering 2 packs of these plugs from Amazon over Christmas, and ended up with a newer hardware version (8.0.0) that used Bluetooth instead of wifi for setup, and ended up returning them. I don't know if version 7.0.0 is readily available anymore, but would gladly offer one if I can and it would help out.

bytespider, 2025-01-16:
Thanks for the updates.

I'll try to delve into the android app and see of there is anything specific for the MSS110 + version 7.3.46 firmware and respond back if I discover anything.

If you do find a spare device I'll happily accept it for science. You can email me to arrange.

maxroulstone, 2025-01-19:
Hi, just for recording keeping I am also having the same issues with the same device (homekit enabled) 

┌─────────────┬─────────────────────────────────────────────────────────────────
│Device       │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.37)             
├─────────────┼─────────────────────────────────────────────────────────────────
│UUID         │2212269185815051030248e1e9b62c49                                 
├─────────────┼─────────────────────────────────────────────────────────────────
│Mac address  │48:e1:e9:b6:2c:49                                                
├─────────────┼─────────────────────────────────────────────────────────────────
│IP address   │10.10.10.1                                                       
├─────────────┼─────────────────────────────────────────────────────────────────
│Credentials  │User: 48:e1:e9:b6:2c:49                                          
│             │Password: 0_14ef86a5626abee0feec7585d51feee5                     
├─────────────┼─────────────────────────────────────────────────────────────────
│MQTT topics  │Publishes to: /appliance/2212269185815051030248e1e9b62c49/publish
│             │Subscribes to: /appliance/2212269185815051030248e1e9b62c49/subscr
└─────────────┴─────────────────────────────────────────────────────────────────

best of luck on this :)

bytespider, 2025-01-20:
I've added some field to the header of the messages. I couldn't see anything else in the source that's missing.
When you have a moment and a free device, please checkout https://github.com/bytespider/Meross/tree/88-mss110-firmware7346 and test.

lcwsoft, 2025-01-20:
I'm off today, so I had a chance to give that updated version with one of the MSS110 plugs running firmware 7.3.46 (hardware version 7.0.0). Unfortunately no luck:

```
lawrence@lawrence-PC:~/Downloads/Meross-88-mss110-firmware7346/bin$ NODE_OPTIONS='--insecure-http-parser' ./meross-setup --wifi-ssid '<XXX-iot>' --wifi-pass '111111' --verbose
Setting up device with IP 10.10.10.1
> POST undefined
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '',
    method: 'GET',
    namespace: 'Appliance.System.All',
    messageId: '8ad0c33265b415d5fdf4b8f044b33fcf',
    timestamp: 1737403865,
    sign: 'b3c34c48a32af53cea71baa75e0c7a03'
  },
  payload: {}
}

(node:5919) Warning: Using insecure HTTP parsing
(Use `node --trace-warnings ...` to show where the warning was created)
< 200 OK
< content-type: application/json
< connection: close
<
{
  header: {
    messageId: '8ad0c33265b415d5fdf4b8f044b33fcf',
    namespace: 'Appliance.System.All',
    method: 'GETACK',
    payloadVersion: 1,
    from: '/appliance/2307065833780651200148e1e9cf4534/publish',
    uuid: '2307065833780651200148e1e9cf4534',
    timestamp: 478,
    timestampMs: 272,
    sign: 'd6e15da7f91a4e7be6cda309a72efea0'
  },
  payload: {
    all: {
      system: {
        hardware: {
          type: 'mss110',
          subType: 'un',
          version: '7.0.0',
          chipType: 'rtl8710cm',
          uuid: '2307065833780651200148e1e9cf4534',
          macAddress: '48:e1:e9:cf:45:34'
        },
        firmware: {
          version: '7.3.46',
          homekitVersion: '6.3',
          compileTime: '2024/08/21-10:15:09',
          encrypt: 1,
          wifiMac: '',
          innerIp: '10.10.10.1',
          server: '',
          port: 0,
          userId: 0
        },
        time: { timestamp: 478, timezone: '', timeRule: [] },
        online: { status: 2, bindId: '', who: 0 }
      },
      digest: {
        togglex: [ { channel: 0, onoff: 1, lmTime: 0 } ],
        triggerx: [],
        timerx: []
      }
    }
  }
}

┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.46)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │2307065833780651200148e1e9cf4534                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │48:e1:e9:cf:45:34                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: 48:e1:e9:cf:45:34                                                        │
│             │Password: 0_a4bcc912c38285327ab6b0aaeae92ee3                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/2307065833780651200148e1e9cf4534/publish              │
│             │Subscribes to: /appliance/2307065833780651200148e1e9cf4534/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
> POST undefined
> Host: 10.10.10.1
> Accept: application/json, text/plain, */*
> Content-Type: application/json
>
{
  header: {
    from: '/app/0-App_2307065833780651200148e1e9cf',
    method: 'SET',
    namespace: 'Appliance.Config.Wifi',
    messageId: 'ff9fbd3501b254eb2b9560c41be40f27',
    timestamp: 1737403865,
    sign: '2b162648dfffd3db552564f3399b00f3'
  },
  payload: { wifi: { ssid: 'bGFyMjQtaW90', password: 'OTQ2NzEzMjU=' } }
}

Error Unable to connect to device
Are you connected to the device's Access Point?
lawrence@lawrence-PC:~/Downloads/Meross-88-mss110-firmware7346/bin$

```

lcwsoft, 2025-01-20:
meross-info output:

```
lawrence@lawrence-PC:~/Downloads/Meross-88-mss110-firmware7346/bin$ NODE_OPTIONS='--insecure-http-parser' ./meross-info --include-wifi
Getting info about device with IP 10.10.10.1
(node:5548) Warning: Using insecure HTTP parsing
(Use `node --trace-warnings ...` to show where the warning was created)
┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.46)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │2307065833780651200148e1e9cf4534                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │48:e1:e9:cf:45:34                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: 48:e1:e9:cf:45:34                                                        │
│             │Password: 0_a4bcc912c38285327ab6b0aaeae92ee3                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/2307065833780651200148e1e9cf4534/publish              │
│             │Subscribes to: /appliance/2307065833780651200148e1e9cf4534/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
⠇ Getting WIFI list…
Error Unable to connect to device
Are you connected to the device's Access Point?
lawrence@lawrence-PC:~/Downloads/Meross-88-mss110-firmware7346/bin$ NODE_OPTIONS='--insecure-http-parser' ./meross-info 
Getting info about device with IP 10.10.10.1
(node:5557) Warning: Using insecure HTTP parsing
(Use `node --trace-warnings ...` to show where the warning was created)
┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.46)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │2307065833780651200148e1e9cf4534                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │48:e1:e9:cf:45:34                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: 48:e1:e9:cf:45:34                                                        │
│             │Password: 0_a4bcc912c38285327ab6b0aaeae92ee3                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/2307065833780651200148e1e9cf4534/publish              │
│             │Subscribes to: /appliance/2307065833780651200148e1e9cf4534/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
lawrence@lawrence-PC:~/Downloads/Meross-88-mss110-firmware7346/bin$
```

bytespider, 2025-01-21:
Thanks for trying. Looking further into the code and the responses you provided, I have noticed a flag regarding encryption. It's possible the device expects the Homekit data to be passed encrypted to the device before it's willing to continue.

I don't have the time to look into this currently but maybe towards the weekend.

bytespider, 2025-04-06:
@lechercheur123 @lcwsoft @maxroulstone @fuomag9 
If you are able please try with the latest beta that includes encrypted communication where required.

`npx meross@beta setup`

Please feedback your results.

lcwsoft, 2025-04-08:
I tried the new beta and it was able to connect to and configure the device:

`
lawrence@lawrence-PC:~$ nodejs --version
v20.16.0
lawrence@lawrence-PC:~$ npx meross@beta info
(node:18137) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
(node:18148) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
┌───────────┬──────────────────────────────────────────────────────────────────┐
│Device     │mss110 un rtl8710cm (hardware:7.0.0 firmware:7.3.46)              │
├───────────┼──────────────────────────────────────────────────────────────────┤
│UUID       │2307069048441951200148e1e9cf4546                                  │
├───────────┼──────────────────────────────────────────────────────────────────┤
│Mac address│48:e1:e9:cf:45:46                                                 │
│           │                                                                  │
├───────────┼──────────────────────────────────────────────────────────────────┤
│IP address │10.10.10.1                                                        │
├───────────┼──────────────────────────────────────────────────────────────────┤
│Credentials│User: 48:e1:e9:cf:45:46                                           │
│           │Password: 0_aec8f4be8f573d5dacd177953269e16e                      │
├───────────┼──────────────────────────────────────────────────────────────────┤
│MQTT topics│Publishes to: /appliance/2307069048441951200148e1e9cf4546/publish │
│           │Subscribes to:                                                    │
│           │/appliance/2307069048441951200148e1e9cf4546/subscribe             │
└───────────┴──────────────────────────────────────────────────────────────────┘
✓Exchanging public keys…
lawrence@lawrence-PC:~$ npx meross setup --wifi-ssid 'xxxxx-iot' --wifi-pass 'xxxxxxxx' --mqtt 'mqtt://192.168.4.144:8883'
(node:18260) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
(node:18271) ExperimentalWarning: Importing JSON modules is an experimental feature and might change at any time
(Use `node --trace-warnings ...` to show where the warning was created)
✓Fetching device information…mss110 (hardware: 7.0.0, firmware: 7.3.46)
✓Fetching device abilities…
✓Exchanging public keys…
✓Configuring MQTT brokers…
✓Configuring WIFI…
Device will now reboot…
lawrence@lawrence-PC:~$
`
However it doesn't appear as a new device to add with MerossLan in HomeAssistant. Everything on the HA side is up to date and there's no matching device in MerossLan already (so I would expect it to appear as a new device to add), I created the HA user as needed for Mosquitto MQTT and the log shows it connects to MQTT successfully. It does show up as a new device for HomeKit and asks me if I want to add it. I suspect that this may just an issue on my HA installation, so I'll debug that a bit further.

bytespider, 2025-04-08:
In Meross LAN you need to add a new configuration with the key `meross` if you didn't set one yourself. Sorry I should have mentioned the default key change. 
The reason for this is that the empty key was rejected by the devices I tested when encrypting the messages. I was unable to work out why, so I just defaulted to something simple. @krahabb you might need to know about this change also.

lcwsoft, 2025-04-08:
I can confirm that manually adding the device to Meross LAN with the IP assigned by my router and key set to **meross** works.

bytespider, 2025-04-09:
@lcwsoft as explained above to be auto detected the device has to have communicated with the MQTT broker.

Please make sure the device can communicate with the broker. You will likely need to add a user login to the broker before pairing. I don't use the default Mosquitto broker because, unless things have changed, it doesnt allow for Mac Addresses as usernames, which Meross devices use.

krahabb, 2025-04-10:
I confirm this:
Once the proper ('meross') key is set in the 'MQTT Hub' every message incoming from the HA/MQTT interface will be managed and replied by meross_lan eventually proposing the 'discovery' of new devices in order to confirm them.
Once a device is configured the key used for the device is stored in every device configuration (so to not rely anymore on the MQTT Hub: it's purpose is to just correctly detect incoming new devices and 'discover' them)

At any rate, if the key in the MQTT Hub prove to be wrong you should anyway see the discovery, just the software cannot communicate correctly and inspect the source device. This will be done once you set the proper key in the discovered device.

As a last 'issue', some devices (I don't remember which) expect to dialogate a bit with the software 'behind' the broker when initially connecting (this is the phase where the MQTT Hub key is used since the device is not yet configured in meross_lan). At this point they might fail if not receiving the proper replies from the software stack and so they reset themeselves (or enter a dead-loop).
This happens if the MQTT Hub key is wrong. At this point the discovery process, even if started might fail.

lcwsoft, 2025-04-13:
I am using the default Mosquitto MQTT broker (version 6.5.0) from HomeAssistant and can confirm it works fine. I just add the required user for each device (user name being the mac address and password being the related password).

I ended up setting up the last of my MSS110 devices today using the beta and everything is working as expected.

lcwsoft, 2025-04-13:
Closing this as resolved by the latest beta.

closed 2025-04-13 by lcwsoft