bytespider/Meross #96: Problem setting up MS200

gnulabis, 2025-02-23:
Hi,

Thank you very much for your work. I ordered an MS200 floor-heating thermostat and used your tool to configure it to use my local MQTT in home assistant, it all worked great.

So much that I ordered a second one for another part of my house. However, with the second thermostat I cannot complete the setup.

I connect to the device's AP and I can see it:

```
$ ./meross-info 
Getting info about device with IP 10.10.10.1
┌─────────────┬───────────────────────────────────────────────────────────────────────────────┐
│Device       │mts200 eu rtl8710cm (hardware:7.0.0 firmware:7.6.11)                           │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│UUID         │xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx                                               │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Mac address  │c4:e7:ae:0f:19:48                                                              │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│IP address   │10.10.10.1                                                                     │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│Credentials  │User: c4:e7:ae:0f:19:48                                                        │
│             │Password: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx                                   │
├─────────────┼───────────────────────────────────────────────────────────────────────────────┤
│MQTT topics  │Publishes to: /appliance/24110618749011601203c4e7ae0f1948/publish              │
│             │Subscribes to: /appliance/24110618749011601203c4e7ae0f1948/subscribe           │
└─────────────┴───────────────────────────────────────────────────────────────────────────────┘
```

But when I try to set it up I get:
```
$ ./meross-setup --gateway 10.10.10.1 --wifi-ssid ABC --wifi-pass pswd --mqtt mqtt://core-mosquito:8183
Setting up device with IP 10.10.10.1
┌────────────────────────────────────────────────┬────────────────────────────────────────────┐
│Primary MQTT broker                             │core-mosquito:8183                          │
├────────────────────────────────────────────────┼────────────────────────────────────────────┤
│Failover MQTT broker                            │core-mosquito:8183                          │
└────────────────────────────────────────────────┴────────────────────────────────────────────┘
Error Unable to connect to device
Are you connected to the device's Access Point?
```

I then checked my first thermostat, the one that I had configured without any issue:

```
$ ./meross-info -g 192.168.4.25
Getting info about device with IP 192.168.4.25
┌────────────────────┬────────────────────────────────────────────────────────────────────────┐
│Device              │mts200 eu rtl8710cm (hardware:7.0.0 firmware:7.6.4)                     │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│UUID                │xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx                                        │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Mac address         │48:e1:e9:a9:a0:18                                                       │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│IP address          │192.168.4.25                                                            │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Current MQTT broker │core-mosquito:8183                                                      │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│Credentials         │User: 48:e1:e9:a9:a0:18                                                 │
│                    │Password: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx                            │
├────────────────────┼────────────────────────────────────────────────────────────────────────┤
│MQTT topics         │Publishes to: /appliance/2209284982828160120348e1e9a9a018/publish       │
│                    │Subscribes to: /appliance/2209284982828160120348e1e9a9a018/subscribe    │
└────────────────────┴────────────────────────────────────────────────────────────────────────┘
```

So, it seems that there's a newer firmware version in my latest thermostat which introduces some breaking change? How can I debug this further?

bytespider, 2025-04-06:
This issue is likely caused by the introduction of encrypted messages for configuring the device.

If you are able please try with the latest beta that includes encrypted communication where required.

`npx meross@beta setup`

Please feedback your results.

gnulabis, 2025-05-21:
Apologies for the late reply, yes it's all working now. Many thanks.