bytespider/Meross #97: Problem with MS210HK

ruizcrp, 2025-03-24:
Hi
I didn't manage to connect with the MS210HK (home kit).
mss210 un rtl8710cm (hardware:7.0.0 firmware:7.3.9) 
I was following the tutorial here: https://www.creatingsmarthome.com/index.php/2022/08/28/guide-local-mqtt-broker-with-meross-and-home-assistant/

Actually first it had firmware 7.3.3.
meross-info worked with that version:
./meross-info --verbose --include-wifi
but meross-setup did not work:
./meross-setup --gateway 10.10.10.1 --wifi-ssid YourNetworkSSID --wifi-pass YourNetworkPassword --mqtt mqtts://192.168.1.100:8883 --wifi-cipher 6 --wifi-encryption 7 --wifi-channel 1 --verbose

I got Error Unable to connect to device
Are you connected to the device's Access Point?

Then I grudgingly downloaded the Meross-app in order to do a firmware update hoping this would solve the issue, new to 7.3.9.
Now, not even meross-info connects to it, with the same error message. I checked that I was well connected to the wifi of the device. 

I am using 2.4Ghz frequency.

Ran out of ideas what to do now.

bytespider, 2025-03-27:
Unfortunately theres an ongoing issue with the newer firmwares requiring the messages are encrypted after the initial connection.

I hope to find the time to fix this at the weekend.

lcwsoft, 2025-04-05:
Hmmm I wonder if addressing this will address my reported issues as well. I'll be keeping an eye on this and will give any new versions a try when they're available to see what happens.

bytespider, 2025-04-05:
@lcwsoft @ruizcrp I might need you to test a version later this weekend. I have successfully paired a MSS110 (hardware:7.0.0 firmware:7.3.46) device and set it up in HA with MerossLAN

bytespider, 2025-04-06:
The fixes should address any device which had the ability Appliance.Encrypt.ECDHE.
Typically this has been devices with a 7.x firmware version.

Changes are currently in the feature/rewrite [branch](https://github.com/bytespider/Meross/tree/feature/typescript-rewrite) if your comfortable running the build script.

I will build a version for NPM in several hours

bytespider, 2025-04-06:
@ruizcrp 

If you are able please try with the latest beta that includes encrypted communication where required.

`npx meross@beta setup`

Please feedback your results.

ruizcrp, 2025-04-06:
Hi @bytespider 
Thank you very much for your efforts to fix it! I’m unfortunately right now on travels and not home until Saturday! I will try it as soon as I’m home!
Kind regards!

ruizcrp, 2025-04-12:
Hi  @bytespider 
I tried the above command. For the moment it says for npx meross@beta setup :
crypto is not defined.

bytespider, 2025-04-12:
@ruizcrp You're absolutely right—thank you for looking into the cause; that was really helpful. I realized I forgot to test on versions 18 and 20. I've now updated the fix and republished it.

ruizcrp, 2025-04-12:
I'm on it. It seems to be working:
npx meross@beta setup
✓Fetching device information…mss210 (hardware: 7.0.0, firmware: 7.3.9)
✓Fetching device abilities…
✓Exchanging public keys…
and finishes without error.

ruizcrp, 2025-04-12:
Yes, it seems to be working! It is talking to mqtt. 
Good job, and thank you again!

closed 2025-04-12 by ruizcrp