bytespider/Meross #105: How to install 2.0.0 version?

lcwsoft, 2025-08-20:
Hello,

I'm trying to install the latest version after a fresh install of Ubuntu 24.04 with nodejs and npm from apt, but none of the instructions are working. The nodejs package link at https://github.com/bytespider/Meross is broken: https://github.com/bytespider/Meross/actions/workflows/npm-ghr-publish.yml

The instructions at https://github.com/bytespider/Meross/releases/tag/2.0.0 result in 1.0.16 being installed:

```
lawrence@lawrence-PC:~$ sudo npm install -g meross

added 48 packages in 3s

7 packages are looking for funding
  run `npm fund` for details
lawrence@lawrence-PC:~$ sudo npm exec -c 'meross info'
Getting info about device with IP 10.10.10.1
^C
lawrence@lawrence-PC:~$ sudo npm exec -c 'meross info --version'
1.0.16
```

And the npx instructions there just don't work at all:

```
lawrence@lawrence-PC:~$ npx meross-info --help
npm ERR! code E404
npm ERR! 404 Not Found - GET https://registry.npmjs.org/meross-info - Not found
npm ERR! 404 
npm ERR! 404  'meross-info@*' is not in this registry.
npm ERR! 404 
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/lawrence/.npm/_logs/2025-08-20T04_26_14_561Z-debug-0.log
lawrence@lawrence-PC:~$ 
```

I checked https://www.npmjs.com/package/@meross/lib?activeTab=readme and it shows that the latest version is 2.0.0-beta-3 from 4 months ago.

I also checked the wiki section and the only instructions are for older versions: https://github.com/bytespider/Meross/wiki/MQTT

I don't use nodejs nor npm on a daily basis, so I feel like I'm missing something obvious here. However I have no idea as to what it could be.

If we can figure it out, I'd be happy to write some documentation to help avoid this in the future.

bytespider, 2025-08-21:
Thanks for the reminder. I botched the 2.0.0 deployment to NPM so I pulled it.
I have now fixed and redeployed. I'm still working on the documentation but I ran out of time. I'll try to help when I can.

Basic flow is here for [WIFI pairing](https://github.com/bytespider/Meross/blob/fix-readme/docs/wifi-pairing.md), any suggestions for updates to the documentation are welcome.

lcwsoft, 2025-08-24:
FInally gotten back to this. Updated from npm and its working good :) The only suggestions would be to fix the broken Node.js link at https://github.com/bytespider/Meross and update both sample commands on that page (both use hyphens which don't work when I checked - they need spaces instead).